meatpiHQ/wican-fw #118: Feature request: MQTT rx subtopic

jenniferlee1818, 2024-04-15:
All of my sensors are setup in home assistant as MQTT sensors all listening to the same topic, so it would be more useful to receive MQTT on a subtopic like wican/xxxxxxxxxxxx/can/rx/PID.  Is it possible?

harkoprinssen, 2024-06-21:
Would be nice indeed.
Currently set up 3 filters, but as they are put in the same topic my HA mqtt sensors get emptied as the json value is only present in 1 of 3 messages

amicomit, 2024-07-24:
Hi yes, I'm having excellent results using the Wican filters, and actually, if it supported two byte PIDs and the filter Delete glitch was fixed, I wouldn't be having any problems at all, it works really well. Below is a sample RX and TX for Tyre Pressure 1 - here the PID hex is 2076 which is 8310 dec, and represented by bytes 2 and 3 (32,118) in the TX and RX - but I can only use byte 3 with index 3 in the Wican filter (bottom). This works OK, but in the end I'm likely to run into a duplicate number for the PID

Type Pressure 1
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}

CAN ID___Name___PID___Index___Start Bit__Length___Expression
1881_____TPMS1___118_____3________0_________1______ (B5*0.01373)*14.5038

[image]