meatpiHQ/wican-fw #119: SoC and AT of FIAT e-Ulysse (same as: Toyota Proace EV, Citroen Spacetourer, Peugeot Traveller) Stellantis EV

tqma1, 2024-04-17:
Hi,

i have done all like in the description "EV Battery Examples".

I have now got for the SoC (value is 75,72%) :
{"bus":"0","type":"tx","ts":38585,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":51733,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,151,113,0,0]}]}

and the Ambient Temperature (value is 5°C)
{"bus":"0","type":"tx","ts":46736,"frame":[{"id":1698,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,52,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":34437,"frame":[{"id":1666,"dlc":5,"rtr":false,"extd":false,"data":[4,98,212,52,5,0,0,0]}]}

But now i am in trouble - i do not know how to setup the CAN to JSON interrupter correct.

I am on the latest FW 2.98 - there are also more fields as on all screenshots...

BR, Thomas

tqma1, 2024-04-17:
some more snippets of SoC (maybe it is helpfull - value of Soc at this time i can´t remember)

{"bus":"0","type":"tx","ts":20179,"frame":[{"id":1698,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,52,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}
{"bus":"0","type":"rx","ts":23868,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,69,19,0,0]}]}

meatpiHQ, 2024-04-18:
Hey @tqma1 I need more data points like this with the SoC value to be able to find the right expression.

> I have now got for the SoC (value is 75,72%) :
> {"bus":"0","type":"tx","ts":38585,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
> {"bus":"0","type":"rx","ts":51733,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,151,113,0,0]}]}

tqma1, 2024-04-19:
Here we go:

SoC: is 69,75% (from CarScanner App)

{"bus":"0","type":"rx","ts":51504,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,139,127,0,0]}]}
{"bus":"0","type":"rx","ts":10066,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,139,127,0,0]}]}
{"bus":"0","type":"rx","ts":19105,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,139,127,0,0]}]}

Ambient Temperature is 6°C
{"bus":"0","type":"rx","ts":2177,"frame":[{"id":1666,"dlc":5,"rtr":false,"extd":false,"data":[4,98,212,52,6,0,0,0]}]}

meatpiHQ, 2024-04-19:
The values are very close..

[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,151,113,0,0]}]}.  **151/2 = 75.5%** 

{"bus":"0","type":"rx","ts":51504,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,139,127,0,0]}]} **139/2= 69.5%**

tqma1, 2024-04-19:
So it should be like this for SoC:

CAN ID = 1684
PID = 212
Start Bit = 0
Bit Length = 1
Expression = B4/2
Cycle ms = 1000

meatpiHQ, 2024-04-19:
Yup that should do it

tqma1, 2024-04-21:
Thank you so long, but now I stuck in the same issue:

https://github.com/meatpiHQ/wican-fw/issues/121

and I readed and tried everything from here what michi741 did:
https://github.com/meatpiHQ/wican-fw/issues/91

meatpiHQ, 2024-04-23:
@tqma1 Can you please post your device configuration page.

tqma1, 2024-04-24:
Got it - now it works!
I did not use the correct index number. (correct is 2)
Your answer here https://github.com/meatpiHQ/wican-fw/issues/121 pointed me to the right way.

Thank you for your support!

[image]

atze09, 2024-05-03:
> Got it - now it works! I did not use the correct index number. (correct is 2) Your answer here #121 pointed me to the right way.
> 
> Thank you for your support!
> 
> <img alt="WiCAN" width="574" src="https://private-user-images.githubusercontent.com/1890699/325209316-d8df9e04-6795-42ad-9a3d-59d31ff6c82a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3MzYxMDUsIm5iZiI6MTcxNDczNTgwNSwicGF0aCI6Ii8xODkwNjk5LzMyNTIwOTMxNi1kOGRmOWUwNC02Nzk1LTQyYWQtOWEzZC01OWQzMWZmNmM4MmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDNUMTEzMDA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmUxOTI3NDllYTkxZTIyZWEyYjFlYjBhMGFkZDY4MGZlNWJkMmZkZTM0YTgwYWJiY2UwZDZmYjJmMTM4ZGQ5MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gkWKugjo4na_3M8MP22TJKqZezqziFIhPtNDO0m7EGE">

Which Stellantis car do you exactly have? I own a Opel Corsa and i am not able to get some RX data out of the car when engine is off. Is it working for you with engine off?
How do you trigger TX? i sent it via home assistant but currently this parts is also not working for me.
Thanks for answering the above questions.

tqma1, 2024-05-03:
Hi!
I have a FIAT e-Ulysse as written in the title of my ticket.
When ignition is off no data can be received.
But you can put the ignition on without pressing the breake pedal and you are able to receive data.
I do it in homeassistant as here described:
https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example
You can check if it works with the MQTT Explorer.

atze09, 2024-05-03:
> Hi! I have a FIAT e-Ulysse as written in the title of my ticket. When ignition is off no data can be received. But you can put the ignition on without pressing the breake pedal and you are able to [...]

Thanks for your feedback. My plan was to get data also during charging and if car is in garage without anyone pressing the ignition button. Is there a way to start ignition from remote?
What do you do with the doc and AT values? Are those only for Viewing them?because the time slot of ignition on and car in wifi area is quite small.

tqma1, 2024-05-25:
Sorry for my late response, I have been on vacation.
When I am coming home the WiCAN connect to my WiFi and sends the SoC (and other values) to my home assistant and there is calculated how much energy should be loaded into my car that I am reaching a SoC of 80%.
That’s it - no need to have the ignition on.

naseweiss99, 2024-06-27:
Puuh gents, this gave me a bit of confusion.
I configured the Wican right as above  described. 
[image]

However I was unaware that the Wican doesn't start transmitting in the time intervall defined in the web frontend.
What are this milliseconds for ? Every Mqtt device I have see is automatically transmitting - Wican not.

Does it really mean I need to pull the data ?
in Fhem I use the implemented MQTT server and every time i publish this:

_set MQTT2_FHEM_Server publish wican/XXXXXXXXXX/can/tx {"bus":"0","type":"tx","frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}_

I get the SOC - however, how to force automatic transmissions from the Wican?

thanks

tqma1, 2024-06-27:
I do not use Fhem, I use Homeassistant.

When the Wican is available in my network I start a automation which publish the Mqtt call every 5 seconds …

naseweiss99, 2024-06-27:
but you need to pull the data - right ?

If I look at Tasmota and many other MQTT devices, they send in determined time intervalls.

tqma1, 2024-06-27:
Both statements are correct.

naseweiss99, 2024-06-28:
And data is only received while ignition is on ?

tqma1, 2024-06-28:
Yes, see my post from may the 3th.

atze09, 2024-08-16:
> Puuh gents, this gave me a bit of confusion. I configured the Wican right as above described. [...]

Did you already pulled some more data out of your car and maybe can provide the tx and the Wican filter for those?

I would like to also have odometer(mileage), soh (State of health) maybe someone already implemented these?

atze09, 2024-09-21:
@meatpiHQ 
With this information we now could build a profile?

We only need the right calculation or?

`#######
####### Setup commands
#######
ATWS ---> Warm reset set back to defaults
ATI ---> Show interface ID
ATE0 ---> Turn off terminal echo
AT@1 --->
ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
ATAT1 --->
ATL0 ---> Disable line feed
ATS0 ---> Disable sending spaces between hex characters
ATH1 ---> Turn on odb2 header
ATCAF1 ---> Turn on CAN auto formatting
ATSH79B ---> Set request message header 79B
ATFCSH79B ---> Set flow control header (CAN ID)
ATFCSD300000 ---> Set flow control data
ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data

#######
####### Data commands
#######
ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
ATFCSH6A2 ---> Set flow control header (CAN ID)
ATCRA682 ---> Set the CAN hardware filter to 682
22D4341 ---> Ambient temperature
22D8EF1 ---> Battery temperature
22D4021 ---> Speed
ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
ATFCSH6B4 ---> Set flow control header (CAN ID)
ATCRA694 ---> Set the CAN hardware filter to 694
22D8161 ---> Battery current
22D8151 ---> Battery voltage
22D86F1 ---> Battery minimum voltage
22D8701 ---> Battery maximum voltage
22D4101 ---> SOC calibrated
22D8651 ---> kWh available
22D8601 ---> SOH
22D8101 ---> SOC`

meatpiHQ, 2024-09-22:
@atze09 we still need the calculation. See the profile example below 

https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles/toyota/rav4.json

tqma1, 2025-04-26:
Profile for Fiat 600e works!

tqma1, 2025-05-02:
This works now for me:

AT= Ambient temperature
BT= Battery temperature

      {
        "Name": "AT",
        "Init": "ATSH6A2;ATFCSH6A2;ATCRA682;",
        "PID": "22D4341",
        "Expression": "B4",
        "MinValue": "",
        "MaxValue": "",
        "Period": "5000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "BT",
        "Init": "ATSH6A2;ATFCSH6A2;ATCRA682;",
        "PID": "22D8EF1",
        "Expression": "B4",
        "MinValue": "",
        "MaxValue": "",
        "Period": "5000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "min_mV",
        "Init": "ATSH6B4;ATFCSH6B4;ATCRA694;",
        "PID": "22D86F1",
        "Expression": "[B4:B5]",
        "MinValue": "",
        "MaxValue": "",
        "Period": "5000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "max_mV",
        "Init": "ATSH6B4;ATFCSH6B4;ATCRA694;",
        "PID": "22D8701",
        "Expression": "[B4:B5]",
        "MinValue": "",
        "MaxValue": "",
        "Period": "5000",
        "Type": "Default",
        "Send_to": ""
      }