meatpiHQ/wican-fw #121: No RX when TX published to MQTT

edlucky1, 2024-04-21:
I am trying to read the odometer for a Hyundai Sonata. I was able to read with Car Scanner and captured these in MQTT Explorer:

```
{"bus":"0","type":"tx","ts":53619,"frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,166,170,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":53625,"frame":[{"id":2024,"dlc":8,"rtr":false,"extd":false,"data":[6,65,166,0,20,164,202,170]}]}
```

I published the TX line to the wican/xxxxxxxxxx/can/tx topic in MQTT but never see RX response.

Settings:

[image]

I also tried to get Engine Speed (it seemed more straight forward, not sure that my expression is correct for Odometer). Frames captured in MQTT:

```
{"bus":"0","type":"tx","ts":25061,"frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,12,170,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":25069,"frame":[{"id":2024,"dlc":8,"rtr":false,"extd":false,"data":[4,65,12,10,120,170,170,170]}]}
```

Same result, TX shows but no RX. I have read through a lot of the discussion and not sure where to look next.

meatpiHQ, 2024-04-23:
@edlucky1  Please change the index to 2 in both filters. The index refers to PID index, that's byte 2.

meatpiHQ, 2024-04-23:
This was not mentioned in the documentation. I have added it now.

https://github.com/meatpiHQ/wican-fw/tree/main?tab=readme-ov-file#7-can-to-json-interpreter---filtring

edlucky1, 2024-04-24:
That fixed it, thanks!

On to the next piece of the puzzle...

closed 2024-04-24 by edlucky1