meatpiHQ/wican-fw #123: Please help with BMW CAN bus mileage

alexdan-77, 2024-05-02:
Hello everyone!

Trying to ask by CAN bus this request (odometer mileage):
{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,166,170,170,170,170,170]}]}

but no reply at all.

but when ask with this request (get fuel level):
{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,47,170,170,170,170,170]}]}
i have reply with correct result.

Is there any correct PIDs to achieve my goal?

thanks.

meatpiHQ, 2024-05-02:
@alexdan-77 it's possible that your car doesn't support this PID. 

Have you tried Carscanner app? Are you able to see the Odometer value in the app?

alexdan-77, 2024-05-03:
> @alexdan-77 it's possible that your car doesn't support this PID.
> 
> Have you tried Carscanner app? Are you able to see the Odometer value in the app?

nope.
[image]

alexdan-77, 2024-05-03:
maybe somebody knows request that works?or is it impossible by can bus?

meatpiHQ, 2024-05-04:
@alexdan-77 Try other apps other than carscanner. I also suggest you try and look at some BMW forum, it's possible that someone has already figured it out.

alexdan-77, 2024-05-06:
tell me. if i use canhacker app will it show something useful?

alexdan-77, 2024-05-07:
i use savvycan scanner. of course my engine is on.

meatpiHQ, 2024-05-08:
@alexdan-77 some cars don't expose the internal CAN bus on the OBD connector. The 0x7E8 is the OBD response, but I don't know what is 0x130.

closed 2024-05-12 by alexdan-77