meatpiHQ/wican-fw #124: Odometer PID 166 Value

edlucky1, 2024-05-05:
I am able to get this response:

```{"bus":"0","type":"rx","ts":24909,"frame":[{"id":2024,"dlc":8,"rtr":false,"extd":false,"data":[6,65,166,0,20,187,199,170]}]}```

Wican settings:

[image]

This results in a value of 173925333. My car odometer (in miles converted to km, km is standard unit for this PID) is 135,086.

edlucky1, 2024-05-05:
After some testing start bit should be 24 in order to get correct value. Wican settings:

[image]

I included the conversion from km to miles. For some reason my car (Hyundai Sonata) appears to have an offset of 2 miles. I will monitor this to make sure the offset doesn't change over time.

meatpiHQ, 2024-05-05:
@edlucky1 looks good!

closed 2024-05-05 by edlucky1