meatpiHQ/wican-fw #128: IOBroker no rx is created and no response to MQTT tx for ID which works in CarScanner via WiFi

stwa2020, 2024-05-20:
Hi All,

I have a Skoda Enyaq and the MeatPi works with the CarScanner and I can see SoC and many other values.
I've tried to get any rx data by using the information created with the CarScanner when ELM323 MQTT is enabled, with debug enabled in IOBroker for the instance, in this case mqtt.3

When the CarScanner is running I have the following in the IObroker MQTT debug log for ID 416941553 **tx** followed by **rx** information.

`mqtt.3 2024-05-20 16:29:41.358	debug	Server received "wican/5432048dee7d/elm327" (string): "{\"bus\":\"0\",\"type\":\"rx\",\"ts\":9390,\"frame\":[{\"id\":417001733,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[34,51,48,48,48,48,48,63]}]}"`

`mqtt.3 2024-05-20 16:29:41.357	debug	Client [ESP32_8dEE7C] received publish package {"cmd":"publish","retain":false,"qos":0,"dup":false,"length":152,"topic":"wican/5432048dee7d/elm327","payload":{"type":"Buffer","data":[123,34,98,117,115,34,58,34,48,34,44,34,116,121,112,101,34,58,34,114,120,34,44,34,116,115,34,58,57,51,57,48,44,34,102,114,97,109,101,34,58,91,123,34,105,100,34,58,52,49,55,48,48,49,55,51,51,44,34,100,108,99,34,58,56,44,34,114,116,114,34,58,102,97,108,115,101,44,34,101,120,116,100,34,58,116,114,117,101,44,34,100,97,116,97,34,58,91,51,52,44,53,49,44,52,56,44,52,56,44,52,56,44,52,56,44,52,56,44,54,51,93,125,93,125]}}`

`mqtt.3 2024-05-20 16:29:41.356	debug	Server received "wican/5432048dee7d/elm327" (string): "{\"bus\":\"0\",\"type\":\"rx\",\"ts\":9385,\"frame\":[{\"id\":417001733,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[33,49,56,52,71,32,49,48]}]}"`

`mqtt.3 2024-05-20 16:29:41.355	debug	Client [ESP32_8dEE7C] received publish package {"cmd":"publish","retain":false,"qos":0,"dup":false,"length":152,"topic":"wican/5432048dee7d/elm327","payload":{"type":"Buffer","data":[123,34,98,117,115,34,58,34,48,34,44,34,116,121,112,101,34,58,34,114,120,34,44,34,116,115,34,58,57,51,56,53,44,34,102,114,97,109,101,34,58,91,123,34,105,100,34,58,52,49,55,48,48,49,55,51,51,44,34,100,108,99,34,58,56,44,34,114,116,114,34,58,102,97,108,115,101,44,34,101,120,116,100,34,58,116,114,117,101,44,34,100,97,116,97,34,58,91,51,51,44,52,57,44,53,54,44,53,50,44,55,49,44,51,50,44,52,57,44,52,56,93,125,93,125]}}`

`mqtt.3 2024-05-20 16:29:41.354	debug	Server received "wican/5432048dee7d/elm327" (string): "{\"bus\":\"0\",\"type\":\"rx\",\"ts\":9383,\"frame\":[{\"id\":417001729,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[3,127,9,34,170,170,170,170]}]}"`

`mqtt.3 2024-05-20 16:29:41.353	debug	Client [ESP32_8dEE7C] received publish package {"cmd":"publish","retain":false,"qos":0,"dup":false,"length":155,"topic":"wican/5432048dee7d/elm327","payload":{"type":"Buffer","data":[123,34,98,117,115,34,58,34,48,34,44,34,116,121,112,101,34,58,34,114,120,34,44,34,116,115,34,58,57,51,56,51,44,34,102,114,97,109,101,34,58,91,123,34,105,100,34,58,52,49,55,48,48,49,55,50,57,44,34,100,108,99,34,58,56,44,34,114,116,114,34,58,102,97,108,115,101,44,34,101,120,116,100,34,58,116,114,117,101,44,34,100,97,116,97,34,58,91,51,44,49,50,55,44,57,44,51,52,44,49,55,48,44,49,55,48,44,49,55,48,44,49,55,48,93,125,93,125]}}`

**tx**
`mqtt.3 2024-05-20 16:29:41.347	debug	Server received "wican/5432048dee7d/elm327" (string): "{\"bus\":\"0\",\"type\":\"tx\",\"ts\":9380,\"frame\":[{\"id\":416941553,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[48,0,10,170,170,170,170,170]}]}"`

I've tried to send the same ID request but there is no response at all, 

This is the **tx** request as JSON entered in **tx** topic:
**{"bus":"0","type":"tx","frame":[{"id":416941553,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}**

This is the IObroker debug information for this request:
`mqtt.3 2024-05-20 19:01:08.590	debug	Client [ESP32_8dEE7C] send to this client "wican/5432048dee7d/can/tx": {"bus":"0","type":"tx","frame":[{"id":416941553,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}`

`mqtt.3 2024-05-20 19:01:08.589	debug	onStateChange mqtt.3.wican.5432048dee7d.can.tx: {"val":"{\"bus\":\"0\",\"type\":\"tx\",\"frame\":[{\"id\":416941553,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[48,0,10,170,170,170,170,170]}]}","ack":false,"ts":1716224468585,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1716224468585}`

**tx**
`mqtt.3 2024-05-20 19:01:08.588	debug	stateChange mqtt.3.wican.5432048dee7d.can.tx: {"val":"{\"bus\":\"0\",\"type\":\"tx\",\"frame\":[{\"id\":416941553,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[48,0,10,170,170,170,170,170]}]}","ack":false,"ts":1716224468585,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1716224468585}
`

I do not use the red node, I send the request directly to the **MQTT tx** topic,

[image]

[image]

Is there something wrong in the way how the tx JSON is send ?
Is there a possibility to see the MQTT information and how it's received on the MeatPi, any debug function ?
Has someone IOBroker running and can see a **rx** when a valid **tx** request was sent ?

Thank you and best regards,

Stefan

stwa2020, 2024-05-21:
I got now a response, after I have disabled the MQTT elm327 log I got a response,
Tx {"bus":"0","type":"tx","frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,0,170,170,170,170,170]}]}
Rx {"bus":"0","type":"rx","ts":49086,"frame":[{"id":417001738,"dlc":8,"rtr":false,"extd":true,"data":[3,127,1,17,170,170,170,170]}]}

Now I have to find which command I have to send to get a response to calculate SoC and how many Km are on the speedometer

meatpiHQ, 2024-05-21:
@stwa2020 have a look at this issue it might help you
https://github.com/meatpiHQ/wican-fw/issues/19

stwa2020, 2024-05-21:
@meatpiHQ Thx I'll check #19 

To make it a little more interesting, I've resent the JSON string that worked yesterday, but now,
Tx {"bus":"0","type":"tx","frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,0,170,170,170,170,170]}]}
Rx **no response**.

I've opened and connected the CarScanner and received a Rx response via MQTT, now I've disconnected the CarScanner and sent the JSON String again now with a response, in both cases the WiCan status was online.

Tx {"bus":"0","type":"tx","frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,0,170,170,170,170,170]}]}
Rx {"bus":"0","type":"rx","ts":22233,"frame":[{"id":417001738,"dlc":8,"rtr":false,"extd":true,"data":[3,127,1,17,170,170,170,170]}]}

I waited more than 15 minutes and sent the JSON String via MQTT again, again no Rx.
I've started the CarScanner and it can connect to the ELM but not to the ECU, disconnected and reconnected, now ELM and ECU are connected.
I was standing in front of my car and I can hear a few quiet clicks (not the loud click) when CarScanner connected to the ECU, I guess I need a MQTT wake up JSON string as well.