meatpiHQ/wican-fw #134: Retained MQTT Data

HaareistNase, 2024-05-30:
Dear Sirs,
Is there a posibility to make the MQTT Data Wican sends „retained“.

I use Home Assistant to Display the MQTT Data from Wican, and every time I restart Home Assistant while wican is not online (because the car is not home or wican is in sleep mode) I get an unknown Sensor Status in Home Assistant.

HaareistNase, 2024-05-30:
Ah. I Export the values to MQTT again with retained flag, and than they do stay. No Need to change. Sorry for bother:-)

justynbell, 2024-07-06:
@HaareistNase could you go into details about what you mean?  Is this a setting in the HA MQTT sensor configuration?  Is this a configuration in the WiCAN settings somewhere?

HaareistNase, 2024-07-06:
>Is this a setting in the HA MQTT sensor configuration? Is this a configuration in the WiCAN settings somewhere?

Hey. Im not a programmer or homeassistant-nerd. So maybe Im wrong. Its just my experience.

So you can't change settings about this in wican. There is no option. So all you can do is handle it in homeassistant.

I create a mqtt-sensor to "fetch" the SOC from my e-Niro in configuration.yaml

```
mqtt:
  sensor:
    - name: "E-UP SOC MQTT"
      unique_id: "sensor.e_up_soc_mqtt"
      state_topic: "wican/5432048fef81/can/rx"
      value_template: "{{ value_json.BMS_SOC|float(0)| round(1) }}"
      icon: mdi:battery
      unit_of_measurement: "%"
```

but I can't make the values from this sensor "permanent".
Every time I reset homeassistant (and I reset it quite regularly because of the many updates this software get, and the work I put into it) it will "forget" the last value and I get a "unknown" in my display. I do not know how to fix this with a setting or parameter.

So I do "Export" this value again (I know, its double-work for homeassistant, but I don't care".

This is the automatisation for export

```
service: mqtt.publish
data:
  qos: 0
  retain: true
  topic: export/soc
  payload_template: "{{ states('sensor.e_up_soc_mqtt') }}"
```

Here I can set "retain: true" so homeassistant "saves" the value in his database, and it will be preserved over a system-restart and I do not get a "unknown" in my display.

Hope this helps. And maybe you know a better way :-)

atze09, 2024-08-14:
I would also like to have retained values,@meatPi could you please implement this setting in your code, this would be perfect and would improve the usability.
Thank you

meatpiHQ, 2024-09-22:
@atze09 already in the main branch will be available in the new release.

atze09, 2025-08-22:
> > [@atze09](https://github.com/atze09) already in the main branch will be available in the new release.
> 
> [@meatpiHQ](https://github.com/meatpiHQ) yesterday I have updated to latest version. But Matt messages are still not retained? Isn't it already implemented?
> 
> <img alt="Image" width="1080" height="2000" src="https://private-user-images.githubusercontent.com/6229866/479922258-c173a47f-f388-4568-a421-7cebf3bb8eda.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU4NTI4NzksIm5iZiI6MTc1NTg1MjU3OSwicGF0aCI6Ii82MjI5ODY2LzQ3OTkyMjI1OC1jMTczYTQ3Zi1mMzg4LTQ1NjgtYTQyMS03Y2ViZjNiYjhlZGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MjJUMDg0OTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTFlZmEzYzg1YjliNWY2Y2VlZjdlZDQ2YWQ2NDVmMTgxOThiNzRlZDRlNDYwYThkZjYyYTQzOTZmYjhiMjdlYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.HCfosjVESycDw6ai43bPXYnqT3JZzB4-Ell1UotADg0">

Sorry i have mixed up the things, MQTT topic is now retained, but the picture shows the entity of the wican which is not visibale when car and wican is turned off..