meatpiHQ/wican-fw #138: wican-usb - bluetooth not broadcasting

pantsnotnecessary, 2024-06-13:
wican-usb running 2.98 in AP. cannot see bluetooth broadcast from phone/tablet/computer despite resetting and reconfiguring multiple times. can pull data from wifi to realdash on w11 computer.

meatpiHQ, 2024-06-14:
@pantsnotnecessary did you disconnect from the device AP after enabling the BLE?

As long as you have any device connected to the device AP BLE won't show. And when you connect to BLE the device wifi AP will be disabled.

pantsnotnecessary, 2024-06-27:
i did disconnect everything wifi from it first. i’m going to disconnect the can connection tonight and see if it sparks it to broadcast bt. will report back later.

meatpiHQ, 2024-06-28:
@pantsnotnecessary I suggest you disconnect from the device access point and power cycle. Another thing to note, WiCAN supports BLE 4.2

pantsnotnecessary, 2024-08-08:
Zero luck still. i've updated to 3.11 beta as well. is it possible my bluetooth isn't functional? i've tried all sorts of combinations including reconfiguring and trying different devices.

meatpiHQ, 2024-08-11:
@pantsnotnecessary please post a screenshot of your device configuration webpage.

meatpiHQ, 2024-09-02:
@pantsnotnecessary looks OK. Can you try and install "nrf connect" from the app store, and then try to scan.

Just to confirm, if you connect to device AP and view the configuration page bluetooth will shutdown. Make sure you disconnect from the device wifi connection and unplug then plug the device back in.

pantsnotnecessary, 2024-09-10:
nrf seems to be the missing link. thank you for your help.

meatpiHQ, 2024-09-11:
To be honest, I don’t fully understand the issue. Which phone are you testing with, and what’s the Android version?