meatpiHQ/wican-fw #142: Can't connect to ECU, Zeekr 001

harkoprinssen, 2024-06-19:
Driving a brand new Zeekr 001.
When using a Vlink elm327 ble dongle it connects and I am able to read data from the car using Car Scanner.
When connecting the Wican-obd it can't seem to connect to the ECU.

I have turned off mqtt, tried every CAN bitrate and ended up setting it to AUTO,
Car Scanner can connect to the wican, but it can't seem to connect to the ECU.

Anyway to figure out what the vlink does different or any tips to get this to work?

In Car Scanner the profile is also used for the Volvo EX30, Smart 1, Zeekr X and Zeekr 001l

harkoprinssen, 2024-06-19:
Log info found from Car Scanner

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>0100
BUS INIT: ...ERROR

>03
BUS INIT: ...ERROR

>07
BUS INIT: ...ERROR

>0A
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>
InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>0100
BUS INIT: ...ERROR

>03
BUS INIT: ...ERROR

>07
BUS INIT: ...ERROR

>0A
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>
InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>0100
BUS INIT: ...ERROR

>03
BUS INIT: ...ERROR

>07
BUS INIT: ...ERROR

>0A
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>0100
BUS INIT: ...ERROR

>0100
BUS INIT: ...ERROR

>ATDPN
0

>
Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped

harkoprinssen, 2024-06-19:
[log.txt](https://github.com/user-attachments/files/15905225/log.txt)

This logfile shows succes with the Vlink BLE and the errors with the Wican.

If I connect the Wican to my 2007 Prius the orange light immediately starts blinking. on my Zeekr it stays off

meatpiHQ, 2024-06-20:
@harkoprinssen can you please post a screenshot of your device configuration webpage?

harkoprinssen, 2024-06-20:
[image]
[image]
[image]

Behaviour is the same in BLE mode.
When connecting to the prius the orange led immediately starts blinking, on the Zeekr it never lights up.

meatpiHQ, 2024-06-21:
@harkoprinssen please disable CAN auto bitrate. The ELM327 protocol will try different bitrates until it finds the correct one.

harkoprinssen, 2024-06-21:
Tried that before already, but tried again just now. After setting can bitrate I took out the the wican and reconnected it to be sure it restarted properly.
Behaviour is identical. 

Log from car scanner below

===== LOG STARTED: 21.06.2024 05:59:14=====
=====Version=1.108.0 Build=1.108.0 P=False NoDelayELM327Init=False
====Connection profile=Zeekr EX30 electric / Smart #1 (EV) / Smart #3 (EV) / Zeekr 001 / Zeekr X / Zeekr Mix
[Connection type: WIFI]
[WiFi name: <null>]

[Connecting to 192.168.1.92:3333]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
client.Connected
got stream from client

[Connected to 192.168.1.92:3333]
ATZ
ATE0

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]

[ELELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>0100
BUS INIT: ...ERROR

>03
BUS INIT: ...ERROR
[... 2039 lines cut ...]
VWTP20OpenSessionForDTCOperations=True
WiFiPort=3333
WiFiServer=192.168.1.92
ZeroConsumptionPIDId=-1
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=False
FPGC=6
CC=0

OS version=17.5
Platform=iOS
Idiom=Phone
DeviceModel=iPhone13,3
DeviceManufacturer=Apple

OrigCulture=en-US
OrigUICulture=en-US
Version=1.108.0/1.108.0

PrVersion=1.108.4

[***/// END OF CONTACT DEVELOPER REPORT]

harkoprinssen, 2024-06-21:
[image]

meatpiHQ, 2024-06-21:
@harkoprinssen please try to set the ECU protocol in carscanner to: 6) 11bit 500Kbaud

If that doesn't work try the other protocols 7, 8 or 9.

harkoprinssen, 2024-06-21:
Will look into that, but as I don't plan on using car scanner (MQTT to Home assistant) what will that tell me?
Also when connected to the prius the orange light starts blinking immediately, on the zeekr nothing happens. I am leaning more towards an issue in the firmware not being able to connect at all.

Will update once I figured out how to change the protocol etc..

meatpiHQ, 2024-06-21:
This will tell us what is the CAN bitrate. 

The Prius probably broadcasts on the CAN, but the Zeekr does not. Once you figure out the CAN bitrate on the Zeekr you'll be able to make an OBD request.

harkoprinssen, 2024-06-21:
I can find an option to auto-detect prorocol, this doesn't work. 
How can I force a protocol in car scanner? 

Also would trying all available bitrates (until 1 works) the same information?

harkoprinssen, 2024-06-21:
[image]
Here you go, setting the wican to 500kbit doesn't work. Tried turning options related to the 15765-4 protocol ON or OFF, but no luck. 
The ble connects instantanious.

harkoprinssen, 2024-06-21:
> > How can I force a protocol in car  scanner?
> 
> Have a look at this comment. [#17 (comment)](https://github.com/meatpiHQ/wican-fw/issues/17#issuecomment-1454783099)
> 
I used the setting 'try all other protocols' before already, no luck

meatpiHQ, 2024-06-21:
Ok, please set to 11 but 500kbits in car scanner, and send me the log.

harkoprinssen, 2024-06-21:
[image]

=====Version=1.108.0 Build=1.108.0 P=False NoDelayELM327Init=False

====Connection profile=Zeekr EX30 electric / Smart #1 (EV) / Smart #3 (EV) / Zeekr 001 / Zeekr X / Zeekr Mix

[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 192.168.1.92:3333]

client = new TcpClient();

BEFORE await client.ConnectAsync(server, port);

client.Connected

got stream from client

[Connected to 192.168.1.92:3333]

ATZ
ATE0

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)

ATE0
OK

>ATE0
OK

>STI
?

>

[STSupported=False(ByResponse)]VTI
?

>

[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK

>ATD0
OK

>ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
[... 1097 lines cut ...]
P=False

FPGC=7

CC=0

OS version=17.5

Platform=iOS

Idiom=Phone

DeviceModel=iPhone13,3

DeviceManufacturer=Apple

OrigCulture=en-US

OrigUICulture=en-US

Version=1.108.0/1.108.0

PrVersion=1.108.4

[***/// END OF CONTACT DEVELOPER REPORT]

harkoprinssen, 2024-06-21:
[blelog.txt](https://github.com/user-attachments/files/15927632/blelog.txt)
Attached a part of the log where car scanner connects using the ble reader

harkoprinssen, 2024-06-21:
2.98, received last wednesday

Kept fiddling around and now this.. Not sure what I did, but it's connecting now.

[image]
[image]

It is however very unstable in reading data as it keeps disconnecting and connecting to the ECU. In fact the screenshot was the only time I was able to read sensor data. can't get it to read any data anymore now (though still says connected).
When selecting the Zeekr profile in car scanner it warns me updates only come every 30 seconds, could it be related to some timeout in the firmware?

meatpiHQ, 2024-06-21:
Did you set it to AP mode?

I think this is good enough, if you only plan to use it with home assistant. You should be able to make PID request over MQTT. Have a look at this link

https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

harkoprinssen, 2024-06-21:
when selecting the SoC I see this in the ELM327 log which 
{"bus":"0","type":"tx","ts":3077,"frame":[{"id":500176437,"dlc":8,"rtr":false,"extd":true,"data":[3,34,72,1,170,170,170,170]}]} 
Looking at the link there is no mention on what to fill out for index (I assumed 0)
Not sure how to consider PID, will that always be 3?

[image]

meatpiHQ, 2024-06-21:
What is the RX frame?

meatpiHQ, 2024-06-21:
Ok, the filter should be:
PID: 72
Index: 2

meatpiHQ, 2024-06-21:
What was the SoC for this reading?

harkoprinssen, 2024-06-21:
[image]

[image]

It is reading something, but the value should only go up so besides a wrong calculation I am thinking it probably reads the wrong bytes

Also I need to keep car scanner open so I am guessing I have the wrong tx frame in my automation

meatpiHQ, 2024-06-21:
The expression is incorrect. I think it should be: B4/2

harkoprinssen, 2024-06-21:
must.try.to.understand :)

here's some more lines, it seems the 6th byte is going up.
Does that mean I have to use my math skills to figure out the equation to get from 179 to (at that moment about 56%?

{"bus":"0","type":"rx","ts":9765,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9665,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9596,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9535,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9456,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9365,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9296,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9196,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9096,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":9036,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8965,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8885,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8816,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8755,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8696,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8636,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8496,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,179,0,0]}]}
{"bus":"0","type":"rx","ts":8256,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":8085,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7985,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7916,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7815,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7761,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7656,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7555,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7485,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7395,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7296,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":7056,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6985,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6896,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6835,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6736,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6636,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6561,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6485,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6385,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6275,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6161,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6081,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}
{"bus":"0","type":"rx","ts":6015,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,108,176,0,0]}]}

harkoprinssen, 2024-06-21:
Stopped charging, SoC @ 56.82%

{"bus":"0","type":"rx","ts":41935,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,110,248,0,0]}]}
{"bus":"0","type":"rx","ts":41845,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,110,248,0,0]}]}
{"bus":"0","type":"rx","ts":41745,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,110,248,0,0]}]}

harkoprinssen, 2024-06-21:
It seems I need byte 4 and 5 as those seem to go up when the SoC raises

meatpiHQ, 2024-06-21:
> Stopped charging, SoC @ 56.82%
>
> {"bus":"0","type":"rx","ts":41935,"frame":[{"id":516337280,"dlc":8,"rtr":false,"extd":true,"data":[5,98,72,1,110,248,0,0]}]} [...]

B4: 110
B5: 248

The SoC is some combination of these 2 bytes. But most like it's something likely (B4/2)+B5/100

Or something like that. Also note that whatever carscanner display and what the car is displaying might be different. The best way is to get as many data points as possible and put those math skills to use :)

harkoprinssen, 2024-06-21:
whahaha, thanks so much for helping me out. Will get datapoints and try to figure this out.

harkoprinssen, 2024-06-21:
> Ok, the filter should be: PID: 72 Index: 2

Last question. how did you get to these two numbers

meatpiHQ, 2024-06-21:
The PID is always the send byte, that's B2. So the index is 2. In some special cases were you have multiple frame response you might want to change the index and look a different byte.

In this filter it's looking for a frame with B2=72

closed 2024-06-21 by harkoprinssen