meatpiHQ/wican-fw #144: ELM327 unable to connect to car

stibbzy, 2024-06-21:
I just received my WiCan (USB) and having trouble on my BMW i3. When I use  Carscanner with a BT OBD2 adaptor it connects just fine and I can get sensor data from the car. 
However, when I connect Carscanner to the WiCan, it says “succesfully connected to ELM327, but ELM327 wasn't able to establish connection with your car.”
The green LED is on and the orange LED is flashing. 
I’ve tried turning MQTT on and off, and disabled battery alerts. Different bitrates don’t work either. The Carscanner log has a lot of “BUS INIT: …ERROR” messages.
I've attached screenshots from Carscanner when it is working via the BT adaptor and when it isn't via WiCan. I've attached the log as well.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[Carscanner log.txt](https://github.com/user-attachments/files/15921317/Carscanner.log.txt)

harkoprinssen, 2024-06-21:
Last screenshot says it's connected to the ECU. Still not able to read data?

stibbzy, 2024-06-21:
> Last screenshot says it's connected to the ECU. Still not able to read data?

No, that is a screenshot illustrating how the Bluetooth OBD2 adaptor connects just fine to the ECU. Carscanner cannot connect to the ECU when using WiCAN

harkoprinssen, 2024-06-21:
Ah, now I see.
Behaviour seems like the issue I am facing, however I do not get the orange light blinking connected to my car

stibbzy, 2024-06-21:
> @stibbzy do you have both adapters connected to the same bus at the same time?

No, one at a time. I turn off the car in between.

meatpiHQ, 2024-06-21:
@stibbzy Try setting it to AP mode, disable mqtt logging and MQTT. Submit the settings and do not use the Monitor tab. The. Then try to connect.

meatpiHQ, 2024-06-21:
@stibbzy looking at the logs... seems that there are some elm327 commands that are not supported by WiCAN. I will try to add them and upload new firmware version for you to try.

harkoprinssen, 2024-06-21:
Could that possibly also help me out? https://github.com/meatpiHQ/wican-fw/issues/142

> @stibbzy looking at the logs... seems that there are some elm327 commands that are not supported by WiCAN. I will try to add them and upload new firmware version for you to try.

stibbzy, 2024-06-21:
@meatpiHQ  Thanks so much!
I’ll also try your other suggestion in the morning (Sydney time)

meatpiHQ, 2024-06-21:
@harkoprinssen I'm looking at your issue now.

stibbzy, 2024-06-21:
@meatpiHQ I just tried setting to AP mode, disabled MQTT and MQTT logging, and didn't use the Monitor tab. Connected to the WiCAN's wifi and ran Carscanner, same result - can't connect to ECU.
I also set Carscanner to run through 17 different CAN protocols to try to detect the right one (different bit count, baud rate etc) and none of those worked.

stibbzy, 2024-06-22:
@meatpiHQ I just spent another hour in the car trying different things but nothing worked. For what it's worth, Carscanner has a toggle to detect fake ELM327 clones. When I toggle that on, it detects the WiCAN as fake... not sure if that means anything or not. 
Also, under Carscanner advance settings, it can display AT commands that aren't supported by the adaptor. When I use the BT adapter, that field is blank, but with the WiCAN, it displays all these.

[image]
[image]

meatpiHQ, 2024-06-22:
@stibbzy WiCAN only emulates an ELM327 device and is not intended for diagnostics. It allows for extracting and logging data from the bus for reverse engineering and integration with Home Assistant. While this has worked with most cars, some vehicles require specific commands to function properly.

One approach is to log data from your other BLE adapter and use this information to configure WiCAN.

stibbzy, 2024-06-22:
@meatpiHQ Yep, all I want to be able to do is to pull the battery SOC into Home Assistant. Would an updated firmware likely start giving responses in CarScanner?

meatpiHQ, 2024-06-22:
> Would an updated firmware likely start giving responses in CarScanner?

Yes, that's the plan. But it might be faster if you log the SoC data using the other EM327 adapter, by clicking on the sensor on carscanner. All we need is the PID request and response.

stibbzy, 2024-06-22:
Ok, here is the log once I clicked the sensor and let it go for about 10 seconds...
[Uploading BMW i3 Batteru SOC.txt…]()

meatpiHQ, 2024-06-22:
22DDBC is the OBD request.

These frames are the PID request and response. We need to translate these into CAN frames and send it over MQTT. I'll have a look at it later tonight.

stibbzy, 2024-06-22:
Thanks so much!

meatpiHQ, 2024-06-22:
I think the PID request should be:

> {"payload":{"bus":"0","type":"tx","ts":0,"frame":[{"id":1777,"dlc":8,"rtr":false,"extd":true,"data":[3,34,221,188,170,170,170,170]}]}}

After you send this frame you should expect a rx frame with the first data byte is 16, once you receive that, you have to send a response frame:

> {"payload":{"bus":"0","type":"tx","ts":0,"frame":[{"id":1777,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

This issue is a good reference. 

https://github.com/meatpiHQ/wican-fw/issues/97#issuecomment-2089644589

stibbzy, 2024-06-22:
I'm still not understanding... Do I publish that to the topic directly under wican (wican/ecda3ba6d52d)? I tried, but I didn't get any rx response. That's what I don't understand, I have never seen a rx response in MQTT Explorer:
[image]

harkoprinssen, 2024-06-22:
I think you should be publishing in the ../tx topic. 

Mine goes to wican/f412fa377d01/can/tx

stibbzy, 2024-06-22:
@harkoprinssen Ahh, thanks very much for the suggestion. Unfortunately it didn't give a response on either wican/ecda3ba6d52d/can/tx or wican/ecda3ba6d52d/elm327/tx

meatpiHQ, 2024-06-22:
Yes to the tx topic. Make sure you have " MQTT ELM327 logging" disabled.

stibbzy, 2024-06-22:
@meatpiHQ OMG, I finally have a response after disabling logging!
{"bus":"0","type":"rx","ts":37927,"frame":[{"id":304,"dlc":5,"rtr":false,"extd":false,"data":[5,240,252,255,255,0,0,0]}]}

meatpiHQ, 2024-06-22:
That's great, but it doesn't look like the frame we're after.

Btw you can have the BLE and WiCAN adapter connected at the same time if you want, you can see Rx frames over MQTT.

stibbzy, 2024-06-22:
I'm using the wiCAN adaptor with a spliced OBD2 cable at the moment, so the BLE can't be plugged in as well unfortunately. Is there only supposed to be one frame with the 16 byte? If so, I would have missed it with all the messages coming in

meatpiHQ, 2024-06-23:
> If so, I would have missed it with all the messages coming in. All the messages are identical, apart from the ts value of course.

what are the messages coming in? 

Note: You should only add the filter after we figured out the message structure, if you added a filter you won't see a more messages coming only the filtered ID.

stibbzy, 2024-06-23:
@meatpiHQ Without that filter, there are ~1000 messages, and they all say {"bus":"0","type":"rx","ts":37927,"frame":[{"id":304,"dlc":5,"rtr":false,"extd":false,"data":[5,240,252,255,255,0,0,0]}]}

I was just trying the filter to try to see if there was my desired response buried in there. But I don't know how to determine the CAN ID number or the PID that the reply would come back on. Based on this BLE response from before, I just tried CAN ID 1543 (Hex 607) and PID 241 (Hex F1) but that didn't work either.

meatpiHQ, 2024-06-23:
Since your car broadcasts over the CAN bus we can try a different approach.

Try adding this filter: 

ID: 1074
Name: SoC
PID: -1
index: 0
Start Bit: 0
Bit Length: 1
Expression:  B4/2
Cycle ms: 1000

I found a DBC file for I3 not sure if its compatible with your car model, but you should give it a go.

https://github.com/damienmaguire/BMW-i3-CCS/blob/main/i3_LIM_dbc1.dbc

stibbzy, 2024-06-23:
@meatpiHQ Thanks for finding the DBC file....  it would very likely be compatible, but I have no idea what a DBC file is or how to use it.

I tried your filter, then published the tx json, but there was nothing received. Are my screenshots correct?

[image]
[image]

meatpiHQ, 2024-06-23:
Since your car is always broadcasting then there is no need to publish any tx frame. Did you store and submit ? If it works you should see an mqtt message like : {"SoC": 51}

The DBC file is a database file that would tell you have to interpret CAN frames:

The CAN frame id is 1074. The Display_SoC is what ever your car display is showing, 32|8@1+ means that the SoC value is in Byte 4 
[image]

stibbzy, 2024-06-23:
@meatpiHQ Yes, I stored and submitted, and after refreshing browser, the filter was there.
The rx topic is not receiving any messages. I guess my question was, with no filter, why is every received response identical?

meatpiHQ, 2024-06-23:
You mean that the car is always broadcasting the same frame? is the engine ON?

stibbzy, 2024-06-23:
@meatpiHQ There is no engine as such (electric), but it is powered on, in drive, with my foot on the brake. Without a filter, every single frame (1000s per second) is:  {"bus":"0","type":"rx","ts":37927,"frame":[{"id":304,"dlc":5,"rtr":false,"extd":false,"data":[5,240,252,255,255,0,0,0]}]}

meatpiHQ, 2024-06-23:
Ok. Motor then :) 

This frame is the like a keep alive frame, not useful for what your trying to do. This tells us that the DBC file is not compatible.

stibbzy, 2024-06-23:
@meatpiHQ Ok, so I probably do need a TX frame, and then somewhere within the keep alive frames will be my response frame, once we figure out the frame structure?

Is there any way to filter out the keep alive messages lol?

meatpiHQ, 2024-06-23:
> so I probably do need a TX frame, and then somewhere within the keep alive frames will be my response frame, once we figure out the frame structure?

Yes you need to send the TX frame.

Try this firmware, it will publish the CAN frame by ID. So each frame with different CAN ID will go to a different topic.

wican/ecda3ba6d52d/can/rx/xxxxx

(Extract the bin file from the zip and upload it to the device)
[wican-fw_v299_hv300.zip](https://github.com/user-attachments/files/15942546/wican-fw_v299_hv300.zip)

stibbzy, 2024-06-23:
@meatpiHQ OK, I'll give it a go.

stibbzy, 2024-06-23:
@meatpiHQ 304 is the only frame, unless I push a tx frame...
[image]

meatpiHQ, 2024-06-23:
What are the other IDs if you push the tx frame?

No the carscanner is not working because sending an unsupported command, that changes the message format so it's failing to parse the response.

stibbzy, 2024-06-23:
@meatpiHQ I don't get any other IDs if I push a TX frame. In my screenshot, I show the response with a TX frame, it just has 304 rx, plus the transmitted frame

meatpiHQ, 2024-06-23:
What about if you send this frame?

{ bus : 0 , type : tx , frame :[{ id :2015, dlc :8, rtr :false, extd :false, data :[2,1,0,170,170,170,170,170]}]}

stibbzy, 2024-06-23:
@meatpiHQ No change:
[image]

meatpiHQ, 2024-06-23:
seems like its a problem with the mqtt explore, try this

{
   "bus":0,
   "type":"tx",
   "frame":[
      {
         "id":2015,
         "dlc":8,
         "rtr":false,
         "extd":false,
         "data":[
            2,
            1,
            0,
            170,
            170,
            170,
            170,
            170
         ]
      }
   ]
}

stibbzy, 2024-06-23:
@meatpiHQ No change, I've also tried issuing these from the Home Assistant MQTT add on with no luck.
[image]

meatpiHQ, 2024-06-23:
Something does not add up.. Just to confirm please double check the tx topic in the configuration page?

stibbzy, 2024-06-23:
It matches:
[image]

meatpiHQ, 2024-06-23:
try this frame again in this format

{
   "payload":{
      "bus":"0",
      "type":"tx",
      "ts":0,
      "frame":[
         {
            "id":1777,
            "dlc":8,
            "rtr":false,
            "extd":true,
            "data":[
               3,
               34,
               221,
               188,
               170,
               170,
               170,
               170
            ]
         }
      ]
   }
}

stibbzy, 2024-06-23:
@meatpiHQ Nope... are we sure these are real frames that are being received? 
[image]

meatpiHQ, 2024-06-23:
> { "payload":{ "bus":"0", "type":"tx", "ts":0, "frame":[ { "id":1777, "dlc":8, "rtr":false, "extd":true, "data":[ 3, 34, 221, 188, 170, 170, 170, 170 ] } ] } }

I accidently had the extended flag turned ON "extd":true , try the frame below. 

{
"payload":{
"bus":"0",
"type":"tx",
"ts":0,
"frame":[
{
"id":1777,
"dlc":8,
"rtr":false,
"extd":false,
"data":[
3,
34,
221,
188,
170,
170,
170,
170
]
}
]
}
}

> are we sure these are real frames that are being received?

Yes, I'm just translating them into CAN frames.

stibbzy, 2024-06-23:
@meatpiHQ That didn't have any effect, however, I had left the car just sitting with RX frames pouring in, and a few other IDs showed up, but at much less frequency than the 304 ID:
[image]

meatpiHQ, 2024-06-23:
can you show me what's in the frames? part of them are hidden in the screenshot

stibbzy, 2024-06-23:
@meatpiHQ This is for 1815:
{"bus":"0","type":"rx","ts":35227,"frame":[{"id":126,"dlc":15,"rtr":true,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":75126616,"dlc":10,"rtr":true,"extd":true,"data":[0,0,0,0,0,0,0,0]},{"id":160979277,"dlc":3,"rtr":false,"extd":true,"data":[99,6,33,0,0,0,0,0]},{"id":443501415,"dlc":2,"rtr":true,"extd":true,"data":[0,0,0,0,0,0,0,0]},{"id":1815,"dlc":1,"rtr":true,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}

For 443501415:
{"bus":"0","type":"rx","ts":37716,"frame":[{"id":286,"dlc":10,"rtr":true,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":607,"dlc":15,"rtr":true,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":161241413,"dlc":3,"rtr":true,"extd":true,"data":[0,0,0,0,0,0,0,0]},{"id":1441,"dlc":3,"rtr":false,"extd":false,"data":[37,205,16,0,0,0,0,0]},{"id":443501415,"dlc":2,"rtr":true,"extd":true,"data":[0,0,0,0,0,0,0,0]}]}

For 1967:
{"bus":"0","type":"rx","ts":37718,"frame":[{"id":475920656,"dlc":3,"rtr":true,"extd":true,"data":[0,0,0,0,0,0,0,0]},{"id":1967,"dlc":12,"rtr":true,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}

meatpiHQ, 2024-06-23:
I've never seen anything like this before... dlc= 15 ?? how is that even possible. DLC in a CAN frame is max 8.

stibbzy, 2024-06-23:
@meatpiHQ Oh dear lol.
All this was brought about because the 3g network in Australia was shut down this month, and now a group of us on Facebook are trying to find a solution to get at least the SOC out of the car.

meatpiHQ, 2024-06-23:
ah so the i3 has a 3G connection?  

I can send you the new WiCAN pro adapter, it supports all the possible OBD commands. And will let you connect to the LTE using an accessory. https://www.crowdsupply.com/meatpi-electronics/wican-pro

stibbzy, 2024-06-23:
@meatpiHQ  Oh really? Should that work out of the box like carscanner then if it supports all commands?

meatpiHQ, 2024-06-23:
yes absolutely, that's what it's supposed to do. It's WiCAN on steroids or more like an OBDLINK MX with an esp32 as an application processor. The firmware is not 100% yet but I will have something soon.

Send me an email with your shipping address: https://www.meatpi.com/contact

> It is working with OVMS... is there anything in the GitHub code for that which would help?

I saw the code it's massive, it will take me a while to find it.

stibbzy, 2024-06-23:
@meatpiHQ OK, will do!
This is the i3 portion of the OVMS anyway, for what it's worth:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/master/vehicle/OVMS.V3/components/vehicle_bmwi3

meatpiHQ, 2024-07-07:
Hey @stibbzy please try this on WiCAN Pro. Extract the firmware from the zip file and update the device. 

[wican-fw_obd_pro_v298-5-gc28f6be-dirty.zip](https://github.com/user-attachments/files/16119291/wican-fw_obd_pro_v298-5-gc28f6be-dirty.zip) 

First set the protocol to: AutoPID

[image]

Then add a new custom pid and fill in the parameters below, then press the **store** button then reboot the device. If it works you should see a mqtt message on topic "soc_topic" with the soc value.

**Initialisation**: ATAT1;ATCEA07;ATTAF1;ATFCSH6F1;ATSH6F1;ATFCSD07300000;ATFCSM1;ATCRA607;
**Name**:  SoC
**PID**: 22DDBC
**Expression**: [B6:B7]/10
**Period**: 2000
**Send_to**: soc_topic

[image]

stibbzy, 2024-07-08:
@meatpiHQ Ahh so close!  So after configuring as above, the soc_topic is returning 410.5... My battery SoC reported by Carscanner (when in elm327 mode) is 46.4%...
I tried with 'Expression' set at just 'V' but it returned 0.

[16 more replies]