meatpiHQ/wican-fw #145: MQTT filter deletes wrong filter

JabeBRD, 2024-06-21:
I have issue with my two WiCan OBD 
```
Firmware ver: | v2.98
Hardware ver: | v3.00_obd
```
If i have entered multiple MQTT JSON filters, when I want to delete one filter, it might delete right one or it might delete some other, usualy it seems to delete lowest one.
Example. If i have 3 filters, and i want to delete 2, 3 gets deleted.
Would it be possible feature to edit filters, instead of delete and rewrite??

harkoprinssen, 2024-06-22:
Had this happen too, even filters I deleted before coming back

JabeBRD, 2024-06-22:
[image]
I wanted to delete odo1, Battery_SoC got deleted

meatpiHQ, 2024-06-22:
@JabeBRD sounds like a bug in the HTML, but I'm unable to reproduce. Did you store the filters? what happens if you refresh the page?

JabeBRD, 2024-06-23:
Yes I `store` before `submit changes`. Sometimes if i refresh, new filters disappear but are still working, those will be visible if i `reboot` under `About`, using `submit changes` wont make differense . Im using Opera, same thing using Opera on Windows or Android.
I think this happens, when multiple new filters are added.

harkoprinssen, 2024-06-23:
I can confirm I see similar behaviour

meatpiHQ, 2024-06-23:
I tested with Opera and I did not find this issue. Can you give me steps to reproduce? or is it just random ?

If you do not press store and refresh the page the filter will disappear, that's normal.

JabeBRD, 2024-06-23:
Its not random.
[image]
Here i had these filters set up, i wanted to delete soc_raw, Battery_SoC got deleted, now soc_raw is first, press delete and it got deleted.
Now i rewrite Battery_SoC, add it and store refresh page. Now it is last in list, had typo and press delete, odo1 got deleted, tried again, raw_soc got deleted, tried again, odo2 got deleted, now only Battery_SoC is left and it can be deleted.
I press `Delete` then `Store` then `Submit changes` and now refresh page.