meatpiHQ/wican-fw #146: TX frames not coming through after using car scanner?

harkoprinssen, 2024-06-22:
Here's the situation.
I set up an automation in home assistant as mentioned to send a tx-frame every 5 seconds. Works like a charm.
However when I use car scanner at the same time then stop it, I no longer receive any data.
Than when I raise the frame id in the mqtt publish it starts working again.

It seems once newer/higher ids are published it no longer accepts the 'lower' id?

meatpiHQ, 2024-06-24:
@harkoprinssen What do you mean by "raise the frame id" ?

harkoprinssen, 2024-06-24:
Currently sending this

{"bus":"0","type":"tx","ts":37511,"frame":[{"id":500176437,"dlc":8,"rtr":false,"extd":true,"data":[3,34,72,1,170,170,170,170]}]}

I got this from using car scanner. 
car scanner seems to +1 the 37511 with each call.

If I stop car scanner and use the last tx message all works fine. if I than however start car scanner again and stop it I no longer receive anything on mqtt.
Than when I raise the number (of what seems to be some kind of id) it starts working again.

meatpiHQ, 2024-06-24:
The "ts":37511 is not related to car scanner, this is the time stamp. Actually you don't need to send it in the tx frame it is just ignored in the firmware. In the rx frame you can use it to measure the time between 2 frames, and it rollover every 60 seconds (60000ms).

harkoprinssen, 2024-06-24:
hmm, I'll try leaving it out and see how that goes than. Thanks!