meatpiHQ/wican-fw #147: Wifi Questions

wh1t3f1r3, 2024-06-23:
Hello,

I received my WiCan this week and have been struggling to get it setup as a Station.  Obviously I can connect to the device as an AP but the device appears to refuse to join the SSID that I'm trying to get it to connect to.  The SSID I'm trying to connect to uses WPA2-PSK with the following encryption types enabled:

- CCMP
- CCMP-256
- GCMP
- GCMP-256

I also have an outdoor AP mounted about 15ft from where I park my truck.  I'm just looking for confirmation that the WiCan will work with these Wifi settings before I start moving AP arounds.

meatpiHQ, 2024-06-24:
@wh1t3f1r3 you should be able to connect these encryption types. Is it possible to get any logs from the router?

The current firmware only WPA2-PSK is supported, If you want I can send you a firmware with WPA2/WPA3 PSK enabled ?

wh1t3f1r3, 2024-06-24:
Logs from the router(s) is possible however there aren't any yet.  This could be due to an issue with the initial handshake or a weak signal.  My plan for next steps are I have a small router that I use while traveling.  I'll set it up with the same SSID & encryption that my AP's are using and place it inside the cab of the truck.  If the WiCan is able to connect to it then that will narrow the issue down to a weak signal problem.  If it isn't able to connect to the travel router then I'll take you up on the firmware to see if that fixes the issue.