meatpiHQ/wican-fw #150: Reading out SoC of Hyundai Ioniq 2017 28KWh

eterpstra, 2024-06-27:
Hello,

I want to read out the SoC of my Ioniq 28 KWh.

- WICAN is installed to Wifi and Mqtt is connected.
-
The next thing is now... How can i extract the data from the WICAN? I have found the folllowing excel file with some specs:

https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/tree/master/Ioniq%20EV%20-%2028kWh

But i dont understand how to start. Where do i fill in the details and connect this to my home-assistant setup?

meatpiHQ, 2024-06-28:
@eterpstra first I suggest you try and use carscanner to read the data.

What you need to do is setup home assistant to send an mqtt message to WiCAN to request the SoC and setup a filter in WiCAN to calculate the SoC. You should read the link below, you can ask me if you think there's anything not clear.

https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

jornodberg, 2024-08-16:
@meatpiHQ : I tried following that guide, but what I cannot understand is at the bottom it shows some result. But where does one find that result?

I'll admit I'm still dumb to Home Assistant. But I do believe there should be added a sensor to the Home Assistant config, which listens for the expected reply from MQTT ? Is there something essential missing from this guide, or is it me who doesn't understand how this works?

meatpiHQ, 2024-08-17:
@jornodberg  yes adding a sensor is missing in the guide, you can find that in Home assistant documentation. 

Btw I'm trying to enable discovery in the next release. So it will automatically create the sensor for you. Please check for firmware update next week.

eterpstra, 2024-09-02:
@meatpiHQ  Hi... I just installed the latest firmware and enabled auto discovery for mqtt.

But the senors are not automaticully created. How can i check if this is done correct?

in the mqtt broker in HA, i can see that the client (192.168.1.72 = wican) is connected to the broker. So the MQTT connection is oke.

meatpiHQ, 2024-09-02:
@eterpstra what's your current firmware version?

And how are you trying to add the sensors?

eterpstra, 2024-09-02:
> @eterpstra what's your current firmware version?
> 
> And how are you trying to add the sensors?

I have the latest version of the firmware. V3.30

meatpiHQ, 2024-09-02:
Did you try to add the sensor?

Edit dashboard -> Add Card -> by entity