meatpiHQ/wican-fw #160: JSON filtering rule isn't saved

jmalm, 2024-07-18:
Hi, I just got my WiCAN OBD unit and I can't get JSON filtering to work.

I have a working MQTT connection set up. (I get the status message, and on my diesel car I get lots of CAN messages.)

I enter a rule, hit Submit Changes, wait for the unit to restart, refresh the page and the rule is gone and is not sent over MQTT. Any ideas what I might be doing wrong?

A rule that I'm trying to set up for my Kia e-Niro 2019:

2028, soc_bms, 33, 0, 0, 1, B2/2, 1000

jmalm, 2024-07-19:
Filling in some more info...

I connected the unit to my computer with a PuTTY session. During reboot I see the following interesting logs:

```
W (933) mqtt_load_filter: mqtt_load_filter start
E (938) mqtt_load_filter: error parsing canflt_json
```

Can I somehow inspect what is in the flash?

meatpiHQ, 2024-07-19:
@jmalm did you click on the store button? :)

jmalm, 2024-07-19:
Oooh... I hadn't noticed that button at all! Thanks!

Logs look better, and the setting is there after a reboot:

```
W (956) mqtt_load_filter: mqtt_load_filter start
I (961) mqtt_load_filter: Loaded CAN Filter 0: CAN ID=2028, PID=33, PIDIndex=0, Name=soc, Start Bit=0, Bit Length=1, Expression=B2/2, Cycle=1000
```

closed 2024-07-19 by jmalm