meatpiHQ/wican-fw #162: Not entering sleep mode, even voltage under limit

JabeBRD, 2024-07-20:
I have two Wican, second one is not going to sleep mode, even when voltage reads under limit.
It goes to sleep and wakes up instantly again, it does this for long time after it goes to sleep.
I had voltage limit at 12,9V and changed it to 13,1V but it didnt make much differense.
Car is in sleep mode as there is no RX messages returning. And voltage reads 12,6-12,8V.
```binary_sensor.e_up_status,off,2024-07-20T16:00:00.000Z
binary_sensor.e_up_status,on,2024-07-20T17:09:02.075Z
binary_sensor.e_up_status,off,2024-07-20T17:09:49.478Z
binary_sensor.e_up_status,on,2024-07-20T17:45:52.492Z
binary_sensor.e_up_status,off,2024-07-20T17:49:43.753Z
binary_sensor.e_up_status,on,2024-07-20T17:49:43.767Z
binary_sensor.e_up_status,off,2024-07-20T17:50:58.409Z
binary_sensor.e_up_status,on,2024-07-20T17:51:16.849Z
binary_sensor.e_up_status,off,2024-07-20T17:51:35.136Z
binary_sensor.e_up_status,on,2024-07-20T17:51:42.752Z
binary_sensor.e_up_status,off,2024-07-20T17:52:48.887Z
binary_sensor.e_up_status,on,2024-07-20T17:52:48.907Z
binary_sensor.e_up_status,off,2024-07-20T17:55:50.704Z
binary_sensor.e_up_status,on,2024-07-20T17:55:50.728Z
binary_sensor.e_up_status,off,2024-07-20T17:58:53.848Z
binary_sensor.e_up_status,on,2024-07-20T17:58:53.868Z
binary_sensor.e_up_status,off,2024-07-20T18:01:56.281Z
binary_sensor.e_up_status,on,2024-07-20T18:01:56.299Z
binary_sensor.e_up_status,off,2024-07-20T18:08:18.874Z
binary_sensor.e_up_status,on,2024-07-20T18:08:19.198Z
binary_sensor.e_up_status,off,2024-07-20T18:11:04.272Z
binary_sensor.e_up_status,on,2024-07-20T18:11:04.302Z
binary_sensor.e_up_status,off,2024-07-20T18:13:17.146Z
binary_sensor.e_up_status,on,2024-07-20T18:13:26.235Z
binary_sensor.e_up_status,off,2024-07-20T18:14:08.455Z
binary_sensor.e_up_status,on,2024-07-20T18:14:08.480Z
binary_sensor.e_up_status,off,2024-07-20T18:17:29.051Z
binary_sensor.e_up_status,on,2024-07-20T18:17:29.068Z
binary_sensor.e_up_status,off,2024-07-20T18:20:27.870Z
binary_sensor.e_up_status,on,2024-07-20T18:20:27.889Z
binary_sensor.e_up_status,off,2024-07-20T18:21:35.157Z
binary_sensor.e_up_status,on,2024-07-20T18:22:19.505Z
binary_sensor.e_up_status,off,2024-07-20T18:22:19.507Z
binary_sensor.e_up_status,on,2024-07-20T18:22:25.158Z
binary_sensor.e_up_status,off,2024-07-20T18:23:31.203Z
binary_sensor.e_up_status,on,2024-07-20T18:23:31.482Z
binary_sensor.e_up_status,off,2024-07-20T18:26:17.434Z
binary_sensor.e_up_status,on,2024-07-20T18:26:17.445Z
binary_sensor.e_up_status,off,2024-07-20T18:29:39.160Z
binary_sensor.e_up_status,on,2024-07-20T18:29:39.173Z
binary_sensor.e_up_status,off,2024-07-20T18:32:40.525Z
binary_sensor.e_up_status,on,2024-07-20T18:32:40.548Z
binary_sensor.e_up_status,off,2024-07-20T18:35:35.284Z
binary_sensor.e_up_status,on,2024-07-20T18:35:35.339Z
binary_sensor.e_up_status,off,2024-07-20T18:38:31.977Z
binary_sensor.e_up_status,on,2024-07-20T18:38:31.999Z
binary_sensor.e_up_status,off,2024-07-20T18:39:19.478Z
binary_sensor.e_up_status,on,2024-07-20T18:39:27.616Z
binary_sensor.e_up_status,off,2024-07-20T18:39:32.495Z
binary_sensor.e_up_status,on,2024-07-20T18:39:40.320Z
binary_sensor.e_up_status,off,2024-07-20T18:41:38.199Z
binary_sensor.e_up_status,on,2024-07-20T18:41:38.231Z
binary_sensor.e_up_status,off,2024-07-20T18:44:35.108Z
binary_sensor.e_up_status,on,2024-07-20T18:44:35.132Z
binary_sensor.e_up_status,off,2024-07-20T18:47:38.968Z
binary_sensor.e_up_status,on,2024-07-20T18:47:38.970Z
binary_sensor.e_up_status,off,2024-07-20T18:50:41.845Z
binary_sensor.e_up_status,on,2024-07-20T18:50:41.857Z
binary_sensor.e_up_status,off,2024-07-20T18:52:15.817Z
binary_sensor.e_up_status,on,2024-07-20T18:52:17.283Z
binary_sensor.e_up_status,off,2024-07-20T18:53:45.295Z
binary_sensor.e_up_status,on,2024-07-20T18:53:45.551Z
binary_sensor.e_up_status,off,2024-07-20T18:57:19.485Z```
See here log from HA.

meatpiHQ, 2024-07-21:
@JabeBRD can you please post a screenshot of you device configuration webpage?

JabeBRD, 2024-07-21:
[image]

meatpiHQ, 2024-08-01:
@JabeBRD I've just tested this it it works fine here. What happens if you increase the sleep voltage to 13.4V?

meatpiHQ, 2024-08-05:
@JabeBRD  Can you try to set the sleep voltage to 13.2V or 13.3V?

The value of sleep voltage does not matter mush as long as it sleeps when you turn OFF the ignition and wakes up when you turn it ON. 

I plan to make improvements on the sleep mode but that will take sometime.

JabeBRD, 2024-08-08:
Tried 13,2V but then it wont wake up every time.

meatpiHQ, 2024-08-11:
@JabeBRD what is your car make and model? This is an unusual issue. 

What is the voltage when the ignition is ON? and OFF? Disable sleep to read the voltage.

JabeBRD, 2024-08-11:
VW e-Up 2022
When car in driving condition "started" 14,4V, ignition off and waited few minutes 12,6V.
And this is read from Wican status page.

meatpiHQ, 2024-08-18:
@JabeBRD I couldn't reproduce this issue. Have you tried measuring for voltage spikes?

I think I will introduce another condition for wakeup, if voltage is higher then X voltage for a set period of time. Now it immediately wakes up once it detects high voltage.

JabeBRD, 2024-08-18:
Logged today, but it didnt have this issue present.
I will try again next day.

Meaby there could be separate wakeup voltage and sleep voltage, it might help.

JabeBRD, 2024-08-21:
Looking at my logs, device is now entering sleep mode as expected.
I have same voltage limit 12,9V, what i have changed is how device is installed, before it was connected directly to OBD port, facing down (hanging), now it has extension cord and it is facing up (standing).
I'll try to measure if it starts acting funny again.

JabeBRD, 2024-09-04:
I have updated to latest firmware 3.30 and im still seeing this behavior.
When i notice that it is going online-offline-online in 3min cycle, i have tried to measure battery voltage, but when i open driver door to open hood, car wakes up for short moment and battery voltage falls down, then it rises very slowly, my multimeter shows voltage of 12,89V and it stays offline.

I think this has to do with 12v battery SoC and temperature.
Adding separate `enable` and `disable` limit for wakeup behavior would take care of this problem, so that it could be adjusted if needed.
I think using same voltage limit for enable and disable without any delta is root cause for this problem.

I have now added sensor to monitor 12V battery voltage from control unit, lets see how it sees voltages when this problem occurs.