meatpiHQ/wican-fw #163: PID new AUTOMATE tab setup info in fw 3.11

amicomit, 2024-07-22:
Request for setup assistance regarding the new, as yet undocumented tab on fw 3.11,set WiCAN to request and publish the the PID automatically:

meatpiHQ, 2024-07-24:
@amicomit Have you used the filters with the TX and RX frame before ? If you have those for your car it will make things easier.

Please use car scanner to connect to your vehicle, and post the logs from the car scanner app here, so we can find the proper initialisation.

amicomit, 2024-07-24:
Hi, yes, I've already achieved an excellent level of success using the filters - if the Wican supported 2 bytes for the PID and the Delete filter glitch was fixed, I wouldn't have any problems at all currently, it works superbly (although at the moment, I have to use the 2nd byte of the PID, indexed to Byte 3).  Below is an example log: (here the PID is 32,118, but I can only filter the RX frame using byte 3, in this instance 118)

{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}

amicomit, 2024-07-24:
Hi yes, I'm having excellent results using the Wican filters, and actually, if it supported two byte PIDs and the filter Delete glitch was fixed, I wouldn't be having any problems at all, it works really well. Below is a sample RX and TX for Tyre Pressure 1 - here the PID hex is 2076 which is 8310 dec, and represented by bytes 2 and 3 (32,118) in the TX and RX - but I can only use byte 3 with index 3 in the Wican filter (bottom).  This works OK, but in the end I'm likely to run into a duplicate number for the PID

Type Pressure 1
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}

CAN ID    Name          PID   Index    Start Bit    Length     Expression
1881	TPMS1	    118      3           0               1	  (B5*0.01373)*14.5038

amicomit, 2024-07-24:
Hi yes, I'm having excellent results using the Wican filters, and actually, if it supported two byte PIDs and the filter Delete glitch was fixed, I wouldn't be having any problems at all, it works really well. Below is a sample RX and TX for Tyre Pressure 1 - here the PID hex is 2076 which is 8310 dec, and represented by bytes 2 and 3 (32,118) in the TX and RX - but I can only use byte 3 with index 3 in the Wican filter (bottom). This works OK, but in the end I'm likely to run into a duplicate number for the PID

Type Pressure 1
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}
[image]

meatpiHQ, 2024-07-25:
Here's what I think it should look like, please delete the filter. And set the protocol to "Auto pid"

[image]

**Initialisation:** ATSH751;
**Expression:** (B5*0.01373)*14.5038

amicomit, 2024-07-25:
Hi, this has now been tried with no result.  I deleted all the existing filters and rebooted before testing, and remembered to change the protocol, but no joy.  Out of interest, why is there the Initialisation field and also the Init field ?.  Would it be possible to make the change to the existing filter feature in the meantime to allow for the two byte PID ?  Many thanks for your ongoing support.

meatpiHQ, 2024-07-25:
Can you please post a screenshot of the autopid page?

Ignore the init field for now, it doesn't do anything.

amicomit, 2024-07-25:
[image]

meatpiHQ, 2024-07-25:
You don't see any mqtt messages coming in at topic: any_topic?

meatpiHQ, 2024-07-25:
Is this the correct value? If not what is it?

amicomit, 2024-07-25:
Yes, the value is correct, 37.8 psi

l2smith2, 2024-07-26:
i'm also trying to understand how this works for a byd atto3, I can see the info I want in car scanner but no idea how to filter it with wican

meatpiHQ, 2024-07-26:
@l2smith2 Please open a new issue so we don't highjack this one.

you can record the info you're seeing using carscanner. Once you do, share that file and I can help you set it up.

https://www.carscanner.info/records/

T1983M, 2025-08-05:
> Hi yes, I'm having excellent results using the Wican filters, and actually, if it supported two byte PIDs and the filter Delete glitch was fixed, I wouldn't be having any problems at all, it works really well. Below is a sample RX and TX for Tyre Pressure 1 - here the PID hex is 2076 which is 8310 dec, and represented by bytes 2 and 3 (32,118) in the TX and RX - but I can only use byte 3 with index 3 in the Wican filter (bottom). This works OK, but in the end I'm likely to run into a duplicate number for the PID
> 
> Tyre Pressure 1 {"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]} {"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}
> 
> [image]

Hey @amicomit Could you share all your PIDs you are using, and the Expressions. this wouldbe very helpfull. I assume you also have the PIDs for at least the other 3 Tire values.

Thanks a lot

amicomit, 2025-08-05:
Below are the PIDs and expressions I use - the first two line for each entry are the MQTT TX and RX for my display in Home Assistant - the third line is the actual data I enter up into the WiCan

BatterySoC
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,73,16,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":27567,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[5,98,73,16,18,21,0,0]}]}
2028	BatterySoC	73	2	0	1	((B4*266)+B5)/100

Odometer
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,221,1,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42612,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[6,98,221,1,0,107,21,0]}]}
2028	Odometer	1	3	0	1	((B5*256)+B6)*0.621371

CabinTemp
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,221,4,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":1076,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[4,98,221,4,59,0,0,0]}]}
2028	CabinTemp	4	3	0	1	((B4-32)*(5/9))+10

TPMS1 (Tyre Pressure)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,118,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42765,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,118,0,196,0,0]}]}
1881	TPMS1		118	3	0	1	(B5*0.01373)*14.5038

TPMS2 (Tyre Pressure)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,119,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":145,"frame":    [{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,119,0,194,0,0]}]}
1881	TPMS2		119	3	0	1	(B5*0.01373)*14.5038

TPMS3 (Tyre Pressure)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,120,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":19174,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,120,0,196,0,0]}]}
1881	TPMS3		120	3	0	1	(B5*0.01373)*14.5038

TPMS4 (Tyre Pressure)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,32,121,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":25544,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[5,98,32,121,0,198,0,0]}]}
1881	TPMS4		121	3	0	1	(B5*0.01373)*14.5038

TTMS1 (Tyre Temperature)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,10,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":5706,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[4,98,42,10,66,0,0,0]}]}
1881	TTMS1	10	3	0	1	B4-50

TTMS2 (Tyre Temperature)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,11,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":5706,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[4,98,42,11,66,0,0,0]}]}
1881	TTMS2	11	3	0	1	B4-50

TTMS3 (Tyre Temperature)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,12,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":5706,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[4,98,42,12,66,0,0,0]}]}
1881	TTMS3	12	3	0	1	B4-50

TTMS4 (Tyre Temperature)
{"bus":"0","type":"tx","ts":12173,"frame":[{"id":1873,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,13,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":5706,"frame":[{"id":1881,"dlc":8,"rtr":false,"extd":false,"data":[4,98,42,13,66,0,0,0]}]}
1881	TTMS4	13	3	0	1	B4-50

T1983M, 2025-08-05:
[image]
[image]

May i also ask, why are you using the MQTT Filter and not the Automate tab? What are the differences here?