meatpiHQ/wican-fw #164: BYD Atto 3 setup

l2smith2, 2024-07-26:
@l2smith2 Please open a new issue so we don't highjack this one.

you can record the info you're seeing using carscanner. Once you do, share that file and I can help you set it up.

https://www.carscanner.info/records/

_Originally posted by @meatpiHQ in https://github.com/meatpiHQ/wican-fw/issues/163#issuecomment-2252269629_

meatpiHQ, 2024-07-26:
@l2smith2 thanks. Can you please export the logs

[image]

l2smith2, 2024-07-26:
[log.txt](https://github.com/user-attachments/files/16390686/log.txt)

ah didnt know about that export log function

meatpiHQ, 2024-07-26:
@l2smith2 Here's what you need to set it up, 

**Initialisation:** ATS0;ATAT1;ATST96;ATSH7E7;ATFCSH7E7;ATFCSD300000;ATFCSM1;

SOC:
**Name:** is the json key for 
**PID:** 221FFC1
**Expression:** (B5*256)+B4
**Send_to:** MQTT topic where the SoC will be published 
**Period:** Cycle time in ms

Some more info:

The BYD PIDs are available this is a very useful post, although they are not using Auto pid request.

The PID request for SoC is 221FFC, notice there is an additional 1 at the end. This means that it is expecting a single frame response. 221FFC without the 1 will also work.

https://github.com/meatpiHQ/wican-fw/discussions/122

These are the parameters that I think is useful for the Initialisation.
[image]

l2smith2, 2024-07-27:
[image]

Ok so its now publishing soc which initially I thought was just 10% too high but it never seems to change while charging so i'm not sure what value it actually is reading.

I know nothing about OBD2 and PIDs so this is a trial by fire. I'm trying to find the right documents/guides to read but I feel I'm missing something.

Can you run me through how you get the PID for different values and the expression from the tx rx frames? and what those initializations are doing?

Thanks for the help with this.

meatpiHQ, 2024-07-27:
@l2smith2 something does not look right. Can you please post a screenshot of your Automate page?

l2smith2, 2024-07-27:
[image]
[image]

I tried with and without the 1 after 221FFC also autopid and elm327 protocol also mqtt log enable/disabled. Car scanner doesn't work on AutoPID protocol is that normal?

l2smith2, 2024-07-27:
{"bus":"0","type":"tx","ts":5192,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,5,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":44476,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[4,98,0,5,69,170,170,170]}]}

Ok so using car scanner and looking at HV Battery SOC which is 69% currently I get this returned. 
I found another page on your wiki with more info on this, I'm still a bit lost on how to convert these returned values into appropriate PID filters or autopid settings.

Looking at the BYD seal desciption he gets

{"bus":"0","type":"rx","ts":37872,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[7,98,31,254,44,26,152,58]}]}
{"bus":"0","type":"rx","ts":38043,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[4,98,0,5,57,170,170,170]}]}

and converts it to 
* CAN ID=2031 
* Name=SoC 
* PID=4 
* Index=0 
* Start Bit=32 
* Bit length=8 
* Expression=V 
* Cycle ms=1000
to get 57% as the value

meatpiHQ, 2024-07-28:
@l2smith2 please go back to my original configuration and change the 
Initialisation to
 ATAT1;ATST96;ATSH7E7;ATFCSH7E7;ATFCSD300000;ATFCSM1

I just removed the ATS0 command.

> Car scanner doesn't work on AutoPID protocol is that normal?

Yes that's normal. And keep the mqtt log disabled.

l2smith2, 2024-07-28:
[image]
[image]

Ok so that setup is reporting the correct soc % without the decimal, can that be solved by dividing the expression by 100? or should I do that in home assistant.

So how do I now get the other values reporting like cell temps voltages etc. 
12v battery voltage is a useful one to have

meatpiHQ, 2024-07-28:
What is the reported SoC?
 Try to change the expression to:
((B5*256)+B4)/100

> So how do I now get the other values reporting like cell temps voltages etc.
> 12v battery voltage is a useful one to have

You need to find the PID, checkout this post https://github.com/meatpiHQ/wican-fw/discussions/122

12v battery voltage will be supported in the next firmware release.

l2smith2, 2024-07-28:
((B5*256)+B4)/100 reports the correct state of charge now, currently soc = {"SoC":87.4,"raw":"07621FFC2422653A"}

I dont understand how to find the PID from [https://github.com/meatpiHQ/wican-fw/discussions/122](url) Complete Noob

meatpiHQ, 2024-07-28:
Great!

No worries, let me know which parameters you want and I can try to find it for you later.

l2smith2, 2024-07-28:
[image]
[image]

Doesn't seem to go to sleep anymore. Battery voltage is below the threshold. Have I missed something.

l2smith2, 2024-07-28:
If we could figure out these values to start, This is from the BYD seal guy but seems the same.
There's a few other values he didn't list I'll have to work out with car scanner.

lowest voltage cell (3,32V):
TX: {"bus":"0","type":"tx","ts":12342,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,43,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":12348,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[5,98,0,43,252,12,170,170]}]}

highest voltage cell (3,33V):
TX: {"bus":"0","type":"tx","ts":31858,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,45,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":31864,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[5,98,0,45,0,13,170,170]}]}

lowest temperature cell (22 °C)
TX: {"bus":"0","type":"tx","ts":35633,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,47,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":54490,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[4,98,0,47,62,170,170,170]}]}

highest temperature cell (23 °C)
TX: {"bus":"0","type":"tx","ts":45461,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,49,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":18397,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[4,98,0,49,63,170,170,170]}]}

Battery total Voltage (572 V)
TX: {"bus":"0","type":"tx","ts":13789,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,8,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":7048,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[5,98,0,8,60,2,170,170]}]}

Battery Average temperature (22 °C)
TX: {"bus":"0","type":"tx","ts":17603,"frame":[{"id":2023,"dlc":8,"rtr":false,"extd":false,"data":[3,34,0,50,170,170,170,170]}]}
RX: {"bus":"0","type":"rx","ts":8303,"frame":[{"id":2031,"dlc":8,"rtr":false,"extd":false,"data":[4,98,0,50,62,170,170,170]}]}

l2smith2, 2024-08-16:
Any Idea what these values may be?

Also the meatpi seems to be the highest wifi traffic user on my network. 
Over 6GB in 24hrs seems a bit outrageous for simple MQTT msgs

meatpiHQ, 2024-08-17:
@l2smith2 I'll add these to BYD car profile soon.

Does your car broadcast CAN frames all the time? If so please add any fake filter to stop it. I will add a "disable mqtt publish" to stop all these frames.

l2smith2, 2024-08-17:
I think it has been doing that. Never really goes to sleep anymore even with voltage set high

meatpiHQ, 2024-08-17:
Which firmware version are you on?

l2smith2, 2024-08-17:
3.11 beta I believe

meatpiHQ, 2024-08-17:
Ok so probably it's keeping the ECU on all the time. I'll fix it and send you a firmware to try.

l2smith2, 2024-08-17:
It doesn't report Rx and TX msgs under separate topics in MQTT explorer either

l2smith2, 2024-10-17:
@meatpiHQ great job on the new firmware and home assistant integration. can confirm I'm getting SoC on the traction battery and voltage for the auxiliary battery.
I'll monitor network traffic to see if it's still using a large amount of data.
Also if having the wican plugged in is draining the battery faster by waking up and charging the 12v auxiliary more regularly.

It would be great to get the additional information available like the BYD Seal as mentioned in [https://github.com/meatpiHQ/wican-fw/discussions/122](url) looks like my atto reports the same info.