meatpiHQ/wican-fw #165: WiCAN attached =  cant control climate on Hyundai Ioniq5

tung256, 2024-07-28:
when unit is attached to my Hyundai Ioniq 5 (2023 SEL AWD), i cannot control the car's climate at all.  the touchscreen on dashboard are greyed out. the physical black bar below the touchscreen is also blanked out. no way to turn the fans nor temp on.  
[image]

any idea why?
once i remove the WiCAN, i can control everything again.

here are my settings for the WiCAN module:
[image]

meatpiHQ, 2024-07-28:
@tung256 are you sending anything on MQTT to the CAN bus?

Try disabling mqtt and changing CAN mode to silent instead of normal. You cannot really use the adapter in silent mode, but this will tell us that the issue is caused by CAN. 

It's possible that the CAN bitrate is not 500k, I know other users are using it with Ioniq. Don't leave the adapter plugged in for long until we figure it out.

tung256, 2024-07-31:
ok. so i enable MQTT to get the CAN ID. once i get CAN ID, i should disable MQTT?
as of now, everything is enabled and i am still able to control Climate.  i guess once i disconnect the WiCAN, and reconnect (without making any changes on the WiCAN settings), the car works as normal.

meatpiHQ, 2024-08-05:
@tung256 what if you set CAN mode to "Normal" does the  climate control stops working? Keep the mqtt disabled and test with carscanner see if you can see values other than the 12V battery voltage. The 12V battery reading is not done through CAN.

First we need to figure out what is causing the  climate control to act up before we set it up.

tung256, 2024-08-05:
> @tung256 what if you set CAN mode to "Normal" does the climate control stops working? Keep the mqtt disabled and test with carscanner see if you can see values other than the 12V battery voltage. [...]

climate control is back to normal now. climate working with these settings:
[image]
[image]
[image]

meatpiHQ, 2024-08-06:
@tung256 OK That's great. Now update to beta version https://github.com/meatpiHQ/wican-fw/releases/tag/v3.11Beta

Then in the automate tab add these values and click store. Reference:  https://github.com/meatpiHQ/wican-fw/discussions/157#discussioncomment-10168102

initialisation: ATSH7E4;ATST96;
expression: B10/2
Pid: 2201019

[image]

tung256, 2024-08-07:
fw updated as you wanted
[image]
Automate tab done as you wanted. i rebooted WiCAN too
[image]
CAN settings as before
[image]
[image]
no SOC in my MQTT as seen here
[image]

meatpiHQ, 2024-08-07:
@tung256 I forgot to mention you should change the protocol to "autopid" instead of elm327

tung256, 2024-08-07:
@meatpiHQ im very happy to say it works once i change to autopid!
values came through every 2s:
[image]

now, is there a CSV file containing more Init, PID, and Expression?
i see you shared this file from [this post](https://github.com/meatpiHQ/wican-fw/discussions/157#discussioncomment-10168102):
[image]
but it makes absolutely no sense for my Ioniq5 as i dont see PID 2201019  in that csv file

meatpiHQ, 2024-08-07:
@tung256 seems like the Ioniq 5 has the same PIDs as the Kona. I haven't had a chance to investigate this yet. 

Note that, the PID is 220101 and the 9 at the end indicates it's expecting 9 frames.

tung256, 2024-08-07:
@meatpiHQ 
in our example of SOC, i see the equation you wrote is B10/2:
[image]
BUT in the CSV file, i see it is just E/2   so this is very confusing.

_can we please do another example?_
[image]
what should i put in for init, PID and expression?
your time and help is greatly appreciated!

tung256, 2024-08-07:
@meatpiHQ   just want to let you know SOC is now available in my Home Assistant once i add this code into my configuration.yml file
[image]

it parses the MQTT from SoC_topic to give me this:
[image]
your product is amazing!

now just waiting on your help to get the voltage reading from the 12v battery, which i believe is labeled as "Aux Batt Volts" from your csv file

meatpiHQ, 2024-08-08:
@tung256 glad to hear it's working for you.

Here is an example of how to interrupt the data based on the CSV file.

Below is the raw data from all the frames read by wican, the letters on top of the bytes, is how the CSV file represents the bytes, 

'a' is B5
'e' is B10 
'ad' is  B38

If you want read the battery voltage you should use the expression: B38*0.1

In the future you don't have to worry about doing this, the new update will include your car model and all the supported parameters in the PID response.

```
               a   b  c
10 3E 62 01 01 EF FB E7
   d  e  f  g  h   i  j
21 EF 84 20 40 01 72 00
   k  l  m   n  o  p  q
22 FF D0 1D 1A 19 17 18
    r  s  t  u  v  w  x  
23 19 18 18 18 00 3F C2
   y  z  aa ab ac ad af 
24 16 C1 0E 00 00 90 00
   ag 
25 00 F1 D5 00 00 ED DF
26 00 00 B4 D5 00 00 AD
27 12 00 8D 9C 5F 00 02
28 E7 00 00 00 00 07 07
```

tung256, 2024-08-09:
i still dont know how to read raw. will try again and again since im not smart.
but at least i got these 2 working and very happy thanks to you!
[image]

[image]

yes, i think future update to just select make, model, and year would alleviate a lot of confusion for amateurs like myself. thank you for all your help

closed 2024-08-09 by tung256

erpalma, 2024-08-10:
I'm trying to replicate this on a Ioniq 6, which has the very same PIDs. Indeed I can confirm that 220101 works by looking at the logs in car scanner, and trying using the elm327 protocol manually. The car responds and B10/2 seems correct if I manually parse the response frames. My problem is that the Automate feature does not seems to work correctly: i can see that WiCAN is sending the request since I can see the response coming back at the "rx" topic. However I can't see the parsed JSON topic (i.e. "SoC_topic in this case). Everything is configured as per this issue, I've triple-checked everything. How can I debug this issue?

meatpiHQ, 2024-08-10:
@tung256 @erpalma this might be of interest to you https://github.com/meatpiHQ/wican-fw/discussions/157#discussioncomment-10298587

I will add more parameters for the Ioniq later.

tung256, 2024-08-16:
im sorry it does not make any sense to me on how to decode.

  in your example, i see "a" is directly over EF.  how does "a" becomes B5?
 i see "ad" is over 90,  but it becomes B38?

is there a video guide that can explain? or what keywords should i search on YouTube? i searched for CAN bus but not of them explained how to read MQTT raw files...