meatpiHQ/wican-fw #176: Fiat Ducato communication

savikko, 2024-08-18:
Hi!

Has anyone been able to get any "significant" information from Fiat Ducato?

If I have understood correctly, the Fiat OBD is somehow non-standard.

Anyways, I am mostly interested about few very simple data points (including, but not limited to):
* Battery voltage (probably will get this in very near future directly via WiCAN)
* Engine running/not running boolean (or possibly the RPM)

meatpiHQ, 2024-08-18:
@savikko are you able to read these parameters using carscanner or other apps?

savikko, 2024-08-18:
> @savikko are you able to read these parameters using carscanner or other apps?

Nope. If I remember correctly (its been a while since I digged into this deeper) there was some very special Fiat Android App which read the values.

Now actually going through data which I get from CAN and tried to understand the data format.

Example data:
```
{
    "bus": "0",
    "type": "rx",
    "ts": 46223,
    "frame": [
        {
            "id": 102277121,
            "dlc": 8,
            "rtr": false,
            "extd": true,
            "data": [0, 40, 6, 206, 40, 26, 68, 25]
        },
        {
            "id": 1089542,
            "dlc": 8,
            "rtr": false,
            "extd": true,
            "data": [0, 0, 0, 0, 0, 0, 0, 0]
        },
        {
            "id": 2138113,
            "dlc": 8,
            "rtr": false,
            "extd": true,
            "data": [0, 0, 3, 57, 110, 0, 40, 224]
        }
    ]
}
```

It seems that `102277121` has bytes 2 and 3 (6,206) which are as 16bit value 1734 (possibly RPM).

Now tried to add this as JSON format, with these settings:
[image]

Probably PID and Index should be actually `-1` but admin panel does not allow setting -1 values to those..

And sorry if I am asking questions which are already on the documentation, but considering myself as a beginner here so questions might be little stupid :)

savikko, 2024-08-18:
Actually, after Store + Reboot i got this info there and seems that PID and Index does not matter.

Now I "only" have to investigate what CAN IDs contain the needed data here..

meatpiHQ, 2024-08-29:
@savikko I suggest you try again with carscanner. Make sure to set the protocol manually to 11bit 500K  in carscanner.

stav242, 2024-10-12:
> Well, after some investigation it seems that this basic data is not broadcasted all the time so it has to ve requested.
> 
> If anyone have any idea how this could be done, I am very interested.
> 
> Using Home Assistant, but I would like WiCAN to do polling and then send receives data - parsed - via MQTT.

Hi , im also interested in this.

i have been writing up my setup.
https://sites.google.com/view/homeassistant-campervan/can-bus-integration

aim to get B-CAN data into Home Assistant.

ubarni, 2025-01-28:
I just setup a meatPi in my camper van (Fiat Ducato from 2016) and used the Standard PID Scan.
This is the result (which varies a bit when repeating) but works for me to get and send some values to HA via MQTT.
Well there are some values missing (i.e. odometer)...

[image]

meatpiHQ, 2025-01-28:
@ubarni which firmware version are you running? 

Can you please post the car scanner logs?

ubarni, 2025-01-28:
@meatpiHQ I just updated to the latest 4.04 Version.
Sorry but i am new to this. How do i get the Car Scanner Logs?
[image]

meatpiHQ, 2025-01-28:
@ubarni https://meatpihq.github.io/wican-fw/car_scanner/usage

https://meatpihq.github.io/wican-fw/car_scanner/logs

ubarni, 2025-01-30:
BTW. I followed this [YT-Video](https://www.youtube.com/watch?v=hx8UjOLD1tw) to setup my meatPi sending data to Home Assistant. Well he got a Sprinter Van. But the steps also fit to my Fiat Ducato.
But as mentioned. Just a few data (i.e. Odometer) are missing in the Standard PID Scan...

meatpiHQ, 2025-01-30:
@ubarni some manufacturers do not support getting the odometer reading on Standard PID, some have a vehicle specific PID for it. Did you see the odometer in car scanner? If so I can create a profile for your car to fetch the odometer, you need to follow the instructions in the link to export the logs for the odometer.

ubarni, 2025-01-30:
I was not able to see any odometer values yet. I will try a bit more during the weekend...

meatpiHQ, 2025-01-30:
You could also look on Fiat forums, someone must have figured it out.

7xy, 2025-02-03:
> You could also look on Fiat forums, someone must have figured it out.

Can you please post a link to the Fiat forum you mentioned. I got my meatpi WicanOBD today and also have a Fiat Ducato Camper Van.

Tidle999, 2025-02-07:
I also have a fiat camper but never was able to get values like odo or fuel state. Just the one you mentioned already. Also car scanner does not show more. 
So I think it’s not possible to get more information, right?
With the pid scanner and the ha integration (I guess alpha but it’s working) it’s a very simple setup now. Thanks!
Should the information also update when ignition is off?

stav242, 2025-02-15:
It would be possible, but needs a decent professional automotive odb
scanner .. that will have the correct codes. Then watch what is sent using
wican.

Tidle999, 2025-02-15:
Tell me what to Nix and to do and i will provide the Information here.

Tidle999, 2026-05-09:
Hi! I’ve been following your Site (https://sites.google.com/view/homeassistant-campervan/can-bus-integration) — amazing work!
Since you mentioned you were looking for the Odometer: I found it on the 50kbps B-CAN (mapped Pins 1 & 9 to 6 & 14) at ID 0x603 (1539 dec).

[image]
[image]

My Problem:
My WiCAN Pro reads these frames perfectly in the CAN Monitor (see Image 1), but the Automate section (Webhook/MQTT) keeps reporting "No frame" or read errors (see Image 2). It seems the background service is "blind" to the 50kbps bus while the monitor is active.
Two quick questions:
1.	Do you have a tip on why the WiCAN background services fail to process these 50kbps frames?
2.	Did you manage to find the ID for the Fuel Level during your reverse engineering?
Thanks for your help

ubarni, 2026-05-28:
Hi @stav242, i appreciate your work getting more data out of the Fiat Ducato CANbus 👏
I ordered some TAJ1055 and maybe there is an issue with your layout on your blog?
It seems that TX should be PIN 2 (not 4) and 3V should also be rerouted?
Could you please take a look?
THX
Barni

[image]

ubarni, 2026-08-31:
> Hi! I’ve been following your Site (https://sites.google.com/view/homeassistant-campervan/can-bus-integration) — amazing work! Since you mentioned you were looking for the Odometer: I found it on the 50kbps B-CAN (mapped Pins 1 & 9 to 6 & 14) at ID 0x603 (1539 dec).
...
> My Problem: My WiCAN Pro reads these frames perfectly in the CAN Monitor (see Image 1), but the Automate section (Webhook/MQTT) keeps reporting "No frame" or read errors (see Image 2). It seems the [...]

Hi @Tidle999. Are you able to get the MeatPi Pro get to work?
I also tried to check it. But it seems the UI and the parameter changed?

[image]

Tidle999, 2026-08-31:
For me it’s working now. I fixed the code with Claude now everything works like expected.

ubarni, 2026-08-31:
> For me it’s working now. I fixed the code with Claude now everything works like expected.

@Tidle999 : It would be great if you share your results so others are able to repeat your steps to a working result.

ubarni, 2026-09-01:
> For me it’s working now. I fixed the code with Claude now everything works like expected.

@Tidle999: And why fixing code? I hoped you got it working with MeatPi? Or did you realized it with @stav242 ESP32?
Anyway. A repeatable solution would be fine...

Tidle999, 2026-09-08:
yes i fixed it with using the meatpi (pro), here the claude summary:

## Passive/listen-only CAN sniffing via MQTT filter is silently broken (base: `ed5304c`, `wican-pro` branch)

### Context

Trying to passively read a non-standard-baudrate CAN bus (Peugeot Boxer / Fiat
Ducato comfort/body bus, 50 kbps) in listen-only mode and forward selected
signals to Home Assistant via the MQTT CAN Filter feature (odometer @ 0x603,
fuel level + ignition bit @ 0x380). This exercised several bugs that likely
affect anyone using MQTT CAN Filter outside the "500k + AutoPID" happy path.

None are fixed as of the current `wican-pro` HEAD (`43f0d1d`) — verified by
diffing against the base commit `ed5304c` this work started from; the three
commits since then (IDF v6.0.2 bump, ELM327 header-parsing alignment, a
whitespace tweak) don't touch any of the affected files.

### Root cause #1 (the actual blocker): MQTT broker URL gets double-prefixed, `esp_mqtt_client_init()` silently fails

`main/web/src/main.js`: the config-save handler unconditionally prepends
`mqtt://`/`mqtts://` to the `mqtt_url` field's current value instead of first
stripping any scheme that's already there. If the field already displays
`mqtt://<host>` (e.g. after a previous save, or because a user typed it in
manually before understanding the field is meant to hold just the bare host),
saving again produces `mqtt://mqtt://<host>` in `config.json`. The matching
load handler only strips a fixed 7/8-char prefix (`.slice(7)` / `.slice(8)`),
so a *doubly*-corrupted value still displays as if it had only one scheme —
masking the corruption in the UI while it persists in storage.

Downstream, `esp_mqtt_client_init()` rejects the malformed URI and returns
`NULL`; `mqtt_init()` (`main/mqtt.c`) then does an early `return` — **the
entire MQTT task never starts**. Nothing in the UI indicates this; the only
trace is an `ESP_LOGE` line only visible over serial/USB.

**Fix:**
- `main.js`: both save and load now strip any existing `mqtt(s)://` prefix(es)
  via regex before use, making the round-trip idempotent.
- `mqtt.c`: added `normalize_mqtt_uri()`, called unconditionally before
  `esp_mqtt_client_init()` — strips any (possibly repeated) leading scheme and
  re-applies exactly one based on the configured security setting. This makes
  the fix self-healing even for already-corrupted stored configs, without
  requiring the user to re-save settings.

### Root cause #2: CAN Filter UI can't accept hex CAN IDs, silently filters on the wrong ID

`main/web/homepage_full.html` / `main/web/src/homepage.html` /
`main/web/src/main.js`: the CAN ID input is `type="number"`. The CAN Monitor
displays frame IDs in hex (e.g. `603h`), but the filter table stores/compares
decimal. A user typing `603` (copying what they saw in the monitor) actually
filters on decimal 603 (`0x25B`), not `0x603` — the filter then never matches,
with no indication why. There's also a known related issue (#650) where
leaving PID empty (the documented way to mean "no PID, passive frame") throws
a misleading validation error meant for the PID *Index* field.

**Fix:**
- CAN ID field is now free-text and hex-aware (`0x603` parses correctly);
  the filter table displays stored IDs in hex to match the CAN Monitor.
- Empty PID now defaults to `-1` (passive), empty PID Index to `0`, with
  corrected validation messages.

### Root cause #3: one filter's cooldown blocks *all* filters sharing its CAN ID

`main/mqtt.c`, filter-processing loop: when a CAN ID has multiple filters
(e.g. two signals decoded from the same frame — fuel level and an ignition
bit both from `0x380`), and the first matched filter's publish cycle isn't
due yet, the loop does `break` — aborting the whole `while` loop instead of
just skipping that one filter. Any sibling filter on the same ID that *was*
due never gets evaluated or published that cycle.

**Fix:** changed to `continue` with `start_index = found_index + 1`, so only
the not-yet-due filter is skipped.

### Root cause #4: no listen-only protocol option

`Protocol` dropdown only offers active protocols (realdash66, slcan,
savvyCAN, elm327, AutoPID) — there's no way to run purely passively for an
MQTT-only setup without an active protocol handler engaged.

**Fix:** added `PROTOCOL_NONE` ("None (Listen-Only / MQTT)") end-to-end —
`config_server.h`/`.c` enum + string mapping, an `app_main()` branch that
just enables the CAN peripheral without starting any protocol handler, and
the corresponding UI option.

### Diagnostics gap that made all of the above hard to find

None of `main/`'s `ESP_LOG*` output is reachable without a serial/USB
connection — the web "Terminal" tab is a command console (`esp_console`),
not a log stream; no `esp_log_set_vprintf()` hook exists anywhere in the
tree. Root cause #1 in particular was invisible from the web UI in any form.

**Added (not a bug fix, but what made #1 diagnosable without a USB cable):**
a 32 KB in-RAM ring buffer (`main/web_log.c`/`.h`) hooked via
`esp_log_set_vprintf()`, exposed through a new `GET /logs` endpoint and a new
`logs` (`-c` to clear) command in the existing terminal console.

### Verification

Tested on WiCAN Pro hardware against a live Boxer B-CAN at 50 kbps, silent
mode: odometer and fuel-level values decoded and cross-checked byte-for-byte
against the dashboard (exact match, e.g. `06 51 9B 55 ...` → 105301 km via
`((B1&0x0F)*65536)+(B2*256)+B3`), delivered over MQTT to Home Assistant.
`/logs` confirmed `MQTT_EVENT_CONNECTED` and broker status flipping to
`online` after the fix, versus `esp_mqtt_client_init failed (client == NULL)`
before it.

Happy to open this as a PR against `wican-pro` if useful — diff is small and
scoped to `main/mqtt.c`, `main/web/src/main.js`,
`main/web/homepage_full.html`, `main/config_server.h/.c`, `main/main.c`,
plus the new `main/web_log.c/.h`.

ubarni, 2026-09-09:
Hi @Tidle999,

putting your Claude summary into my Gemini AI 😅 results in these settings in WiCAN Pro:

[image]
[image]

But even when ignition is on (or motor is running) there are no relevant data send via MQTT
[image]

There are also no data visible when using the CAM Monitor:
[image]

There is now just one option left from your list. **Did you made any changes in the code?**

### Root cause https://github.com/meatpiHQ/wican-fw/issues/4: no listen-only protocol option
Protocol dropdown only offers active protocols (realdash66, slcan,
savvyCAN, elm327, AutoPID) — there's no way to run purely passively for an
MQTT-only setup without an active protocol handler engaged.

Fix: added PROTOCOL_NONE ("None (Listen-Only / MQTT)") end-to-end —
config_server.h/.c enum + string mapping, an app_main() branch that
just enables the CAN peripheral without starting any protocol handler, and
the corresponding UI option.

Any other changes you made or tipps?

Thanks in advance...
Uwe

Tidle999, 2026-09-09:
Hi @ubarni,

To be clear: yes, these were real, compiled source changes — not just a
text analysis. They're verified working (odometer/fuel level decoded and
cross-checked against the dashboard, MQTT connection confirmed via
`MQTT_EVENT_CONNECTED`, broker status flipping to `online`).

The catch: **none of this is merged upstream yet**, so it only exists as a
patched fork, not in the stock firmware you're running. That's exactly why
you can't reproduce it — your Protocol dropdown shows `slcan` because
`PROTOCOL_NONE` ("None (Listen-Only / MQTT)") doesn't exist in your build
at all; it's a firmware source change (new enum value in
`config_server.h`/`.c`, a new `app_main()` branch, a new UI option), not a
setting you can configure into existence on stock firmware. No AI can
conjure a menu entry that isn't compiled into the binary you flashed.

So on that base just ask google to fix the code for you and flash a new firmware. Thats what i did with claude, on my local pc. i can provide you with the firmware (bin file) for meatpi pro, based on 4.47, if you want for testing.

ubarni, 2026-09-09:
Hi @Tidle999,

thanks for clarifying my approach. And yes. It would be great to get a test version.
Will this be also included in a later official release?

THX...
Uwe

Tidle999, 2026-09-09:
ok, feedback appreciated.

https://limewire.com/d/Nylyw#4Bj3RlI8mG

and no, as i am a normal user i wont merge anything, i jsut asked the developer for fixing the bug, and he said he will make a new reease, but this never happened so far.

ubarni, 2026-09-09:
> Hi! I’ve been following your Site (https://sites.google.com/view/homeassistant-campervan/can-bus-integration) — amazing work! Since you mentioned you were looking for the Odometer: I found it on the 50kbps B-CAN (mapped Pins 1 & 9 to 6 & 14) at ID 0x603 (1539 dec).
> 
> [<img alt="Image" width="590" height="1278" src="https://private-user-images.githubusercontent.com/36228711/590040106-6544e0f7-8f7c-43be-9c98-93791e3f5388.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODg5NTg2OTUsIm5iZiI6MTc4ODk1ODM5NSwicGF0aCI6Ii8zNjIyODcxMS81OTAwNDAxMDYtNjU0NGUwZjctOGY3Yy00M2JlLTljOTgtOTM3OTFlM2Y1Mzg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTA5VDEyNTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlZTZhNDNjOTE2ZWM4MmNiMjljNGQ0MTNkMDQ1ODE4NTFhMTMzZDIwOTBjZDBmMDE3YTRlNmFhZDVjODE3ZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.m3933Urk1KGVvXDaexLUFrya_t1y9mlY9TNaQrfsBbI">](https://private-user-images.githubusercontent.com/36228711/590040106-6544e0f7-8f7c-43be-9c98-93791e3f5388.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODg5NTc3NjAsIm5iZiI6MTc4ODk1NzQ2MCwicGF0aCI6Ii8zNjIyODcxMS81OTAwNDAxMDYtNjU0NGUwZjctOGY3Yy00M2JlLTljOTgtOTM3OTFlM2Y1Mzg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTA5VDEyMzc0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4ZDkzOThmNTZmNmQ0NGRiNjMzMDczMGJlMWE4YjU0YTEzZmRmNDE3MjU4NGMxOTNiYjNmNWRjMjVhOGYxY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.MHvbxeGDmagbclsF906p6pxhsVrmhHfsmnF6XcaXTtA) [<img alt="Image" width="590" height="1278" src="https://private-user-images.githubusercontent.com/36228711/590040107-65a29c82-3ba7-4034-a3ad-72a1a7dc06db.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODg5NTg2OTUsIm5iZiI6MTc4ODk1ODM5NSwicGF0aCI6Ii8zNjIyODcxMS81OTAwNDAxMDctNjVhMjljODItM2JhNy00MDM0LWEzYWQtNzJhMWE3ZGMwNmRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTA5VDEyNTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4NTAwMjQxYThiODU0NTNlZjA1MTE3NjQzMDBmMDFiNWJjOTQ5OTU2ZjdmNTdkN2NlY2Q1NmI5NDJmMTgyZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.5U8FKrvp_7bJBTfox0WdDOWOhk9JossRdcLqxlmP1yo">](https://private-user-images.githubusercontent.com/36228711/590040107-65a29c82-3ba7-4034-a3ad-72a1a7dc06db.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODg5NTc3NjAsIm5iZiI6MTc4ODk1NzQ2MCwicGF0aCI6Ii8zNjIyODcxMS81OTAwNDAxMDctNjVhMjljODItM2JhNy00MDM0LWEzYWQtNzJhMWE3ZGMwNmRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTA5VDEyMzc0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNiNzVlYmJiNWE5M2I4ZGE2NjRjYzg5YjJhM2M3MzJmN2U0Y2RjNTZmM2FiNjRhOGZkMGNkNmVhYTc3MDg0YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.wnytK__A1Dht7TtI1XfcMxAK2_zxMPfGjGbBpOxAGmI)
> My Problem: My WiCAN Pro reads these frames perfectly in the CAN Monitor (see Image 1), but the Automate section (Webhook/MQTT) keeps reporting "No frame" or read errors (see Image 2). It seems the background service is "blind" to the 50kbps bus while the monitor is active. Two quick questions:
> 
> 1. Do you have a tip on why the WiCAN background services fail to process these 50kbps frames?
> 2. Did you manage to find the ID for the Fuel Level during your reverse engineering?
>    Thanks for your help

@Tidle999: Did you make any changes in wiring as mentioned in https://sites.google.com/view/homeassistant-campervan/can-bus-integration ? Because i do not get any CAN Monitor data yet?

Tidle999, 2026-09-09:
**The Boxer/Ducato's B-CAN isn't on the standard HS-CAN pins.**

Pin layout on this platform:
- **OBD-II pin 6 / pin 14** = standard HS-CAN (500 kbps) — this is the only
  pair the WiCAN Pro's CAN transceiver is wired to (checked the schematic:
  one single TJA-class transceiver, hard-wired to pins 6/14, no relay or
  mux anywhere on the board — this can't be changed in software, it's a
  hardware limitation).
- **OBD-II pin 1 / pin 9** = the low-speed comfort/body B-CAN (50 kbps) —
  odometer, fuel level, etc. live here. Plugging straight into the OBD-II
  port gets you nothing on this bus, because it's never routed to the
  WiCAN's transceiver at all.

I'm using this adapter to physically remap pin 1/9 onto pin 6/14 so the
WiCAN reads the B-CAN instead of (or in addition to, depending on the
adapter) HS-CAN:
https://www.amazon.de/dp/B0BW9GFS47

ubarni, 2026-09-09:
I ordered a Multi ECU Scan OBD Adapter for FIAT. I will check the new setup on Saturday.

danderik88, 2026-09-24:
Hi! I just published a repo documenting the body CAN of my 2009 Fiat Ducato X250, read with a stock WiCAN (listen-only, 50 kbit/s on pins 6/14) and sent to Home Assistant via MQTT. Maybe it can help:

https://github.com/danderik88/fiat-ducato-x250-wican

It covers engine running, RPM, coolant, speed, odometer, range, fuel, doors, parking brake, lights and turn signals, plus the WiCAN filter, the Home Assistant config, raw captures and small tools to sniff and decode your own van.

Note that my IDs are 11-bit standard, so newer model years may use a different scheme. If you try it on your van, feedback or captures are very welcome!

JeanLucPicard-hub, 2026-09-29:
@Tidle999: I am following all the discussions on Wicn Pro on Ducato. Now I started myself to get it connected and run into the same issues you faced. Unfortunately I am not such a SW buff like you are and cannot fix it myself. Would you mind sharing your firmware image? My is a 2023 Citroen Jumper.
Thank you very much.

Tidle999, 2026-09-29:
Doesn’t it work to follow the link above with the instructions?

https://github.com/danderik88/fiat-ducato-x250-wican

Tidle999, 2026-09-29:
> Hi! I just published a repo documenting the body CAN of my 2009 Fiat Ducato X250, read with a stock WiCAN (listen-only, 50 kbit/s on pins 6/14) and sent to Home Assistant via MQTT. Maybe it can [...]

regarding fuel state....

`b5` = fuel level, most likely **litres**. The same decoding you had in the first version of the
repo before it was dropped as unconfirmed. My WiCAN MQTT filter is identical to yours:
`{"CANID":896,"Name":"fuel","StartBit":40,"BitLength":8,"Expression":"V"}`

Over several weeks of driving, `b5` followed the fuel gauge across almost the whole tank, from
full down to nearly empty and back up after refuelling:

| b5 | when |
|---|---|
| 0x5B = **91** | tank full |
| 0x47 = 71 | after driving |
| 0x24 = 36 | after more driving |
| 0x0B = 11 | nearly empty (reserve) |
| 0x4B = 75 | after refuelling |

Why I think it's litres and not percent:
- The highest value I've seen is **91** with a full **90 L** tank. In percent a full tank would read
  ~100. A tank holding slightly more than its nominal capacity fits litres.
- Your own data point fits too: `b5 = 23` → 23 L = 25.6 % of 90 L, vs 24 % from the OBD PID.

**Sample frames**

Taken from the WiCAN web CAN monitor (no timestamps, so shown as slcan-style lines without `t`):

```
t38082C044857005B0B04   b5=0x5B (91)  full
t38082C04405A00470B04   b5=0x47 (71)
t38082C04435900240B04   b5=0x24 (36)
t38082C044B5B020B0B04   b5=0x0B (11)  reserve
t38082C004059004B0B04   b5=0x4B (75)  after refuelling
t38080C004058020A0B04   b5=0x0A (10)  suspected key-off frame
```

(`b3` in these frames also matches your battery-voltage decoding: 0x57–0x5B × 0.16 = 13.9–14.6 V
with the engine running.)

JeanLucPicard-hub, 2026-09-30:
> Doesn’t it work to follow the link above with the instructions?
> 
> https://github.com/danderik88/fiat-ducato-x250-wican

Unfortunately not. Danderik88 description works on the standard wincan. I have the pro. With that one I cannot get it to work. So to the FE topics you described it just cannot, so I understand. That is why I ask if you could share your private build again, since your old link above does not lead to anything. Thanks a lot

Tidle999, 2026-09-30:
https://limewire.com/d/V14li#LZbKyZLghR

> Unfortunately not. Danderik88 description works on the standard wincan. I have the pro. With that one I cannot get it to work. So to the FE topics you described it just cannot, so I understand. That [...]

JeanLucPicard-hub, 2026-09-30:
> https://limewire.com/d/V14li#LZbKyZLghR

Thank you very much! I already installed on Wican Pro remotely. I will test later this evening and provide feedback.

Tidle999, 2026-09-30:
Looking forward to your reply. I am interested if it works also for other people. If so I can provide it here as a fork

ubarni, 2026-09-30:
I also was able to get some data using WiCAN Pro from my Ducato (2017).
I am in vacation right now and will continue testing end of next week.

danderik88, 2026-09-30:
> regarding fuel state....
>
> `b5` = fuel level, most likely **litres**. The same decoding you had in the first version of the
> repo before it was dropped as unconfirmed. My WiCAN MQTT filter is [...]

Thanks, this is really useful. I just checked on my van with the reserve light on: 0x380 = 20 00 40 59 02 0B 0B 04, so b5 = 11 and b4 = 0x02, range 110 km on the cluster. Your reserve frame has exactly the same b4 = 0x02 and b5 = 11. In all my earlier captures (b5 between 17 and 23) b4 was always 0x00, so b4 bit 1 looks like the low-fuel warning.
I'll add fuel level back to the repo: with your weeks of data from full to reserve, b5 following the gauge is confirmed on two vans. On litres vs percent I'm not fully convinced yet. My single OBD point (b5 = 23 vs 24 % from the fuel PID) actually sits closer to percent than to litres, and a full tank reading 91 would also fit a gauge that never shows 100. My next refuel will settle it: litres from the pump receipt vs the jump in b5. Until then I'll document it as "probably litres".
One small difference: my key-off frame shows b5 = 0, while your suspected key-off frame shows 10. It may just be an ordinary frame.

danderik88, 2026-09-30:
> regarding fuel state....
>
> `b5` = fuel level, most likely **litres**. The same decoding you had in the first version of the
> repo before it was dropped as unconfirmed. My WiCAN MQTT filter is [...]

Update: I'll take your full-tank test as the answer for now. 91 on a 90 L tank fits litres, not percent. The repo documents 0x380 b5 as fuel level in litres and b4 bit 1 as the low-fuel warning, with credit to you. I'll cross-check on my van at the next refuel.

Tidle999, 2026-09-30:
👍 thanks to you I managed to build an automation so Alexa tells me when I forgot to pull the parking break when switching the engine off. Great :)

danderik88, 2026-09-30:
> 👍 thanks to you I managed to build an automation so Alexa tells me when I forgot to pull the parking break when switching the engine off. Great :)

Great! If the RV doors are connected to the doors of the Vehicle like mine, you also have contact sensors and you can do an automation that alerts you if they are open while you are keep driving.