meatpiHQ/wican-fw #177: Toyota Rav4 SoC setup

phanja, 2024-08-19:
Hi, could you help with setting up soc. I tried to set up via tx in home assistant a few months back with no luck but hoping the auto pid request will work.

here is the car scanner log: 
[log r4p soc.txt](https://github.com/user-attachments/files/16658039/log.r4p.soc.txt)

let me know if anything else is needed. thanks

meatpiHQ, 2024-08-19:
@phanja I assume you clicked on the SoC sensor and the repeating PID request is the SoC ?

What was the SoC at the time of the log. It will also be helpful if you make another log with different SoC value.

`
>015B2
7EB03415BE0
7EA03415BE0

>015B2
7EA03415BE0
7EB03415BE0

`

meatpiHQ, 2024-08-19:
@phanja alright try this please, update the firmware then in the Automate tab, 

1- click on choose file and select wican_vehicle_profiles_v142.json
2- click update "Car Data"
3- Refresh the page
4- enable "car specific" 
5- Select Rav4
6- Set your Destination Topic and Time, then click store
7- Reboot

[wican-fw_v322_hv300.zip](https://github.com/user-attachments/files/16661956/wican-fw_v322_hv300.zip)
[wican_vehicle_profiles_v142.json](https://github.com/user-attachments/files/16662055/wican_vehicle_profiles_v142.json)

phanja, 2024-08-19:
Thanks I now get the SoC readings. Any resources on home assistant discovery? I'm gonna go ahead and add it through the config yaml

closed 2024-08-19 by phanja

meatpiHQ, 2024-08-20:
@phanja the discovery feature is not yet functional, I will release this firmware later this week with this feature enabled.