meatpiHQ/wican-fw #179: ABRP compatibility

teodorch85, 2024-08-20:
I would like to ask if someone has used WiCan with ABRP
https://www.iternio.com/abrp-obd

jay-oswald, 2024-08-20:
No i would like to. But you can't currently have station mode and bluetooth active at the same time. I have an issue about this, that is planned to fix this, then should he able to activate bluetooth and use with abrp.

meatpiHQ, 2024-08-20:
So if I understand correctly, you need BLE to be ON during when WiCAN is not connected to wifi. And once BLE disconnected, supposedly when you arrive home you want it to connect to you home wifi. 

The only issue here is that you have to manually disconnect from BLE.

I had issues with BLE and WiFi coexisting on the same radio, it significantly throttled the connection but that was a while ago I have to retest with the new idf to see if this was fixed and how well it works.

jay-oswald, 2024-08-20:
@meatpiHQ I don't think manually disconnecting from BLE is a major issue. Though Ideally do something smart like if ignition has been off for 2+ mins turn off the BLE.

I'd be fine with an option to only have BLE enabled when station can't connect, and then keep station disabled while BLE is connected, especailly if BLE automatically drops the connection when the car is turned off for a few mins

DeepflashX, 2024-10-23:
Hello all togetheter,
how is the status regarding ABRP?
I would be interested in the MeatPi product if ABRP would work and have Wifi running aswell.

meatpiHQ, 2024-10-27:
@DeepflashX It's not yet implemented. I will update this issue once it's ready.

> I would be interested in the MeatPi product if ABRP would work and have Wifi running aswell.

Do you connect ABRP to BLE and have WiFi connected at the same time? 

This is possible but it affect the throughput, I have this enabled at some point but it did not work so well.

DeepflashX, 2024-10-27:
@meatpiHQ ABRP would use BLE to connect to the Dongle. I would set up a Hotspot on my Android Phone and let the Dongle connect to it and upload to MQTT. So yes, Wifi and BLE would be needed same time but of course only a limted number of PIDs.
In the end I need a maximum of 20 PIDs to upload to MQTT.

cr0i, 2024-11-07:
I am also interested in using ABRP with WiCAN.
I have tried disabling WLAN Station Mode so that I can enable Bluetooth (I hope BLE?) in WiCAN.
With my iPhone, the WiCAN is not found as a Bluetooth device.
My Android has found it, but when I connect it, it immediately disconnects.
No device is found in ABRP.

Has anyone managed to get it to work with ABRP at all?

I only want to use ABRP on the go, at home I only need a WLAN connection.
In my use case, BLE is therefore only needed if the WLAN connection is interrupted.

cr0i, 2024-11-07:
Hello @meatpiHQ ,

I had to change the Protocoll from AutoPID (I used with WLAN) to elm327 (when trying BLE), so connection with ABRP was successfully.
This could be important for this solution.

meatpiHQ, 2024-11-10:
@cr0i Thanks. I want to add a "Drive mode" and "Home Mode", so when the car is running it will activate BLE and when it's parked at home it will connect to home wifi.

DeepflashX, 2024-11-10:
Both at the same time is not possible?
My plan was to set up a Hotspot on my mobile and have both ABRP running and wifi to upload to Mqtt..

mdallaire, 2025-05-29:
I am looking at the Wican Pro for my newer car but I still use a OVMS with LTE on my old Nissan Leaf and the way to link it to ABRP is using https://github.com/iternio/ovms-link to push the vehicle metrics to the ABRP api. 

Perhaps this is something that can be adapted to the Wican code? This way the metrics are pushed through the network (either cellphone hotspot or the future LTE dongle) and there is no need to enable the Bluetooth stack.

meatpiHQ, 2025-07-27:
@juggie Yes WICAN-PRO works with ABRP

gerok1984, 2025-07-29:
What settings do you use to make WiCan work with ABRP?
I've enabled BLE and selected ELM327, and it says Not Compatible.

[image]

meatpiHQ, 2025-07-29:
@gerok1984 for now only the PRO version is compatible with ABRP.

meatpiHQ, 2025-07-30:
@gerok1984 More stock should be available in few weeks. The PRO version is much easier to develop on and will be getting more advanced features, I would recommend it over the original WiCAN.

meatpiHQ, 2025-08-26:
I'm working on adding http/https , it's not specific to ABRP. Does ABRP expect a specific data format ?

cr0i, 2025-09-09:
> [@Manouchehri](https://github.com/Manouchehri) [@stevelea](https://github.com/stevelea) ABRP api is now supported please test and let me know if it works for you. I did not test with ABRP, I don't have a token yet. I've only tested on my own python server.
> 
> This new firmware will allow you to use BLE+Station mode and also support HTTPs post, you can upload certificates in the system tab if needed.
> 
> [wican-fw_obd_pro_v430p_beta5.zip](https://github.com/user-attachments/files/22236943/wican-fw_obd_pro_v430p_beta5.zip)

Hello @meatpiHQ ,
will there be an Update for WiCAN (without Pro) too, so we can use BLE one the Road, when WiFi is not connected?
This would not only be nice for ABPR, so I would also be able to use Carscanner with CarPlay/Android Auto.

stevelea, 2025-09-09:
> [@Manouchehri](https://github.com/Manouchehri) [@stevelea](https://github.com/stevelea) ABRP api is now supported please test and let me know if it works for you. I did not test with ABRP, I don't [...]

Loaded up Beta5, i cant update the network configuration as it is greyed out, i can add an additional ssid but it doesnt appear to switch to that.  Also i currenty use homeassistant to fill the data for ABRP and my car profile does not appear to support BT Connection in ABRP. When i switch on BT i get a message saying AP Will disconnect, which it doesnt BTW.

meatpiHQ, 2025-09-09:
@stevelea can you please send me a screenshot of the greyed out network configuration?

Can you confirm you have the correct security configuration wpa2 or wpa3 in the other ssid?

stevelea, 2025-09-10:
> [@stevelea](https://github.com/stevelea) can you please send me a screenshot of the greyed out network configuration?
> 
> Can you confirm you have the correct security configuration wpa2 or wpa3 in the other ssid?

User issue :-) .. ignore the comment :-)

stevelea, 2025-09-10:
> [@Manouchehri](https://github.com/Manouchehri) [@stevelea](https://github.com/stevelea) ABRP api is now supported please test and let me know if it works for you. I did not test with ABRP, I don't [...]

Just testing abrp , so in the url field for abrp_api i have https://api.iternio.com/1/tlm/send? and then in the api field i have apikey=c8b6c1b3-72be-4706-b5ea-xxxxxxxxxxxc&token=88b3baad-0b6a-4308-b6e4-xxxxxxxxxxx  , is that the format you expect ? here is my rest command i use in HomeAssistant which work btw 

 abrp:
    url: >
      https://api.iternio.com/1/tlm/send?token={{ token | default('MISSING_TOKEN') }}&tlm={
        "utc":{{ utc | default(0) }},
        "soc":{{ soc | default(0) }},
        "soh":{{ soh | default(100) }},
        "power":{{ power | default(0) }},
        "speed":"{{ speed | default('') }}",
        "lat":{{ lat | default(0) }},
        "lon":{{ lon | default(0) }},
        "is_charging":{{ is_charging | default(0) }},
        "is_dcfc":{{ is_dcfc | default(0) }},
        "is_parked":{{ is_parked | default(0) }},
        "elevation":"{{ elevation | default('') }}",
        "ext_temp":"{{ ext_temp | default('') }}",
        "odometer":{{ odometer | default(0) }},
        "est_battery_range":{{ est_battery_range | default(0) }}
      }
    method: post
    headers:
      Authorization: "APIKEY {{ api_key }}"

r1st0, 2025-10-05:
> Hello [@meatpiHQ](https://github.com/meatpiHQ) , will there be an Update for WiCAN (without Pro) too, so we can use BLE one the Road, when WiFi is not connected? This would not only be nice for [...]

@meatpiHQ will ABRP support also come to non-pro version (WICAN OBD)? or more generally, will there come further releases for the wican odb in the future?

stevelea, 2025-10-28:
> > > It seems like the latest release has ABRP support in the UI, but it doesn't seem to work as the data doesn't update in ABRP
> > > <img alt="Image" width="1629" height="952" src="https://private-user-images.githubusercontent.com/171674/506529093-7693bfac-9768-481b-adfa-d5ff06170cf2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjE2NTUwODksIm5iZiI6MTc2MTY1NDc4OSwicGF0aCI6Ii8xNzE2NzQvNTA2NTI5MDkzLTc2OTNiZmFjLTk3NjgtNDgxYi1hZGZhLWQ1ZmYwNjE3MGNmMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAyOFQxMjMzMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wODljNzU1NTA5YTA3M2U4MTU0MzQ0NWYxZjc2MTg1ZjBjYmZhZmQzNzFjZDNjZDQ2NzNiNjA5YzU1Y2I3MTM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.DMIR1w3Lr1JWh6SwOz2x-dpsKShCcrlNrmkraRSI2vc">
> > 
> > Do you have both the apikey and token in key field ? Example apikey=c8b6c1b3-72be-4706-b5ea-xxxxxxxxxxxc&token=88b3baad-0b6a-4308-b6e4-xxxxxxxxxxx
> 
> [image]
> 
> Is this the correct format? 

Yes looks ok

camueller, 2025-12-23:
With 4.45 ABRP works out-of-the-box without token etc.

[image]

[image]

guschc, 2026-10-20:
Would be great to have this!