meatpiHQ/wican-fw #186: Toyota Rav4 odometer setup

phanja, 2024-08-29:
Looking to add vehicle odometer reading and distance traveled since codes cleared. Both values are recorded in miles. I'll keep collecting more logs at different values too. Thanks

Here are a couple of logs for the odometer readings:
[(01A61) log odo 43857.txt](https://github.com/user-attachments/files/16794980/01A61.log.odo.43857.txt)
[(01A61) log odo 44046.txt](https://github.com/user-attachments/files/16794981/01A61.log.odo.44046.txt)
[(01A61) odo 44077.84 .txt](https://github.com/user-attachments/files/16794982/01A61.odo.44077.84.txt)

Here are logs for distance since codes cleared:
[(01317) Distance til code clear 7769.63 .txt](https://github.com/user-attachments/files/16795000/01317.Distance.til.code.clear.7769.63.txt)
[(01317) Distance till code clear 7759.txt](https://github.com/user-attachments/files/16795001/01317.Distance.till.code.clear.7759.txt)

meatpiHQ, 2024-08-30:
@phanja In custom PID add these values

ATSP6;ATST96;ATSH7DF;
Name: Odometer
PID: 01A61
Expression: [B3:B6]/10
Send_to: Odometer_Topic

Name: Distance_CR 
PID: 01311
Expression: (256*B3)+B4
Send_to: Distance_CR _Topic

phanja, 2024-08-31:
Alright so I was able to obtain values but some other problems have occurred as well.

- Values are incorrect:
Odometer actual value is 44131 but I am receiving {"Odometer":354020591.5,"raw":"03415BD3"}, {"Odometer":82475987.3,"raw":"0441313128"},{"Odometer":71020.9,"raw":"0641A6000AD641"}

Distance_CR actual value is 7823 but I am seeing {"Distance_CR":12428,"raw":"044131308C"},{"Distance_CR":12587,"raw":"044131312B"}, {"Distance_CR":54019,"raw":"03415BD3"}

I am also receiving some false SoC readings now, it seems to be within the first 30 seconds then sporadic afterwards. Im guessing it has something to do with the initiations  or expressions?
I'll try and collect some more logs, just been a little busy, but should have some time this weekend to browse online to see if I can find any leads on the expressions.

phanja, 2024-09-01:
@meatpiHQ so I figured out I am receiving the correct readings in some of those responses, just needed to convert km to mi. {"Odometer":71020.9,"raw":"0641A6000AD641"} and {"Distance_CR":12587,"raw":"044131312B"}. 

Now I am wondering is there a way to filter out those other responses I am receiving? I think that will fix the SoC readings as well. Thanks again

closed 2024-10-24 by phanja