meatpiHQ/wican-fw #187: New Vehicle: Holden Colorado LTZ 2024

cloudbr34k84, 2024-08-29:


meatpiHQ, 2024-08-30:
@cloudbr34k84 How about some description? 🙈

cloudbr34k84, 2024-08-30:
Hey sorry mate I wrote this then didn come back.

I finally got the OBD2 working etc, updated to the latest firmware.

I used car scanner today. Do you need the logs from car scanner?

meatpiHQ, 2024-08-30:
No worries. First, which parameters are you after?

cloudbr34k84, 2024-08-30:
Can I do all the sensors as I really not sure what I want to see or do with the sensors yet I'm home assistant

meatpiHQ, 2024-09-02:
@cloudbr34k84 here are the standard OBD PIDs, I'm planning to add all these later but for now we can add them as custom PIDs. Let me know which ones you're interested in and I can help you set it up

https://en.wikipedia.org/wiki/OBD-II_PIDs

cloudbr34k84, 2024-09-03:
Hey @meatpiHQ i reckon the below'

Engine RPM (PID 0C)
Vehicle Speed (PID 0D)
|Engine Coolant Temperature (PID 05)
Intake Air Temperature (PID 0F)
 Throttle Position (PID 11)
 Mass Air Flow (MAF) Rate (PID 10)
 Fuel Pressure (PID 0A)
 Short Term Fuel Trim (PID 06)
 Fuel Tank Level Input (PID 2F)

meatpiHQ, 2024-09-05:
@cloudbr34k84 Please try these first, let me know if it works.

This link will help you understand the calculations 

https://www.csselectronics.com/pages/obd2-pid-table-on-board-diagnostics-j1979

Custom Initialisation: ATSP6;ATST96;

Name: speed
PID: 010D1
Expression: B3
Period: 1000

send_to: mqtt_topic
Name: rpm
PID: 010C1
Expression: [B3:B4]*0.25
Period: 1000
send_to: mqtt_topic

Name: coolant 
PID: 01051
Expression: (B3*5)-40
Period: 1000
send_to: mqtt_topic

Name: fuel_level
PID: 012F1
Expression: B3/2.55
Period: 1000
send_to: mqtt_topic

cloudbr34k84, 2024-09-06:
> mqtt_topic

Thanks for this! Does this look right

[image]

cloudbr34k84, 2024-09-07:
@meatpiHQ 
in HA i got the below logs errors
```
Received message on illegal discovery topic 'homeassistant/sensor/5432048f5189/Engine RPM/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/5432048f5189/Vehicle Speed/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
``

meatpiHQ, 2024-09-07:
@cloudbr34k84  please update to the latest firmware.

https://github.com/meatpiHQ/wican-fw/releases

the configuration looks good, disable the "car specific" there is an issue with the Generic profile, the parameter name contains space. 

The custom PID do not create sensors yet those illegal discovery coming from the car specific.

cloudbr34k84, 2024-09-07:
> @cloudbr34k84 please update to the latest firmware.
>
> https://github.com/meatpiHQ/wican-fw/releases
>
> the configuration looks good, disable the "car specific" there is an issue with the Generic [...]

Ok updates to latest firmware
Disabled Vehicle Specific:

meatpiHQ, 2024-09-07:
@cloudbr34k84 anything on MQTT explorer?

cloudbr34k84, 2024-09-07:
> @cloudbr34k84 anything on MQTT explorer?

this is all i see
[image]

meatpiHQ, 2024-09-07:
Is your car engine on?

cloudbr34k84, 2024-09-07:
> Is your car engine on?

lol no, should it be?

meatpiHQ, 2024-09-07:
Lol yes! How else would you be able to read rpm and speed..

cloudbr34k84, 2024-09-07:
> Lol yes! How else would you be able to read rpm and speed..

well shit i thought it would show historic data as i have been driving around all day

cloudbr34k84, 2024-09-07:
> Lol yes! How else would you be able to read rpm and speed.. 

What other sensors can I add that don't require the car to be on?

cloudbr34k84, 2024-09-07:
Ok so how do you use the data then for km traveled per day?

meatpiHQ, 2024-09-07:
If you set it up to read the odometer, once you arrive home it will connect to your wifi and will push the value. Unless you turn off the car before it connects.

cloudbr34k84, 2024-09-07:
Ok so tomorrow I'll check and get back to you. Thanks for helping!

cloudbr34k84, 2024-09-08:
Hey mate, got data when l look at MQTT explorer. ecu_status is switching between online and offline
```
{"bus":"0","type":"rx","ts":3915,"frame":[{"id":460,"dlc":4,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":401,"dlc":8,"rtr":false,"extd":false,"data":[6,193,6,193,6,193,0,0]},{"id":417,"dlc":7,"rtr":false,"extd":false,"data":[0,32,5,64,92,92,0,0]},{"id":463,"dlc":3,"rtr":false,"extd":false,"data":[0,0,5,0,0,0,0,0]},{"id":419,"dlc":1,"rtr":false,"extd":false,"data":[128,0,0,0,0,0,0,0]},{"id":451,"dlc":8,"rtr":false,"extd":false,"data":[6,193,6,193,0,0,0,0]},{"id":465,"dlc":4,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":479,"dlc":4,"rtr":false,"extd":false,"data":[128,0,0,0,0,0,0,0]},{"id":481,"dlc":7,"rtr":false,"extd":false,"data":[64,191,5,0,0,0,0,0]},{"id":485,"dlc":8,"rtr":false,"extd":false,"data":[68,255,219,31,253,0,3,118]},{"id":647,"dlc":3,"rtr":false,"extd":false,"data":[193,0,0,0,0,0,0,0]},{"id":456,"dlc":8,"rtr":false,"extd":false,"data":[64,0,0,0,255,255,63,255]},{"id":707,"dlc":8,"rtr":false,"extd":false,"data":[8,134,6,75,0,0,200,0]},{"id":562,"dlc":8,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":499,"dlc":3,"rtr":false,"extd":false,"data":[128,128,0,0,0,0,0,0]},{"id":840,"dlc":5,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]},{"id":842,"dlc":5,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}
```
```{
  "ecu_status": "offline"
}
```

meatpiHQ, 2024-09-08:
@cloudbr34k84 please send me a screenshot of MQTT explorer

cloudbr34k84, 2024-09-08:
> @cloudbr34k84 please send me a screenshot of MQTT explorer

[image]

cloudbr34k84, 2024-09-08:
@meatpiHQ 
here is a screen recording

https://github.com/user-attachments/assets/34fc2ed8-ee86-435f-92fc-efe51930bbe7

meatpiHQ, 2024-09-08:
@cloudbr34k84 looks like the fuel level is working, but another PID is timing out.

Change the send_to, to a different topic for each parameters it will be easier to see what's going wrong.

[image]

cloudbr34k84, 2024-09-08:
[image]
@meatpiHQ 

that colorado topic is coming from Wican, Under Automate which is disabled?

meatpiHQ, 2024-09-08:
@cloudbr34k84 that's ok for now.

Please change these topics and make them unique to each parameter 

[image]

cloudbr34k84, 2024-09-08:
> @cloudbr34k84 that's ok for now.
> 
> Please change these topics and make them unique to each parameter
> 
> ![Screenshot_2024-09-08-13-19-29-174_com android chrome-edit](https://private-user-images.githubusercontent.com/94690098/365411958-4e5d79d0-9460-4491-831b-865afbdfca73.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3NjYxMTIsIm5iZiI6MTcyNTc2NTgxMiwicGF0aCI6Ii85NDY5MDA5OC8zNjU0MTE5NTgtNGU1ZDc5ZDAtOTQ2MC00NDkxLTgzMWItODY1YWZiZGZjYTczLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA4VDAzMjMzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJjZTMyZDZlM2ZjYTIxMDk1ZTE5NThmYTZkMzNiOTBhODhhMWUzNzhlZGQxMDMzYTFlZGUwYzdiNjk4ZjY1OWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UVhj8WVY80Ktl2wE84ymSz0kgvKIcjch8bse0V0sivo)
[image]

i did that already whcih is

cloudbr34k84, 2024-09-08:
also for coolant when I ask GPT about the formula is says, just learning so GPT could be wrong, but i thought i would ask

The formula (𝐵3×5)−40
(B3×5)−40 is often used to calculate intake air temperature from certain OBD-II PIDs, such as PID 0F (Intake Air Temperature).

Here’s the breakdown:

Name: Intake Air Temperature (or other temperature sensors with similar scaling)
PID: 0F (for Intake Air Temperature)
Expression: Temperature (°C) =(𝐵3×5)−40
Temperature (°C)=(B3×5)−40
Where:

B3 is the raw data byte from the ECU.
Multiply B3 by 5, and then subtract 40 to get the temperature in degrees Celsius.
For example, if B3 = 10 (which is 16 in decimal):

Temperature (°C)=(B3×5)−40°C
(16×5)−40=80−40=40°C
This formula is typically used when the sensor returns data that needs to be scaled by 5 and adjusted by a constant (in this case, subtracting 40).

meatpiHQ, 2024-09-08:
The expression for the coolant temperature should be B3-40

[image]

cloudbr34k84, 2024-09-09:
@meatpiHQ what else do i need to do mate?

meatpiHQ, 2024-09-11:
@cloudbr34k84 do you want to add the rest of the parameters? I assume the coolant temperature worked correctly?

cloudbr34k84, 2024-09-11:
@meatpiHQ 
Before I do add more how do I get this data into home assistant? I don't see these current parameters as sensors in HASS

cloudbr34k84, 2024-09-16:
@meatpiHQ im still unsure how to the home assistant integration works?

meatpiHQ, 2024-09-16:
@cloudbr34k84 sorry about the delay, I'll update you later today. 

I'll add the PIDs to the generic profile which will create the sensors in HA.

cloudbr34k84, 2024-09-16:
> @cloudbr34k84 sorry about the delay, I'll update you later today.
> 
> I'll add the PIDs to the generic profile which will create the sensors in HA.

ahh ok, no worries, then how do i go about adding other PID's?

meatpiHQ, 2024-09-16:
@cloudbr34k84 I'll do the same for the others as well.

meatpiHQ, 2024-09-16:
@cloudbr34k84 Try this please, this will create the sensors but you still have to add them [manually in HA](https://www.home-assistant.io/docs/configuration/customizing-devices/)

https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles.json

cloudbr34k84, 2024-09-16:
> @cloudbr34k84 Try this please, this will create the sensors but you still have to add them [manually in HA](https://www.home-assistant.io/docs/configuration/customizing-devices/)
> 
> https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles.json

OK so im updated the. Do i need to update Vehicle Specific:
[image]

meatpiHQ, 2024-09-16:
@cloudbr34k84 yes, enable car specific and select generic. You should also delete the custom one's it's not needed anymore.

cloudbr34k84, 2024-09-16:
> @cloudbr34k84 yes, enable car specific and select generic. You should also delete the custom one's it's not needed anymore.

@meatpiHQ do i delete Custom Initialisation: ATSP6;ATST96 as well

meatpiHQ, 2024-09-16:
@cloudbr34k84 you can leave that.

cloudbr34k84, 2024-09-17:
@meatpiHQ ok i see the 4 sensor entities. For some reason i have a sensor called 'sensor.colorado_fuel_level that i cant delete and when i search for it in the logs i says the below. i dont have any mqtt sensors for fuel level, so im not sure how this is being created?

26337,1007: ⟪ 806 characters skipped ⟫ensor.colorado_wican_status":{"s":"unavailable","a":{"friendly_name":"Colorado WiCAN Status"},"c":"01J7SJXBTA03EJHHT24N251BWV","lc":1726361808.7147975}}},"id":2},{"type":"event","event":{"a":{"sensor.colorado_fuel_level":{"s":"unavailable","a":{"friendly_name":"Colorado Fuel Level"},"c":"01J7SJXBTBMNVZK7ASFHZRJR3X","lc":1726361808.7150974}}},"id":2},{"type":"event","event":{"event_type":"service_registered","data":{"domain":"text","service":"set_value"},"origin":"LOCAL","time_fired":"2024-09-15T00:56:48.716268+00:00","context":{"id":"01J7SJXBTC3SDWNAMM4C4K8CKJ","parent_id":null,"user_id":null}},"id":8},{"type":"event","event":{"event_type":"component_loaded","data":{"component":"text"},"origin":"LOCAL","time_fired":"2024-09-15T00:56:48.716430+00:00","context":{"id":"01J7SJXBTC1JMXBGAW2C1THRES","parent_id":null,"user_id":null}},"id":6},{"type":"event","event":{"event_type":"component_loaded","data":{"component":"text"},"origin":"LOCAL","time_fired":"2024-09-15T00:56:48.716430+00:00","context":{"id":"01J7SJXBTC1JMXBGAW2C1THRES","parent_id":null,"user_id":null}},"id":13},{"type":"event","event":{"event_type":"service_registered","data":{"domain":"mqtt","service":"reload"},"origin":"LO

meatpiHQ, 2024-09-17:
@cloudbr34k84 can you please check the "Destination topic" using mqtt explore?

cloudbr34k84, 2024-09-17:
> @cloudbr34k84 can you please check the "Destination topic" using mqtt explore?
 checked 
Here is a screenshot of everything with the word Colorado in it
[image]