meatpiHQ/wican-fw #188: boot loop with  v3.27 (and also  v3.28)

rcmcronny, 2024-08-30:
I did an upgrade to 3.27 some days ago and the adapter goes into a bootloop, it did get a short look on seriell and it showed a parsing error , did not have it anymore in detail.
So, today the 3.28 came out and i disassembled the adapter and flashed over usb with full factory reset.
I connected, configured ap+station with new passwortd and the wifi data. Save -> reboot  and all fine.
Then i configured the autpid protokoll -> save and boot loop, as the seriel was not stable cause of the quick reboots, i did not see any detail.

So i factory reset it again, and configured ap+station , new password, uploaded the vehicle json and set it to generic ca and enabled, but leave the protocol on slcan.  This is stable now.  I just do not want to change it and  have to disassemble it again ;)
I wanted to get more data with car scanner,l but no time till today, so only tinkering from time to time shortly.

meatpiHQ, 2024-09-02:
@rcmcronny sorry about that, this is fixed in v3.30

https://github.com/meatpiHQ/wican-fw/releases/tag/v3.30

rcmcronny, 2024-09-02:
Hi, all good,

i reported it, to hopefully get a fixed version, and you nailed it :) 

Thanks Ronny

panlinux, 2024-09-16:
I suppose this issue can be closed? I had the same issue, and 3.30 fixed it for me.

closed 2024-09-16 by rcmcronny