meatpiHQ/wican-fw #189: Difficulty integrating with Home Assistant

ghgeiger, 2024-08-31:
I am having a hard time integrating with Home Assistant. I have the wican connected to my car (2024 Rav4 Prime), network & MQTT. I am able to use car scanner to view sensors over my lan, and I can see activity in MQTT Explorer under topic wican/XXXXX/can(or elm327 with elm logging on). I've followed the instructions to enable the Home Assistant integration, but no new topics are created for the sensors. I initially set up with Destination Topic set to "rav4", so I thought that perhaps you were looking for my specific Home Assistant topic, so I tried "homeassistant/sensor/rav4", also just "homeassistant" & "homeassistant/sensor", which is where my other Home assistant devices publish, but none of these create any new MQTT topics for the sensors. Can you help me?

meatpiHQ, 2024-08-31:
@ghgeiger what's your firmware version? 

Can you please send me a screenshot of your device configuration webpage?

ghgeiger, 2024-08-31:
Sorry that I forgot the version. I'm currently on 3.28, but it did the same thing with 3.27. Here's a screenshot of my setup:

[image]

meatpiHQ, 2024-09-02:
@ghgeiger please update to the latest firmware and try it, you can select Rav4 from the drop down list

https://github.com/meatpiHQ/wican-fw/releases
https://github.com/meatpiHQ/wican-fw?tab=readme-ov-file#getting-started-guide

ghgeiger, 2024-09-02:
I've updated to v3.30 & went through the directions again. I will note that one thing I hadn't initially followed correctly was that I had initially entered my MQTT username & password as set up in the HA Add-on, as this is how I've always used MQTT wit h HA in the past. I set up a user account for wican in HA & tried again, but I haven't progressed much. There hasn't been any new sensors created, but at one point while changing settings, I found one message posted to a new topic under "homeassistant" that was simply "status > online", however that has since disappeared. Is there a simple way for me to perform a factory default & try to start fresh? I've included a few screenshots from MQTT explorer if it helps.

[image]
[image]

meatpiHQ, 2024-09-02:
@ghgeiger did you change the protocol to autopid?

ghgeiger, 2024-09-02:
No, I didn't see that anywhere in the instructions. It was still set to el327 from previous trials to get MQTT working before updates that added the Automation tab. 

That was the problem, it works now. Thanks for everything!

closed 2024-09-02 by ghgeiger

meatpiHQ, 2024-09-02:
This step was missing from the documentation, I fixed it now.