meatpiHQ/wican-fw #194: [REQUEST] Official "WiCAN integration for Home Assistant" from MeatPi Electronics?

Hedda, 2024-09-03:
Note! This is a follow-up to https://github.com/meatpiHQ/wican-fw/issues/3 (which should be moved to [discussions](https://github.com/meatpiHQ/wican-fw/discussions) as it only suggest to get Home Assistant community to do this).

This a formal request for an official "WiCAN integration for Home Assistant" (via a "custom component") from MeatPi Electronics.

@meatpiHQ please consider creating and maintaining an official "WiCAN integration for Home Assistant" from MeatPi Electronics.

Maybe you can consider adding development of a MeatPi official dedicated Home Assistant integration for WiCAN (native "custom component" for Home Assistant) as a stretch goal to represent it as a further funding milestone in your [ongoing crowdfunding campaign for the WiCAN PRO on Crowd Supply](https://www.crowdsupply.com/meatpi-electronics/wican-pro) as that could help spread the news and get more Home Assistant users interested in buying the WiCAN hardware (and get more developers interested in contributing to the firmware)?

* https://www.crowdsupply.com/guide/funding-goals

The reason for this request is that while it is great that the [WiCAN firmware update has improved support for Home Assistant](https://www.crowdsupply.com/meatpi-electronics/wican-pro/updates/wican-firmware-update-improved-support-for-home-assistant) that way of directly using MQTT via Home Assistant is still too advanced for beginners (or users with limited time) and I do not think that is simple enough to use by the broader userbase of mainstream Home Assistant end-users, and what is missing and would be needed to target that mainstream userbase is a have a more streamlined setup for using WiCAN with Home Assistant, and in essense means having a dedicated "Home Assistant integration" (also refered to as "custom component" by developers) and for that WiCAN integration to present each WiCAN adapter as a "device" and then each endpoint attribute on that device as an "entity" ("entities") + if services are available; a set of actions that can include triggers, conditions, and actions that can be used and triggered in Home Assistant automations.

* https://www.home-assistant.io/getting-started/concepts-terminology/

So if Home Assistant users are a part of your targeted market audience then should should look into and try to prioritize developing an official "Home Assistant integration" for WiCAN, an integration that under the hood makes use of MQTT or other available APIs and presents them as native "entities" ("entity") via that WiCAN integration. I believe having even just a minimum WiCAN integration will help get the word out:

* https://developers.home-assistant.io/docs/architecture_components/

* https://developers.home-assistant.io/docs/creating_component_index/

One of the main use cases with Home Assistant integrations is protocol and dependency abstraction, so from an end-users point-of-view they do not need to know or care what APIs that integration is using to communicate with the WiCAN firmware, so in practice the end-users will not need to know that it depends on MQTT (as Home Assistant configuration flow will automatically initiate the installation and setup of an MQTT broker in Home Assistant if the user do not already have one configured).

You as the developer of the integration can choose what endpoints should be used and which attributes of what endpoint should be exposed and presented as native Home Assistant "entities" ("entity") via that WiCAN integration which can be trigged in automations via actions. You can also use templating and templates inside the integration to have it present more advanced features:

* https://www.home-assistant.io/integrations/template/
* https://www.home-assistant.io/docs/configuration/templating/
* https://www.home-assistant.io/docs/automation/trigger/
* https://www.home-assistant.io/docs/automation/templating/

I would recommend that you start by creating a simple WiCAN integration as a "custom component" for Home Assistant and get that published via [HACS (Home Assistant Community Store)](https://www.hacs.xyz/) to get more eyes on it before you begin to look at getting it into the Home Assistant core (as integration components that are included there and ships with Home Assistant core by default have higher requirements). Recommend that you check out these additional links on the topic of creating a custom component for Home Assistant (note that they are a little older so might not be fully up to date with the latest Home Assistant development standard and structure), where many of them contain infromation on how-to write a custom component integration that commuicate with MQTT:

* https://github.com/home-assistant/example-custom-config
  * https://github.com/home-assistant/example-custom-config/tree/master/custom_components/mqtt_basic_async
  * https://github.com/home-assistant/example-custom-config/tree/master/custom_components/mqtt_basic_sync
* https://developers.home-assistant.io/docs/development_environment/
* https://svrooij.io/2023/01/18/home-assistant-component/
  * (and his proof-of-concept example https://github.com/svrooij/home-assistant-mqtt-component )
* https://dev.to/adafycheng/write-custom-component-for-home-assistant-4fce
* https://aarongodfrey.dev/home%20automation/building_a_home_assistant_custom_component_part_1/
* https://github.com/ludeeus/integration_blueprint/tree/main/custom_components/integration_blueprint
* https://github.com/oncleben31/cookiecutter-homeassistant-custom-component

The purpose of having a ready-made integration for WiCAN would be to make it easy for anyone to install/configure and get started and that would really be the best way to get more Home Assistant users interested in the WiCAN hardware and firmware project. The fact is that many (if not most) Home Assistant beginners will first search Home Assistant integrations page (which only includes integration components  built-into Home Assistant core), and more experienced Home Assistant end-users will extend that search to also include HACS (Home Assistant Community Store for integrations shared by community developers there but not included yet as built-in Home Assistant core integration components), and you can expect most only buy a product if it already has a ready-made integration on one of those because otherwise it is often simply to much work for normal end-users to maintain and update it.

* https://www.home-assistant.io/integrations/

* https://www.hacs.xyz/

  * https://www.hacs.xyz/docs/publish/integration/

PS: Check out this new blog post which explain what is new in HACS 2.0 (which made it easier to install community integrations)

* https://www.home-assistant.io/blog/2024/08/21/hacs-the-best-way-to-share-community-made-projects/

Hedda, 2024-09-03:
As also mentioned in -> https://github.com/meatpiHQ/wican-fw/issues/3

Suggest that you as well check out Home Assistant's existing integrations (custom components) that are MQTT-based and present different device types and services as actions which can be triggered. So you can make a new dedicated Home Assistant integration custom component that utilize your existing MQTT support and expose/present relevant functions/features as makee that new dedicated WiCAN integration work as a kind of generic "Car/Truck OBD and CAN Bus" custom integration by having that Home Assistant integration act like a virtual "hub" (also referred to as bridge or gateway) for that in turn makes use of all different existing MQTT integrations under the hood and make the WiCAN Can Bus adapter duplicate and abstract relevant MQTT topics and subscriptions to show them as kind of "hub" that present one of multiple "devices" that expose endpoint attributes as "entities" with different support "device types" (example "sensor", "binary sensor", etc.), as well as "events" for momentary things, to Home Assistant:

https://www.home-assistant.io/integrations/#hub

See example existing integrations "Velbus", "Modbus", and "OpenTherm Gateway" integrations which all acts as a "hub" integration (hubs == gateways/bridges with APIs, meaning they can abstract multiple devices which they, in turn, can present to Home Assistant in a uniformed matter to Home Assistant), as well as the "MQTT Alarm Control Panel" which is an "alarm" category integration:

https://www.home-assistant.io/integrations/manual_mqtt/
https://www.home-assistant.io/integrations/velbus/
https://www.home-assistant.io/integrations/modbus/
https://www.home-assistant.io/integrations/opentherm_gw/
https://www.home-assistant.io/integrations/supla/
https://www.home-assistant.io/integrations/thethingsnetwork/

Anyway, there are loads of MQTT integrations which in turn can be used indirectly from other integrations as dependencies:

https://www.home-assistant.io/integrations/#search/mqtt
https://www.home-assistant.io/integrations/mqtt/
https://www.home-assistant.io/integrations/mqtt_eventstream/
https://www.home-assistant.io/integrations/mqtt_statestream/
https://www.home-assistant.io/integrations/binary_sensor.mqtt/
https://www.home-assistant.io/integrations/sensor.mqtt/
https://www.home-assistant.io/integrations/device_trigger.mqtt/
https://www.home-assistant.io/integrations/number.mqtt/
https://www.home-assistant.io/integrations/select.mqtt/
https://www.home-assistant.io/integrations/button.mqtt/
https://www.home-assistant.io/integrations/switch.mqtt/
https://www.home-assistant.io/integrations/light.mqtt/
https://www.home-assistant.io/integrations/scene.mqtt/
https://www.home-assistant.io/integrations/lock.mqtt/
https://www.home-assistant.io/integrations/alarm_control_panel.mqtt/
https://www.home-assistant.io/integrations/siren.mqtt/
https://www.home-assistant.io/integrations/fan.mqtt/
https://www.home-assistant.io/integrations/climate.mqtt/
https://www.home-assistant.io/integrations/device_tracker.mqtt/
https://www.home-assistant.io/integrations/mqtt_json/

Also check out other "helper", "utility", and "automation" integrations that they have which you can make use of in integrations:

https://www.home-assistant.io/integrations/#helper
https://www.home-assistant.io/integrations/#utility
https://www.home-assistant.io/integrations/#automation

It would also be a nice feature to include the export integration diagnostics for troubleshooting the integration for HA, see:

https://www.home-assistant.io/integrations/diagnostics/

Hedda, 2024-09-03:
Oh, and if you @meatpiHQ would consider this then perhaps add a link to this request under your features wishlist for tracking:

* https://github.com/meatpiHQ/wican-fw/blob/main/README.md#features-wishlist

...or create a roadmap under the "Projects" (Kanban board) section in this GitHub repository for better tracking (and transparency):

* https://github.com/meatpiHQ/wican-fw/projects?query=is%3Aopen

  * https://docs.github.com/en/issues/planning-and-tracking-with-projects/learning-about-projects/about-projects

meatpiHQ, 2024-09-04:
@Hedda I’m planning to create a custom component for WiCAN, especially now that I want to support ESPNetLink as a GPS tracker and LTE gateway for WiCAN PRO. This custom component will include options for an MQTT bridge to connect to public MQTT brokers, along with full integration of WiCAN functionality. Additionally, I want to include support for HTTP and webhooks, which will simplify things even further.

> MQTT via Home Assistant is still too advanced for beginners (or users with limited time), and I don’t think it’s simple enough for the broader Home Assistant userbase.

I’m aware of this, which is why I’m focusing on making it as user-friendly as possible by utilizing automation and car profiles. The firmware development is largely driven by the Home Assistant user base, and that’s where this project is headed.

I’ve looked into examples of custom component integration, and there’s still a lot for me to learn. It’s going to take some time to get everything working because I’m juggling hardware, firmware, and production. However, I think I can start with something basic and gradually build on it.

Hedda, 2024-09-04:
> _I’ve looked into examples of custom component integration, and there’s still a lot for me to learn. It’s going to take some time to get everything working because I’m juggling hardware, firmware, and production. **However, I think I can start with something basic and gradually build on it.**_

That sounds great! I believe that as long as you can provide something that meets your own idea of what a custom WiCAN  integation with the bare-minimum of basic MVP (Minimal Viable Product) features then that will be something than you yourself and the community can build on togther then there should be no need to set the bar too high on an initial version.

typxxi, 2024-09-04:
Can't agree more.

Europe is becoming the next big EV market and that is a key market for the WiCan dongle which works, 
but that was a challenging adventure and people would not buy it if I tell them that I had needed 100 hours or more cause the e-GOLF has not been done before and everything was different than many recommandations which referred to e-GOLF is like an e-Up. 

But the market is growing fast especially for all those used cars which had been sold new as company car and are now bought by private owners who want to get rid of VW cloud and all the other manufacturers.

It feels somehow that the current average driver and non it guy is not your target group considering the small amount of master templates that would work. You need a lot and deep knowledge to get start.

I am happy but my father asked if I am crazy and why it takes so much time. 

If you have such integration than try to get someone on board to speak about it like Andy from the channel off grid garage.
He is famous to push manufacturers to develope & engineer good products, he is an australian resident, he welcomes his viewers (100.000 subscribers) welcom to sunny hot australia. We have 22Ampere ... and shows the sunny sky or clouds.

He does diy solar power systems and batteries, then charging evs and what not.

He is a german emigrant and I guess you should get allong quite well cause even the chinese are cooperating with him.
Search for "off grid garage". His audience is the usual diy guys and therefore the key audience or evs and charging and then of cause meatpi wican dongles, not the usual garage or workshop. 

Good luck.

Hedda, 2024-09-05:
> If you have such integration than try to get someone on board to speak about it like Andy from the channel off grid garage.
> He is famous to push manufacturers to develope & engineer good products, [...]

FYI, Phil Hawthorne who co-hosts the Home Assistant podcast is also an Australian, so maybe once you released an initial pre-alpha version of a WiCAN integration for Home Assistant then you could contract him and ask what car/cars/truck/trucks he owns so can create profiles for his vehicles and then get him to test a prototype or pre-release unit of the upcoming WiCAN Pro in the hopes of getting a interview spot on the Home Assistant podcast where can do a shameless promition for it ;)   You can probably easist reach him via the Home Assistant podcast contact channels:

https://hasspodcast.io

https://bio.link/phil

https://github.com/philhawthorne

https://twitter.com/philhawthorne

https://www.greataustralianpods.com/home-assistant/

meatpiHQ, 2024-09-22:
@Hedda @typxxi it's getting there :)

https://github.com/jay-oswald/ha-wican

jay-oswald, 2024-10-16:
Closing this one off, as MeatPi said a few weeks ago I've got a HA integration, its mostly working, with a few bugs I want to get sorted out before we do a 1.0 release of it, and then can start adding some nice to have features.

@Hedda would be great to get exposure on the HA Podcast, it could certently generate a lot of interest in the project, and hopefully sell a lot of the units for MeatPi.  Though IMO I'd want the integration at least in 1.0, which really just means the basics are working and stable, and would probably be a good idea to also wait for the pro release, since thats not all that far off anyway.

Though once thats done reaching out to him to see if hes interested would be a good idea

closed 2024-10-16 by jay-oswald

Hedda, 2024-10-16:
> would be great to get exposure on the HA Podcast, it could certently generate a lot of interest in the project, and hopefully sell a lot of the units for MeatPi. Though IMO I'd want the integration [...]

@philhawthorne have you preordered the upcoming WiCAN Pro? https://www.crowdsupply.com/meatpi-electronics/wican-pro

philhawthorne, 2024-10-17:
@Hedda I haven't. 

I've only had a brief read over this, and not sure how it checks back in to Home Assistant. Is it WiFi? Getting a data-only sim down under is pretty expensive per month, so unsure about it for me. But this does look very cool. 

Will have to see if my Aussie-Korean (Kia) even has an OBD port

jay-oswald, 2024-10-17:
Hey @philhawthorne I've done the HA integration for WiCAN

Every car since early 2000's has an OBD Port, so your Kia probably does, unless its very old

The regular WiCAN (not pro) doesn't have 4G, that's just going to be an optional extra, so yes the primary method is WiFi.

Currently HA pings the WiCAN every 30 secs to pull data, would also like to support WiCAN pushing data to HA for 4G as well, but thats probably quite a while away, even just for firmware support before I even add it to the integration. The most common use case at the moment is people with EV's getting the battery level of the EV, so that charging can be automated to minimise charging costs, but make sure it always has enough charge, however lots and lots of data can be pulled out.

Here's some HA use cases I've thought of and would like to do at some point

* Getting the car odometer combined with a "last maintenance" helper in HA could automate maintenance reminders in HA.
* Some cars have tyre pressure sensors, so you could get flat tire alerts via HA, rather than finding out when you go to drive.
* You could get "door left open" alerts so if the door doesn't get shut properly the interior light won't be on all night flattening the battery.

There's lots more data that can be gotten out of cars to do various alerts/automation', and some cars expose lots more data via the OBD port for other various automation's and alerts

philhawthorne, 2024-10-17:
Well that was easy to find! My car does indeed have an OBD port, but unfortunately its right behind the panel for the fuse box. So impossible to have anything plugged in (even an extension cable) and have the fuse box closed. Yuk.

I would love all those sensors in HA. Very useful to have. Would find the fuel gauge handy in HA, and we're always leaving the car unlocked overnight, that would be a win. Just so annoying the OBD port is in an awkward place

meatpiHQ, 2024-10-17:
@philhawthorne 

> Getting a data-only sim down under is pretty expensive per month, so unsure about it for me. 

This not necessarily true :). ALDI Australia offers a long expiry SIM, which I've been using, it's the cheapest plan I could find downunder. And there's plenty of of iot data SIMs out there comes down to couple of $ a month. 

https://www.aldimobile.com.au/products/payg

Hedda, 2024-10-17:
> Will have to see if my Aussie-Korean (Kia) even has an OBD port

@philhawthorne all cars and trucks you can buy for the last 20-years in first world countries must have an ODB-II port by law (so does even most but the least expensive ”cars” from second and third world countries too), so unless you have a very old car or import a moped-car from Africa the chances are practically 100% that your car has an OBD-2 port.

OBD-II becomame mandatory in the year 1996 for all cars manufactured in the United States. EOBD (European version of OBD-II) then became
mandatory in 2001 for all gasoline vehicles in the European Union (EU), and since 2003 EOBD becamw mandatory for all diesel vehicles in the EU.

Hedda, 2024-10-17:
> Is it WiFi

Both the existing WiCAN and the upcoming WiCAN Pro use just Wi-Fi and Bluetooth as standard, but you will also have the option to buy an optional LTE (4G) addon-module in the future.

All local only, no cloud, even if would use LTE.

Hedda, 2024-10-17:
> Well that was easy to find! My car does indeed have an OBD port, but unfortunately its right behind the panel for the fuse box. So impossible to have anything plugged in (even an extension cable) and have the fuse box closed. Yuk.

There are flat OBD extension cords that you can buy.

philhawthorne, 2024-10-17:
Very nice! Well when you're ready happy to have someone on the podcast to talk about it, particularly for the Home Assistant integration and some very fun use cases.

I do like the EV angle too. I assumed with things like Tesla having an app, they wouldn't expose an OBD port, so this might be an option for those that don't want to pay for Tesla fleet API access.

> There are flat OBD extension cords that you can buy.

Just ordered one off Amazon to arrive on the weekend. Will give it a go - and if I can still close the fuse box I'll order one of these bad boys to test out

meatpiHQ, 2024-10-17:
@Hedda Some users are connecting the WiCAN-USB directly to the harness using the screw terminal connector for power and CAN. 

https://sites.google.com/view/homeassistant-campervan/can-bus-integration

Hedda, 2024-10-17:
> Some users are connecting the WiCAN-USB directly to the harness using the screw terminal connector for power and CAN.
> 
> https://sites.google.com/view/homeassistant-campervan/can-bus-integration

Yeah but for the new WiCAN Pro you would need to sell an optional OBD to screw-terminal converter adapter for that, or the users need to make their own by cutting a extension cord in half

jay-oswald, 2024-10-17:
@philhawthorne 

> Well that was easy to find! My car does indeed have an OBD port, but unfortunately its right behind the panel for the fuse box. So impossible to have anything plugged in (even an extension cable) and have the fuse box closed. Yuk.

I'm in the exact same situation, Can't even plug in the flat extension cable, I'm planning to 3d print a new cover for the fuse box with enough of a bump for the extension cable, for now I just live with the cover off, or jam wires in the connector so its slimline.

> Very nice! Well when you're ready, we're happy to have someone on the podcast to talk about it, particularly for the Home Assistant integration and some very fun use cases.

I'd be happy to chat on the podcast about it and the HA side of it, just messaged MeatPi and hes happy for me to do it as well, let me know how to get in contact with you and we can sort out details

Hedda, 2024-10-17:
> I do like the EV angle too. I assumed with things like Tesla having an app, they wouldn't expose an OBD port, so this might be an option for those that don't want to pay for Tesla fleet API access.

Again, ODB-II port is required by law in the USA and the EU, even on EV cars like those from Tesla.

Hedda, 2024-10-17:
@meatpiHQ can you as repo admin move this issue to discussions so do do not need to stay closed? https://github.com/meatpiHQ/wican-fw/discussions