meatpiHQ/wican-fw #197: fw3.30 not compatible with Ioniq5?

tung256, 2024-09-06:
i have this for settings:
[image]

unfortunately, i dont see any useful values in MQTT:
[image]

also, i dont see anything new in my MQTT devices
[image]

tung256, 2024-09-06:
i see many bin files in the ZIP file.
i assume i only need to upload this BIN file:
[image]

meatpiHQ, 2024-09-06:
@tung256 It’s working well for other users with the Ioniq.

Is your vehicle a 2017 Ioniq or an Ioniq 5/6?

The cycle time is currently set to 2000ms. I suggest trying a cycle time of 5000ms first.

meatpiHQ, 2024-09-22:
@tung256 the PIDs in the ioniq5-6 profile should work for you, they match the ones mentioned in [here](https://github.com/meatpiHQ/wican-fw/discussions/178#discussioncomment-10421135)