meatpiHQ/wican-fw #205: Under Generic Profile Intake Airtemp

cloudbr34k84, 2024-09-18:
The Intake Airtemp PID is creating a sensor in Home Assistant as % where it should be a Temperature class

OBD2 PID for Intake Air Temperature:
The PID for Intake Air Temperature is 0x0F.

**Formulabfor Calculating Intake Air Temperature:**

- The response for PID 0x0F is a single byte (A).
- The formula to convert the byte (A) into the actual temperature is:
- Intake Air Temperature (°C)=𝐴−40

**How to Use the Data:**

- When you request PID 0x0F, you’ll get a response that contains the value of A (in hexadecimal).
- Convert the value of A from hexadecimal to decimal.
- Apply the formula above to calculate the temperature.

Let’s say the OBD2 scanner reads a value of 0x45 for PID 0x0F.
Convert 0x45 to decimal:
0𝑥45=69 (decimal)
Apply the formula:
Intake Air Temperature (°C)=69−40=29°C

seanauff, 2024-10-03:
This is addressed in my PR #220

jay-oswald, 2024-10-15:
Was fixed in #220