meatpiHQ/wican-fw #206: Generic Profile vehicle_speed appears to be incorrect?

cloudbr34k84, 2024-09-18:
The sensor for vehicle_speed is reporting 0-250km/h
[image]

OBD2 PID for Vehicle Speed:

The PID for vehicle speed is 0x0D.
**Formula for Calculating Vehicle Speed:**

- The response for PID 0x0D is a single byte (A).
- The formula to convert this byte (A) into vehicle speed is:
- Vehicle Speed (km/h)=A

**How to Use the Data:**
When you request PID 0x0D, you will get a value for A in hexadecimal.
Convert the hexadecimal value of A to decimal, and that is the vehicle speed in kilometers per hour (km/h).

**Example:**
Let’s say the OBD2 scanner reads a value of 0x28 for PID 0x0D.
Convert 0x28 to decimal:
0x28=40 (decimal)
This means the vehicle speed is:
Vehicle Speed=40km/h

meatpiHQ, 2024-10-15:
@cloudbr34k84 is this still an issue for you? 

The Generic Profile speed is working for me, this is a standard PID. If you're seeing the wrong value in WiCAN but the correct value in car scanner app, this means there is a different PID specific for your car.

cloudbr34k84, 2024-10-15:
Hey @meatpiHQ 
I haven't had time to check this because I'm having to rebuild my home assistant server. I have a backup but there is something causing high temps 90-100 and subsequently my server died.
Close it off for now