meatpiHQ/wican-fw #207: Generiic Profile Fuel Level % not correct

cloudbr34k84, 2024-09-18:
I just filled up my tank and the fuel level sensor is only at 35%? 
[image]

seanauff, 2024-10-02:
Appears related to #206. Expressions in the profile look correct. Perhaps it's using the wrong byte?

cloudbr34k84, 2024-10-03:
> Appears related to #206. Expressions in the profile look correct. Perhaps it's using the wrong byte?

yeah inthink we have determine that my vehicle does not follow the standard PID's so @meatpiHQ is creeating a car profile my Holden Colorados

meatpiHQ, 2024-10-03:
@cloudbr34k84 let me know if you have found a correlation between the value you're seeing and the actual fuel level.

cloudbr34k84, 2024-10-03:
> @cloudbr34k84 let me know if you have found a correlation between the value you're seeing and the actual fuel level.

hey sorry, ill check it over the weekend when ill fill up again

cloudbr34k84, 2024-10-28:
@meatpiHQ 

My fuel tank is now full.  Data is saying its 35.69 = 76lt

New Scaling Factor Calculation:

Scaling Factor = 76 / 35.69 ≈ 2.13

Applying the Updated Formula:
Multiply the OBD-reported fuel level by this updated factor:

Fuel Level in Liters = OBD Reported Value × 2.13

Example:
If the OBD-reported fuel level is 35.69:
35.69 × 2.13 ≈ 76 liters

This should now provide accurate fuel readings aligned with your 76-liter capacity.

cloudbr34k84, 2024-11-12:
@meatpiHQ do you think we can re look at my use case for getting this and other data?

cloudbr34k84, 2024-11-16:
@meatpiHQ no dramas, i ahve had no time to look at this until now. Have uploaded and will check it tomorrow when I fill up