meatpiHQ/wican-fw #208: Generic Profile: Odometer Issue

cloudbr34k84, 2024-09-18:
attached is my car scanner log file. Im unable to get a reading on my odomter. In car Scanner, i was able to see Odometer Engine that shows my odometer number.

cloudbr34k84, 2024-09-26:
@meatpiHQ any chance of having a look at this?

meatpiHQ, 2024-09-27:
@cloudbr34k84 what was your odometer reading when you exported this log?

cloudbr34k84, 2024-09-27:
> @cloudbr34k84 what was your odometer reading when you exported this log?

Sorry no idea

cloudbr34k84, 2024-09-27:
@meatpiHQ 
[engine_odometer.txt](https://github.com/user-attachments/files/17157868/engine_odometer.txt)
[image]

meatpiHQ, 2024-09-27:
@cloudbr34k84 

Please try adding this to the custom PID first. And use mqtt explore to check

[image]

cloudbr34k84, 2024-09-27:
closing this off Thank you

closed 2024-09-27 by cloudbr34k84

cloudbr34k84, 2024-10-28:
@meatpiHQ  how do we go about setting up a profile for Colorado??