meatpiHQ/wican-fw #211: Encrytped mqtt?

lerra, 2024-09-21:
Hi, thanks for a cool product, the only thing that is holding me back is sending the data over encrypted mqtt, preferably with client certificate but only verifying server certificate is also ok or ev using the esphome encryption over http to use the home assistant support to fetch the information in a secure way. Any plans to enhance the security?

meatpiHQ, 2024-09-22:
@lerra secure mqtt will be supported on the [PRO version](https://www.crowdsupply.com/meatpi-electronics/wican-pro),  the firmware will need some optimization to get it working on the older version.

I'm also working on the http endpoint for the [official HA integration](https://github.com/jay-oswald/ha-wican) and will consider https implementation for the PRO version.

lerra, 2024-09-22:
Awesome, I will buy one when it is supported or in beta :)

jay-oswald, 2024-10-15:
@lerra we have the HA integration working as a beta BTW, the HA integration is over at https://github.com/jay-oswald/ha-wican

lerra, 2024-10-15:
Nice! :) the only downside is that it looks like it is using http endpoints but nice with an integration!

panlinux, 2024-10-15:
Can the wican hardware handle https?

meatpiHQ, 2024-10-15:
@panlinux Yes, it can support HTTPS, but the current firmware needs to be optimized to free up enough RAM for HTTPS usage.

This won't be an issue with the upcoming new hardware, WiCAN-PRO.