meatpiHQ/wican-fw #212: Kia Soul EV 2016 - AutoPID - MQTT "timeout, pid: 01A61"

jornodberg, 2024-09-24:
After giving up the last time I tried getting data from my car into Home Assistant, I now tried the newest firmware, and also uploaded the newest Vehicle Profiles.

Via MQTT Explorer i look at the Destination Topic, and it only recieves frames like:
{"error": "Timeout, pid: 010C1"}
{"error": "Timeout, pid: 012F1"}
{"error": "Timeout, pid: 01A61"}

Wican-settings:
AP+Station
CAN Bitrate: 500K
CAN Mode: Normal
Protocol: AutoPID
TX Topic: wican/KiaSoul/can/tx
RX Topic: wican/KiaSoul/can/rx
Status Topic: wican/KiaSoul/can/status
MQTT elm327 log: Disable

Vehicle Specific: Enable
Vehicle Model: AAA: Generic
Home Assistant Discovery: Enable
Grouping: Disable
Destination Topic: KiaSoul
Cycle Time(ms): 10000

I get a continous stream of data/frames to the topic wican/KiaSoul/can/rx

In the topic KiaSoul, i only get data on KiaSoul/error  , with many different PIDs.

What am I missing?

meatpiHQ, 2024-09-24:
@jornodberg this might sound silly, but is the car on? :)

jornodberg, 2024-09-24:
Not silly at all - I've read many of the issues here. ;-)

The car is on.

I found that choosing Vehicle Model: Hyundai: Ioniq2017 gave me SOC_DISPLAY in HomeAssistant, and the value seems correct. The other Ioniq-profile didn't give anything.

But that vehicle profile only contains SOC. I would love to also get Odometer, and preferably also some Temp-sensors.

But since that profile was working, maybe you would be able to help me get at least Odometer also?

Received my Wican in february 2023, and have never been able to get anything (correct) into HomeAssistant. And, yeah, I do realise that's because of my lack of knowledge... ;-) So this is progress!

jornodberg, 2024-09-24:
And also there seems to be some bug with getting the SOC-values. About every other value is empty. Here are the history from MQTT Explorer (in reverse) : 

09/24/2024 3:26:45 PM(-10.01 seconds) 
{"SOC_DISPLAY":97.5}
09/24/2024 3:26:35 PM(-10.01 seconds) 
{"SOC_DISPLAY":97.5}
09/24/2024 3:26:23 PM(-12.08 seconds) 
{}
09/24/2024 3:26:11 PM(-11.53 seconds) 
{"SOC_DISPLAY":98}
09/24/2024 3:26:03 PM(-8.47 seconds) 
{}
09/24/2024 3:25:51 PM(-11.55 seconds) 
{"SOC_DISPLAY":98}
09/24/2024 3:25:39 PM(-12.06 seconds) 
{}
09/24/2024 3:25:28 PM(-11.52 seconds) 
{"SOC_DISPLAY":98}
09/24/2024 3:25:18 PM(-9.5 seconds) 
{"SOC_DISPLAY":0}
09/24/2024 3:24:59 PM(-19.53 seconds) 
{}
09/24/2024 3:24:49 PM(-10.02 seconds) 
{}
09/24/2024 3:24:37 PM(-11.53 seconds) 
{"SOC_DISPLAY":98}

meatpiHQ, 2024-09-25:
@jornodberg 

> Is it correct that Wican also sends 4-5 messages each second to wican/KiaSoul/can/rx , even after enabling AutoPID-setup?

Yes it does, I will disable that by default and make it optional.