meatpiHQ/wican-fw #214: Automate stopped working (topic setting issue?)

happyjaxx, 2024-09-27:
Hi,

Might have a small issue trying to use Automate with 3.30

Uploaded the vehicles json for my Kia e-Soul (SK3 post 2019 64kWh) using the Ioniq/Kona profile, and, had to change to the AutoPid protocol (that wasn't obvious)...

Initially, it straight out worked, but the 1000ms cycle time* and the initial topic I set to wicanauto didn't suit me.
I changed it to 5000ms and set "wican/auto" to keep the same mqtt root topic as the general one.

And... it doesn't load anything anymore, the topic came back as "wicanauto", chewing the "/" away, and wican/can/rx isn't budging (can is online)

parameter parsing issue ?

*(I which you could set cycle_times per PID and even throttle updates per variable actually, Battery Power at 1 seconds would be great, but I don't need to receive BMS SoC more than every 10 seconds)

happyjaxx, 2024-09-27:
Shocker, it's working again, no clue how it came back to life...

There still is an issue with it not accepting a sub topic I suppose

meatpiHQ, 2024-10-01:
@happyjaxx are you able to reproduce this ? was it just a wifi connection glitch?

> *(I which you could set cycle_times per PID and even throttle updates per variable actually, Battery Power at 1 seconds would be great, but I don't need to receive BMS SoC more than every 10 seconds)

You can use the custom PID and set the cycle per PID

happyjaxx, 2024-10-01:
Well... so i put a subtopic as destination topic (goal being having all in one single branch)
[image]
Hit store...

Now is mostly unresponsive over web .... after a couple of minute, it lost 60 seconds of pings (watchdog?), came back (it did publish offline/online over mqtt though), web is barely responsive, like if it was struggling

```
Client null received PUBLISH (d0, q0, r0, m0, 'wican/can/status', ... (21 bytes))
wican/can/status {"status": "offline"}
Client null sending PINGREQ
Client null received PINGRESP
Client null received PUBLISH (d0, q0, r0, m0, 'wican/can/status', ... (20 bytes))
wican/can/status {"status": "online"}
Client null sending PINGREQ
Client null received PINGRESP
```

Started a climate session to get ECUs online, nothing started flowing besides a mere couple of rx's...

So: being patient (during the climate session) I let the gui load, changed back the topic to "darthmaul"... hit store... it did show after a (slow) refresh, but I had to wait for the wican to reboot on it's own (watchdog again), then all of a sudden data came in again.

Albiet, wonky values sometimes:
```
darthmaul {"SOC_BMS":0,"Max_REGEN":0,"Max_Power":0,"Batt_Current":6553.6,"HV_Volts":0,"Batt_MaxT":0,"Batt_MinT":0,"Batt_Temp_1":63,"Batt_Temp_2":80,"Batt_Temp_3":66,"Batt_Temp_4":104,"Batt_Temp_5":0,"Batt_InletT":100,"Max_Cell_V":4.16,"Max_Cell_V_No":0,"Min_Cell_V":0.06,"Min_Cell_V_No":232,"Aux_Batt_Volts":23.2,"SOH":100,"SOC_D":75.5}
darthmaul {"SOC_BMS":73.5,"Max_REGEN":162.3,"Max_Power":170,"Batt_Current":6552,"HV_Volts":384.2,"Batt_MaxT":22,"Batt_MinT":21,"Batt_Temp_1":22,"Batt_Temp_2":21,"Batt_Temp_3":21,"Batt_Temp_4":22,"Batt_Temp_5":0,"Batt_InletT":22,"Max_Cell_V":3.92,"Max_Cell_V_No":37,"Min_Cell_V":3.9,"Min_Cell_V_No":50,"Aux_Batt_Volts":14.7,"SOH":100,"SOC_D":75.5}
```

Sure I could work on the Wifi quality, but it should distort parsing CAN messages :) (I'll be setting up a local gateway inside the car, it already hosts the broker and is the WiCan's gateway, itself will have a better wireless client stack and will parse/store/send data whenever possible, but for now, it's sitting in the kitchen behind a thick wall) and it was flawless throughout yesterday and this morning

```
Station 54:32:04:90:de:ad (on wlx000f6008bxyz)
		signal:  	-81 dBm
	signal avg:	-79 dBm <- not great
```

meatpiHQ, 2024-10-08:
@happyjaxx  

> but it shouldn't distort parsing CAN messages

I think this could be fixed by adding a filter command to the pid_init in the profile.