meatpiHQ/wican-fw #218: Illegal MQTT Discovery Topic Error for 'Engine RPM' and 'Vehicle Speed' Sensors

cloudbr34k84, 2024-10-01:
Description:
Home Assistant logs an error indicating that the MQTT discovery messages for Engine RPM and Vehicle Speed sensors contain illegal characters in their discovery topics. As a result, these messages are not processed correctly.

```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:229
Integration: MQTT (documentation, issues)
First occurred: 12:47:12 pm (2 occurrences)
Last logged: 12:47:12 pm

Received message on illegal discovery topic 'homeassistant/sensor/5432048f5189/Engine RPM/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/5432048f5189/Vehicle Speed/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
```
**Steps to Reproduce:**
Configure MQTT discovery for sensors with spaces in their names, such as Engine RPM and Vehicle Speed.
Home Assistant attempts to process the discovery messages.
Observe the error logs indicating illegal discovery topics.
**Expected Behavior:**
The MQTT discovery messages should be processed correctly, and sensors should be created without errors.

**Actual Behavior:**
The discovery messages are rejected due to illegal characters (likely spaces in the topic), preventing the sensors from being recognized.

**Proposed Solution:**
Replace spaces or other illegal characters in the discovery topic with underscores or dashes.
Ensure that the sensor discovery topics comply with the MQTT discovery topic guidelines as outlined in [Home Assistant's MQTT documentation](https://www.home-assistant.io/integrations/mqtt/#discovery-topic).

meatpiHQ, 2024-10-01:
@cloudbr34k84 I suggest you update to v3.45 and use the new HA integration. The MQTT discovery will be deprecated.

https://github.com/jay-oswald/ha-wican

[wican-fw_obd_v345.zip](https://github.com/user-attachments/files/17206410/wican-fw_obd_v345.zip)

cloudbr34k84, 2024-10-01:
ok will do thanks

closed 2024-10-01 by cloudbr34k84