meatpiHQ/wican-fw #221: MQTT Discovery should set the "device" map so that all sensors get grouped in home assistant

seanauff, 2024-10-02:
See here for setting the device: https://www.home-assistant.io/integrations/sensor.mqtt/#device

Suggestions for the parameters:

- configuration_url: the IP for the wican in station mode
- connections: Wican MAC address
- hw_version: Wican HW version
- manufacturer: meatPi
- model: Wican OBD (or USB)
- name: could this be user set?

meatpiHQ, 2024-10-03:
@seanauff Thanks for the suggestion, but yes you're right the MQTT discovery will be deprecated soon.

To try the new integration, please update to version v3.45. Make sure to disable MQTT, as the integration now uses the HTTP endpoint. If you face any issues while updating to the latest firmware, make sure that MQTT is disabled before proceeding.

https://github.com/jay-oswald/ha-wican
[wican-fw_obd_v345.zip](https://github.com/user-attachments/files/17246469/wican-fw_obd_v345.zip)

seanauff, 2024-10-04:
I did the update to v3.45 and have it connected to HA via the integration! Looks like it doesn't yet pull in the configured sensors but those still go fine over mqtt for me

closed 2024-10-04 by seanauff