meatpiHQ/wican-fw #222: I need assistance understanding the new firmware's automation features.

alexsahka, 2024-10-05:
Is there a resource (link to help) to learn more about the new automation options? For example, I used to do this before:

[image]

How to use this with automation?

What's the purpose of using this? Does it automatically request data when the device is disconnected from Wi-Fi and then publish the latest values to MQTT?

Thank you.

meatpiHQ, 2024-10-08:
@alexsahka 

> How to use this with automation?

There are multiple ways you can use this feature, either by using the vehicle profile or by adding a custom PID 

> What's the purpose of using this? Does it automatically request data when the device is disconnected from Wi-Fi and then publish the latest values to MQTT?

Yes, it automatically requests data and publish it mqtt. It is also used by the new HA integration over HTTP.

What is your car make and model?

https://github.com/jay-oswald/ha-wican

alexsahka, 2024-10-08:
> 
> What is your car make and model?
> 
Honda Clarity 2018

meatpiHQ, 2024-10-09:
@alexsahka I found your old issue. If you can export the logs from Car Scanner, I will be able to create a profile for your car.

alexsahka, 2024-10-10:
ODO 92034 M  148114.37 km

{"payload":{"bus":"0","type":"tx","ts":50182,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[3,34,112,34,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":50186,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[16,57,98,112,34,236,0,0]}]}}
{"payload":{"bus":"0","type":"tx","ts":50188,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":50207,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,1,103,130,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50216,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50229,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[35,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50237,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[36,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50246,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[37,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50257,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[38,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50267,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[39,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50276,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[40,0,0,85,85,85,85,85]}]}}

{"payload":{"bus":"0","type":"tx","ts":50592,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[3,34,112,34,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":50597,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[16,57,98,112,34,236,0,0]}]}}
{"payload":{"bus":"0","type":"tx","ts":50599,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":50617,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,1,103,130,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50631,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50636,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[35,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50647,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[36,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50657,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[37,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50666,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[38,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50677,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[39,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":50687,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[40,0,0,85,85,85,85,85]}]}}

--------------

Fuel input level 100%

{"payload":{"bus":"0","type":"tx","ts":6180,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":6182,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":6287,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":6291,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":6453,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":6455,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8542,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":8544,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8604,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":8606,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8679,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":8681,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8741,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":8743,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8819,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":8821,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[3,65,47,255,85,85,85,85]}]}}
{"payload":{"bus":"0","type":"tx","ts":8911,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,47,170,170,170,170,170]}]}}
[... cut ...]
{"payload":{"bus":"0","type":"rx","ts":46374,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[32,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46375,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46376,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46377,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,0,0,85,85,85,85,85]}]}}

meatpiHQ, 2024-10-21:
@alexsahka Thanks, but I need the logs from car scanner please see the screenshot

[image]

alexsahka, 2024-11-07:
[FUEL log.txt](https://gith
[ODOM log.txt](https://github.com/user-attachments/files/17670457/ODOM.log.txt)
[SOC log.txt](https://github.com/user-attachments/files/17670458/SOC.log.txt)
[BATT CAPAS log.txt](https://github.com/user-attachments/files/17670454/BATT.CAPAS.log.txt)
ub.com/user-attachments/files/17670453/FUEL.log.txt)

meatpiHQ, 2024-11-10:
@alexsahka I've created a new profile for Honda Clarity 2018, please try and let me know if it works.

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup#integration-device-setup

alexsahka, 2024-11-10:
There is only SOC and here is the status:

MQTT:
{"error": "Timeout, pid: 22202A8"}

Homeassistant: 
WiCAN SoC = 0
[image]

alexsahka, 2024-12-10:
> @alexsahka I've created a new profile for Honda Clarity 2018, please try and let me know if it works.
> 
> https://meatpihq.github.io/wican-fw/home-assistant/integration_setup#integration-device-setup

Could you please look into the issue with the SOC? It appears to be reporting an inaccurate value of zero, even though the (SOC) is 99%. Also, a single PID (SOC) was created for the Honda Clarity. Thanks.

meatpiHQ, 2024-12-13:
@alexsahka sorry for the delay im looking at the logs again, its really hard to guess. Can you please create another log for the SoC at a different SoC level. The more data points you have the easier it will be.

Just an idea what we're dealing with, there is 36 frames ... most cars respond with much less than that.

[image]

alexsahka, 2024-12-15:
[37 SOC log.txt](https://github.com/user-attachments/files/18138278/37.SOC.log.txt)

SOC: 37%

Thanks!

[image]

meatpiHQ, 2024-12-29:
@alexsahka I've update the profile, please sync and try again.

I verified the profile based on this forum, there is a lot of PIDs for the Clarity, let me know if you want to added anything else.

https://www.insideevsforum.com/community/index.php?threads/budget-battery-capacity-readout.10531/

[Car Scanner Clarity PIDs File](https://github.com/user-attachments/files/18269858/CarScanner.Release.1.zip)

alexsahka, 2025-01-02:
> @alexsahka I've update the profile, please sync and try again.
>
> I verified the profile based on this forum, there is a lot of PIDs for the Clarity, let me know if you want to added anything else. [...]

Unfortunately, with the new SOC profile and PID numbers jumping and ranging from 0 to 255, I'm unsure how to proceed. I don't understand how to generate the profile file.
[image]
[image]

alexsahka, 2025-01-02:
> @alexsahka I've update the profile, please sync and try again.
>
> I verified the profile based on this forum, there is a lot of PIDs for the Clarity, let me know if you want to added anything else. [...]

To receive proper SOC Im sending from Nodered: 
msg.payload = { "bus": "0", "type": "tx", "ts": 49627, "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": true, "data": [3, 34, 32, 42, 170, 170, 170, 170] }] };
return msg;

than after 150ms second message:

msg.payload = { "bus": "0", "type": "tx", "ts": 49632, "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": true, "data": [48,0,10,170,170,170,170,170]}] };
return msg;

to wican/xxxxxx/can/tx

and Receiving message from wican/xxxxxx/can/rx/#

and converting with function node:

var obj = {};

for (let i = 0; i < msg.payload.frame.length; i++) {
    // Check frame id and data
    if (msg.payload.frame[i].id == 417001729 && msg.payload.frame[i].data[0] == 39 && msg.payload.frame[i].data[1] == 129) {
        obj.soc = msg.payload.frame[i].data[6];
        msg.payload = obj;
        return msg;
    }
}

meatpiHQ, 2025-01-02:
@alexsahka I've update the profile, please try again and let me know if it works.

> I don't understand how to generate the profile file.

It's based on the elm327 instructions from the car scanner logs.

alexsahka, 2025-01-03:
> @alexsahka I've update the profile, please try again and let me know if it works.
> 
> > I don't understand how to generate the profile file.
> 
> It's based on the elm327 instructions from the car scanner logs.

Thank you for your work, but unfortunately, the updated profile is not working at all. There is no output of any kind of number.

meatpiHQ, 2025-01-05:
@alexsahka Please update to v4.02, the issue was that the buffer was too small, i increased the buffer size and tested on the emulator. It's working fine now.

[wican-fw_obd_v402.zip](https://github.com/user-attachments/files/18310669/wican-fw_obd_v402.zip)

alexsahka, 2025-01-06:
> Thank you

The problem persists even after the update. No data is being published to the MQTT topic.

meatpiHQ, 2025-01-06:
@alexsahka can you please download your device configuration file from the system tab. Make sure to remove any private data like ssid and passwords from the downloaded JSON file.

alexsahka, 2025-01-06:
> @alexsahka can you please download your device configuration file from the system tab. Make sure to remove any private data like ssid and passwords from the downloaded JSON file.

[config_2025-01-06.json](https://github.com/user-attachments/files/18313339/config_2025-01-06.json)

meatpiHQ, 2025-01-06:
@alexsahka Please sync the profile and try again.

alexsahka, 2025-01-06:
> @alexsahka Please sync the profile and try again.

Thank you. It seems to be working now, but I will observe its behavior with different SOC% values later. Is it possible to add more measurements? I previously posted some measurements here:
https://github.com/meatpiHQ/wican-fw/issues/222#issuecomment-2463394538

meatpiHQ, 2025-01-09:
@alexsahka i've added the odometer, you can sync and test. The Fuel is not very clear can you please repeat the log 

https://meatpihq.github.io/wican-fw/car_scanner/logs

alexsahka, 2025-01-10:
Thank you; the odometer is working now!
I will repeat the log for the fuel and post it later.

> @alexsahka i've added the odometer, you can sync and test. The Fuel is not very clear can you please repeat the log
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

alexsahka, 2025-01-10:
> @alexsahka i've added the odometer, you can sync and test. The Fuel is not very clear can you please repeat the log
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

I apologize, I forgot to ask you about the Home Assistant card. I can see the SOC in the Home Assistant card, but not the odometer. What steps are necessary to display the odometer on the card as well? Thank you!

meatpiHQ, 2025-01-11:
try removing the device and add it back again.

alexsahka, 2025-01-12:
> try removing the device and add it back again.

Thanks!

alexsahka, 2025-01-14:
> @alexsahka i've added the odometer, you can sync and test. The Fuel is not very clear can you please repeat the log
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

Here is the 2  fuel logs 
6.41 US Gallons
and
[Fuel log 1.29 Gallons.txt](https://github.com/user-attachments/files/18405675/Fuel.log.1.29.Gallons.txt)
[Fuel log 6.41 Gallons.txt](https://github.com/user-attachments/files/18405676/Fuel.log.6.41.Gallons.txt)

1.29 US Gallons

alexsahka, 2025-02-05:
> [@alexsahka](https://github.com/alexsahka) i've added the odometer, you can sync and test. The Fuel is not very clear can you please repeat the log
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

I've added 2 fuel logs above. Could you please review and assist me with this fuel log? Thanks