meatpiHQ/wican-fw #234: ID3 profile not working on Skoda Enyiaq iV 60

marten-lucas, 2024-10-17:
Hi,

I tried the vehicle profile for the ID.3 on my Skoda Enyaq iv but it did not work. Both are MEB-Platform so they should work.
I confirmed the setting by switching to a vw up, which did work.

I did find some PIDs on the internet
https://www.goingelectric.de/forum/viewtopic.php?t=60894&start=10
and here:
https://github.com/nickn17/evDash/blob/master/src/CarVWID3.cpp
carscanner works as well (see exported pids attached)
[custompids.txt](https://github.com/user-attachments/files/17421273/custompids.txt)

How can I help to improve the profile?

meatpiHQ, 2024-10-20:
@marten-lucas Which firmware version are you on? Can you post a screenshot of your device configuration settigns page and automate page?

marten-lucas, 2024-10-21:
[image]

[image]

[image]

meatpiHQ, 2024-10-21:
Try to disable the MQTT and change the cycle time to 5000ms

cr0i, 2024-10-21:
Hello @meatpiHQ ,
I think, like in my case, the init-string and the pid_init does not work in the Profile.
The ATSP6 or ATSP7 must be in the pid_init, not globally.
In my case with the custom config, it worked best, if the full init string is now in the init field, so it should be in the pid_init perhaps?

meatpiHQ, 2024-10-22:
@cr0i are you able to add these to the profile and test? If it's successful please create a PR

meatpiHQ, 2024-10-23:
@cr0i Sorry I might have miss understood.

The pid_init  should be working in the latest firmware version, it was tested on other cars.

> but I don´t know anything how to create PR.

Don't worry about PR I can add the changes once you confirm that it's working. I wanted you to do the PR so you'd be added as contributor to this project :)

marten-lucas, 2024-10-25:
@meatpiHQ donyou mean confirmation if it works with custom pids in the latest firmware? I can do this next week.

@cr0i : or did you already try it?

cr0i, 2024-10-25:
Hello @marten-lucas ,

with Custom PIDs it´s working, I use in my solution, because I need different MQTT Topics per PID.
It´ll look like this, Type and Sent_to will be different at you:
[image]

I´ll try the vehicle-Profile again and give you feedback.
I think ID3 (My Cupra Born) and ID4 (Your Enyiaq) will be identically.

cr0i, 2024-10-25:
Hello @meatpiHQ ,

I reduced the vehicle-Profile to only one PID with pid_init:
`{
            "car_model": "VW: id-3 Test",
            "init": "ATST96;ATFCSD300000;ATFCSM1;",
            "pids": [
                {
                    "parameters": [
                        {
                            "class": "battery",
                            "expression": "((B4*51)/115)-6.4",
                            "name": "SOC",
                            "pid_init": "ATSP7;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;",
                            "unit": "%"
                        }
                    ],
                    "pid": "22028C1"
                }
            ]
        }`

with this I get only timeout.

But when I put the full init-String in init and don´t use pid_init it works (but only for one PID, because we need different Init-Strings:

`{
            "car_model": "VW: id-3 Test",
            "init": "ATST96;ATFCSD300000;ATFCSM1;ATSP7;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;",
            "pids": [
                {
                    "parameters": [
                        {
                            "class": "battery",
                            "expression": "((B4*51)/115)-6.4",
                            "name": "SOC",
                            "unit": "%"
                        }
                    ],
                    "pid": "22028C1"
                }
            ]
        }`

Answer:
others/wican/test {"SOC":58.35}

So I think pid_init is not working (with me).
I need only Custom PIDs, so I´m not worry in the moment about this,but I can go on testing, if something is to do.

I have to remove my working config for this every time. If you could add Backup of settings in the future, this would be nice?

I´m using Firmware 3.48.

meatpiHQ, 2024-10-27:
@cr0i I you're intersted I suggest that you read about the ELM327 AT commands so you'd understand what do these commands do. For some PIDs you have change the protocol from `ATSP6 `which is 500K 11bit to `ATSP7` 500K 29bit and the set the receive filter. Check this document. https://www.elmelectronics.com/DSheets/ELM327DSH.pdf

What if you put the same string in `init `and `pid_init`? Does it work?

cr0i, 2024-10-27:
I'm interested how it works, but I accepted, that I can't understand the whole world.; -)
Some of this commands I think I understood. Setting beetween 11Bit and 29Bit and setting Request and Response filters, but I don't exactly the reason behind, for this I don't have in deepth protocol knowledge or programming. 

But I thought, vehicle Profiles and Custom PIDs work similar:
Custom Initialisation String is send only once and then the Init Field before each PID. 
In Vehicle Profiles init is send once and pid_init before each PID. 
But it doesn't work so.

I'll try the same string in pid_init later and give you feedback.

cr0i, 2024-10-27:
Hello @meatpiHQ ,

When I put the same init String in init and pid_init it works, but only one value is correct. The other values are calculated wrong or get timeouts:
{"SOC":60.57,"Range":0,"Odometer":9895936}
(SOC OK, Range and odometer wrong)
So I think pid_init is not send like in Custom PIDs?
The JSON is like this (I tried 3 times with the correspondending init):
`{
            "car_model": "VW: id-3 Test",
            "init": "ATST96;ATFCSD300000;ATFCSM1;ATSP7;ATSHFC0076;ATCP17;ATCRA17FE0076;ATFCSH17FC0076;",
            "pids": [
                {
                    "parameters": [
                        {
                            "class": "battery",
                            "expression": "((B4*51)/115)-6.4",
                            "name": "SOC",
                            "pid_init": "ATST96;ATFCSD300000;ATFCSM1;ATSP7;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;",
                            "unit": "%"
                        }
                    ],
                    "pid": "22028C1"
                },
                {
                    "parameters": [
                        {
                            "class": "none",
                            "expression": "(B5*256)+B6",
                            "name": "Range",
                            "pid_init": "ATST96;ATFCSD300000;ATFCSM1;ATSP6;ATSH710;ATCP18;ATCRA77A;ATFCSH710;",
                            "unit": "km"
                        }
                    ],
                    "pid": "222AB62"
                },
                {
                    "parameters": [
                        {
                            "class": "none",
                            "expression": "[B4:B6]",
                            "name": "Odometer",
                            "pid_init": "ATST96;ATFCSD300000;ATFCSM1;ATSP7;ATSHFC0076;ATCP17;ATCRA17FE0076;ATFCSH17FC0076;",
                            "unit": "km"
                        }
                    ],
                    "pid": "22295A"
                }
            ]
        }`

meatpiHQ, 2024-10-28:
@cr0i Please try this profile.

> So I think pid_init is not send like in Custom PIDs?

Yes you are right seems there is an issue with pid_init .

```
{
   "car_model":"VW: id-3 Test",
   "init":"ATST96;ATFCSD300000;ATFCSM1;",
   "pids":[
      {
         "parameters":[
            {
               "class":"battery",
               "expression":"((B4*51)/115)-6.4",
               "name":"SOC",
               "pid_init":"ATAR;ATCP18;ATSHDB33F1;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;ATFCSM1;",
               "unit":"%"
            }
         ],
         "pid":"22028C1"
      },
      {
         "parameters":[
            {
               "class":"none",
               "expression":"(B5*256)+B6",
               "name":"Range",
               "pid_init":"ATSP6;ATSH710;ATFCSH710;ATFCSM1;ATCRA77A;",
               "unit":"km"
            }
         ],
         "pid":"222AB62"
      },
      {
         "parameters":[
            {
               "class":"none",
               "expression":"[B4:B6]",
               "name":"Odometer",
               "pid_init":"ATSP7;ATSHFC0076;ATCP17;ATCRA17FE0076;ATFCSH17FC0076;ATFCSM1;",
               "unit":"km"
            }
         ],
         "pid":"22295A"
      }
   ]
}
```

cr0i, 2024-10-30:
Hello @meatpiHQ ,

with this Profile I get timeouts on all 3 values, i.e.:
{"error": "Timeout, pid: 22295A"}

jeroen-github, 2024-11-06:
I also bought the Wican for my Enyaq 60, I have the same issues, as the topic starter, can someone from meatpi give us some more directions to get this working ?

meatpiHQ, 2024-11-10:
@cr0i @jeroen-github please try this profile, i kept the SoC for now, we can add them one by one and see what breaks.

You can also directly sync the profile directly from the webpage

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup#integration-device-setup

```
{
  "car_model": "VW: id-3",
  "init":"ATST96;",
  "pids":[
    {
       "parameters":[
          {
             "class":"battery",
             "expression":"((B4*51)/115)-6.4",
             "name":"SOC",
             "pid_init":"ATSP7;ATAR;ATCP18;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;ATFCSD300000;ATFCSM1;",
             "unit":"%"
          }
       ],
       "pid":"22028C1"
    }
 ]
}
```

cr0i, 2024-11-10:
With this profile I only get timeouts:
{"error": "Timeout, pid: 22028C1"}

In Monitor Tab there are no frames captured.

meatpiHQ, 2024-11-10:
@cr0i I think I figured it out this time, the issue was that the pid_init was inside the parameters and was not being parsed 

```
{
  "car_model":"VW: id-3",
  "init":"ATST96;",
  "pids":[
     {
        "pid":"22028C1",
        "pid_init":"ATSP7;ATAR;ATCP18;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;ATFCSD300000;ATFCSM1;",
        "parameters":[
           {
              "name":"SoC",
              "expression":"((B4*51)/115)-6.4",
              "unit":"%",
              "class":"battery"
           }
        ]
     }
  ]
}
```

cr0i, 2024-11-10:
Hello @meatpiHQ ,

strike, now it works. ;-)
I´ve tried the other value too:
{"SOC":36.17,"Range":109,"Odometer":22865}

`   "car_model":"VW: id-3 Test",
   "init":"ATST96;ATFCSD300000;ATFCSM1;",
   "pids":[
      {
         "parameters":[
            {
               "class":"battery",
               "expression":"((B4*51)/115)-6.4",
               "name":"SOC",
               "unit":"%"
            }
         ],
         "pid":"22028C1",
         "pid_init":"ATAR;ATCP18;ATSHDB33F1;ATSHFC007B;ATCP17;ATCRA17FE007B;ATFCSH17FC007B;ATFCSM1;"
      },
      {
         "parameters":[
            {
               "class":"none",
               "expression":"(B5*256)+B6",
               "name":"Range",
               "unit":"km"
            }
         ],
         "pid":"222AB62",
         "pid_init":"ATSP6;ATSH710;ATFCSH710;ATFCSM1;ATCRA77A;"
      },
      {
         "parameters":[
            {
               "class":"none",
               "expression":"[B4:B6]",
               "name":"Odometer",
               "unit":"km"
            }
         ],
         "pid":"22295A",
         "pid_init":"ATSP7;ATSHFC0076;ATCP17;ATCRA17FE0076;ATFCSH17FC0076;ATFCSM1;"
      }
   ]
}`

meatpiHQ, 2024-11-11:
@cr0i Great! Do you want to make a pull request to update the [profile](https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles/vw/id-3.json) ?

jeroen-github, 2024-11-11:
Thx for the update, I will try it wednesday, currently traveling for work.

jeroen-github, 2024-11-13:
Hi just confirmed its working also over here, with MQTT explorer I can see that I receive the correct values.
{
  "soc": 68.15,
  "range": 218,
  "odometer": 13223,
  "capacity": 53.85
}

So now its up to us to add more info to the cars profile.

marten-lucas, 2024-11-16:
Great i do get the values as well.
But everytime I requests the values the alarm goes off. 
@jeroen-github  or @cr0: Is it the same for you? What is you strategy to avoid it?

cr0i, 2024-11-16:
> Great i do get the values as well. But everytime it requests the values the alarm goes off. @jeroen-github or @cr0: Is it the same for you? What is you strategy to avoid it?

Hello @marten-lucas ,

you have 3 options to avoid alarm:
1. I have an openWB Wallbox, there is an manual_soc Module. So it´s working for this, if the SoC is only once transmitted to the Wallbox when I drive home. With an OBD Kable you can change PIN1 und PIN16, so the WiCAN only has power, when ignition is on.
https://forum.openwb.de/viewtopic.php?p=115467#p115467

2. You can remove the fuse for Alarm Signal, so the car is only blinking when alarm is on.

3. With ODB11 and an active VAG Subscription it could be possible to disable OBD-Monitoring. With my 2023 Cupra Born it worked, but I don´t know, if it is possible with newer cars with SFD2 already.
https://www.born-forum.de/forum/thread/193-m%C3%B6gliche-codierungen-am-cupra-born-ohne-gew%C3%A4hr-auf-eigene-gefahr/?postID=77395#post77395

marten-lucas, 2024-11-19:
Thank you for the suggestion.

I had option 1 running with evcc and mqtt. and would like to replicate it.

@jay-oswald or @meatpiHQ: does the "/autopid_data" endpoint also trigger the data retrieval? if not is there an endpoint that can be triggered for that? my idea is to set the interval to 0 (or 99999999..) and only trigger via a home assistant automation if the device comes online or the status voltage indicate loading or start of the vehicle.

Do you think it will work like this?

jay-oswald, 2024-11-19:
@marten-lucas the HA integration currently runs every 30 seconds. I think that interval is only used for MQTT, where the WiCAN pushes the data, instead of pulls like HA does. We have plans to make HA polling happen more frequently at some point as well, but want some smarts around it, so HA isn't trying to reach a wican thats off every 5 seconds spamming the network requests.

I think my WiCAN is set to 10 seconds for the interval, with MQTT disabled, and it updates every 30 secs in HA reliably when its online, so I'd just go with that, and let the FW and HA integration take it from there

cr0i, 2024-11-23:
> Hi just confirmed its working also over here, with MQTT explorer I can see that I receive the correct values. { "soc": 68.15, "range": 218, "odometer": 13223, "capacity": 53.85 }
> 
> So now its up to us to add more info to the cars profile.

Which infos are interesting for you?
In ABRP there are some PIDs listed, I think they could be used. Only the Init Strings must be found and the Expressoin has to be "translated":

batt_temp | current | ext_temp | is_charging | is_dcfc | is_parked | kwh_charged | odometer | soc | vehicle_reported_speed | voltage
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
222A0B | 221E3D | 222609 | 227448 | 227448 | 22210E | 221E32 | 22295A | 22028C | 22F40D | 221E3B
A/2-40 | (((A*16777216+B*65536+C*256+D)-150000)/100)*-1 | A/2-50 | {A:2} | ({A:1}&&{A:2}) | B==8 | INT32(I:J:K:L)/8583.07123641215 | INT24(A:B:C) | A*0.4425-6.1947 | A | ((A*256)+B)/4