meatpiHQ/wican-fw #235: More Sensors vor e-up

marten-lucas, 2024-10-17:
Hi,

using the mqtt method I got more sensors out of the WiCan then the profile has.
I have:
- Range
- Charge State
- Plug State
- Remaining Charge Time
- Outside Temp

How can I contribute to the profile or test it locally first?

meatpiHQ, 2024-10-20:
@marten-lucas you can post the mqtt requests and response that you're sending here and I can try and help you add them to the profile

marten-lucas, 2024-10-21:
Here is the script for the requests (mqtt needed alls data in dec):

` alias: Seat OBD2 Request
sequence:
  - alias: Request Outside Temp
    data:
      qos: "0"
      retain: false
      topic: wican/1091a817d1b9/can/tx
      payload: |-
        { 
          "bus": "0", 
          "type": "tx", 
          "frame": [{ 
            "id": 1812, 
            "dlc": 8, 
            "rtr": false, 
            "extd": false, 
            "data": 
              [3, 34, 16, 20, 170, 170, 170, 170] 
          }] 
        }
    action: mqtt.publish
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 500
  - alias: Request SoC
    data:
      qos: "0"
      retain: false
      topic: wican/1091a817d1b9/can/tx
      payload: |-
        { 
          "bus": "0", 
          "type": "tx", 
          "frame": [{ 
            "id": 1893, 
            "dlc": 8, 
            "rtr": false, 
            "extd": false, 
            "data": 
            [3, 34, 29, 208, 170, 170, 170, 170] 
          }] 
        }
    action: mqtt.publish
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 500
  - alias: Request Range
    data:
      qos: "0"
      retain: false
      topic: wican/1091a817d1b9/can/tx
      payload: |-
        { 
          "bus": "0", 
          "type": "tx", 
          "frame": [{ 
            "id": 1812, 
            "dlc": 8, 
            "rtr": false, 
            "extd": false, 
            "data": 
            [3, 34, 34, 224, 170, 170, 170, 170] 
          }] 
        }
[... 200 lines cut ...]
description: ""
triggers:
  - topic: wican/1091a817d1b9/can/rx
    trigger: mqtt
conditions:
  - condition: template
    value_template: "{{ trigger.payload_json.frame[0].id == 2024 }}"
  - condition: template
    value_template: "{{ trigger.payload_json.frame[0].data[2] == 22 }}"
  - condition: template
    value_template: "{{ trigger.payload_json.frame[0].data[3] == 70 }}"
actions:
  - variables:
      plug_state: "{{ trigger.payload_json.frame[0].data[4] == 1 }}"
  - data:
      new_value: "{{ plug_state }}"
    action: esphome.wallbox_controller_set_seat_plugstate
mode: single
`
the plug state did not work as expected:
here is the definition from http://obd-amigos.linuxtech.net/files/amigos_PIDs.pm :
`       plg_ins=>{cmd=>'22 16 46', cu=>'01', desc=>'charging plug inserted',       unit=>'',          d=>1,  formula=>'sprintf("%s", V1==0?"NO":V1==1?"YES":"V1")'},                      # verified correct

    '01'=>{hdr=>'7E0', rhdr=>'7E8', fc=>0,                 desc=>'Engine'},                    # +0x08 <
`

cr0i, 2024-11-15:
pid_init is now working on VW: ID.