meatpiHQ/wican-fw #240: Renault Twizy EV Profile

lucabacchipg, 2024-10-19:
Hello, thas is possibile to have a profile for Renault Twizy ?
This will be fantastic :)
-Soc %
-Range KM
-Charge State
-Plug State 
[Twizy-CanBus.pdf](https://github.com/user-attachments/files/17445794/Twizy-CanBus.pdf)

meatpiHQ, 2024-10-20:
@lucabacchipg while monitoring in car scanner click on the SoC sensor let it log for few seconds then immediately unplug WiCAN from the OBD port. So we can easily find the SoC logs at the end of the file. Also note the last SoC value from car scanner.

lucabacchipg, 2024-10-20:
[log.txt](https://github.com/user-attachments/files/17450522/log.txt)
Done, value of soc is about 93.22 %
should be bit 5+6 converted into decimal and divide by 400

ps. you can look here where use a ATZ command to receive reply
( you can also look at this link https://www.forumelettrico.it/forum/post438519.html where are explained by using 2 image) 
[image]
[image]

meatpiHQ, 2024-10-22:
@lucabacchipg I've created the profile, please follow the instructions in the link below. 

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup

lucabacchipg, 2024-10-23:
That's work perfect. IF possible need to load other value to read

lucabacchipg, 2024-10-27:
> @lucabacchipg I've created the profile, please follow the instructions in the link below.
> 
> https://meatpihq.github.io/wican-fw/home-assistant/integration_setup

Hello, please can you add also these value to integration to Tw
[log.txt](https://github.com/user-attachments/files/17534080/log.txt)
izy ? 
I attach a log of sequentiale read done:
Module 1 Battery Temperature  23c
Module 2 Battery Temperature   23c
Module 3 Battery Temperature   23c
Module 4 Battery Temperature   23c
Module 5 Battery Temperature   23c
Module 6 Battery Temperature   23c
Module 7 Battery Temperature 
[log.txt](https://github.com/user-attachments/files/17534101/log.txt)
  22c

Battery Milleage 67990 KM

Energy BMS 3.7 KW

Traction Battery Compartment Temperature 22C

meatpiHQ, 2024-10-27:
@lucabacchipg please provide a log for each sensor. You should press on the sensor in car scanner, let it run for few seconds then immediately unplug WiCAN so the sensor log would be at the end of the file. And it would be better if you have multiple logs at different values so we can confirm the result.