meatpiHQ/wican-fw #248: how to enable Home Assistant Discovery?

tung256, 2024-10-24:
i got v3.48 for my obd wican
[image]
Enable HA Discovery is greyed out. i cannot select Enable
[image]

these are my settings
[image]
[image]

in HA, i dont see any new devices
[image]

what i really want is to have WiCAN send me a notification if any door is opened as seen in this video  https://www.youtube.com/watch?v=oWluBx78IR4
would that ever be possible?

meatpiHQ, 2024-10-25:
Hey @tung256 so v3.48 works really well with the new HA integration, MQTT discovery is not supported in this version. MQTT itself is still supported without the discovery. If you decide to use the new integration you should disable MQTT. And reduce the cycle time to 10000 (10 sec) you can change the cycle time later. The new integration detects the sensors automatically, and its using HTTP instead of MQTT.

If you decide to keep using MQTT you have to disable ```MQTT ELM327 logging ``` and keep that disabled. And you should add the sensors manually.

So if I understand correctly you want to check the car door is opened? Do you if you car broadcasts the car door state in the CAN bus, if so what is the CAN ID of the message. 

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup

tung256, 2024-11-02:
@meatpiHQ 
mqtt is now disabled
[image]
HA Discovery is greyed out, i cannot change to Enable:
[image]

meatpiHQ, 2024-11-10:
@tung256  The integration will automatically find the sensors you don't need to enable the discovery. 

The discovery is disabled in this firmware version.