meatpiHQ/wican-fw #256: 2023 Chevy Bolt - Trying to link and get SoC values

SAUhrich, 2024-11-10:
I just last week received my WiCAN-ODB and have only gotten as far as connecting it to the local lan, setup and configured with Car Scanner and captured the ELM logs. I've also got the entities created in Home Assistant.

I've used the Automation tab and set my vehicle as "Opel Ampere-E" as it is closer than anything else in the predefined list. This will successfully connect and give good readings all the way through: WiCAN->WIFI->Home Assistant-HA Entities, here is an image of the device in HA:
[image]

The primary purpose of this as a project is to get live State of Charge (SoC) values into Home Assistant. I am off grid with solar and a decent size battery bank. This will allow me to trigger my chargers charging cycle and adjust the current based on the cars SoC relative to the house batteries SoC over a given timeframe.

I have followed the instructions in the documentation under "New Vehicle Profiles" and captured the Car Scanner log output reading the SoC for a while right before unplugging the WiCAN [(attached).

I am ok at various syntaxes (yaml, xml, etc.) but I am pretty lost when it comes to interpreting PIDs and initialization codes :)

Looking for any guidance on how to fill in the init codes and special PID's to retrieve the SoC:

[image]

Thank you very much for any assistance!

[2023ChevyBolt-SOC_focus_ELM_Logs.txt](https://github.com/user-attachments/files/17693349/2023ChevyBolt-SOC_focus_ELM_Logs.txt)

SAUhrich, 2024-11-12:
Welp, it is now working! I think that it must have been updates to the Opel Ampere-E vehicle profile settings, thank you @meatpiHQ. 

I am getting the SoC value passed from the WiCAN-ODB through WiFi and into Home Assistant using auto_pid mode and not MQTT. Here is a view from within Home Assistant:
[image]

I will continue to experiment, but so far this gives exactly the behavior I was looking for: The ECU will give continuous readings, even if the car is off as long as the car is at home/connected to my LAN. It is therefore giving live readings during charging and I can balance my electrical loads between the car battery charging and the home/solar battery charging. 

If it is helpful for anyone else with a Chevy Bolt or Opel here are my config screens in the WiCAN config pages:

[image]

[image]

I left the values I was experimenting with in the Custom PIDs section of the Automate page of the configuration, but I think those are only MQTT communications (?), not sure. Either way, I'm leaving them in the config as it is working!

closed 2024-11-12 by SAUhrich

meatpiHQ, 2024-11-12:
@SAUhrich That's awesome thanks for the update!

meatpiHQ, 2024-11-14:
@brjhaverkamp please feel free to submit a pull request to separate the two profiles

Elibart-home, 2025-05-16:
Many thanks for your labor! I have now succesfully read the SOC of my 2020 Opel Ampera-e.

The key is the init-code. Just a dump for the next lucky one getting it working:

```
  "auto_pid_car_data": {
    "cars": [
      {
        "car_model": "Opel:Ampere-E",
        "init": "ATSP6;ATST96;ATSH7E0;ATSH7E1;ATSH7E4;",
        "pids": [
          {
            "pid": "2283341",
            "pid_init": "",
            "parameters": [
              {
                "name": "SOC",
                "expression": "(B4*100)/255",
                "unit": "%",
                "class": "battery",
                "period": "5000",
                "min": "",
                "max": "",
                "type": "MQTT_Topic",
                "send_to": "wican/soc"
              }
            ]
          }
        ]
      }
    ]
  },

```

meatpiHQ, 2025-05-19:
@Elibart-home the Opel Ampera-e is already supported, the current profile did not work for you?

Elibart-home, 2025-05-19:
@meatpiHQ That's right, the existing profile didn't work for me. The values were empty.

meatpiHQ, 2025-05-19:
@Elibart-home do you want to create a PR so you can take credit for your contribution or would you like me to add these in?

Elibart-home, 2025-05-20:
@meatpiHQ I would like you to add these in :-)

meatpiHQ, 2025-05-20:
@Elibart-home Done. Please try the profile and let me know if it works.

Elibart-home, 2025-05-20:
@meatpiHQ Done. SoC works fine. HV_CAPACITY gives an unknown result, my 2019 Ampera-e has another PID: 2245F91. If I change that, it also works fine.

Elibart-home, 2025-05-23:
For the record: working values are:

```
      {
        "car_model": "Opel: Ampere-E",
        "init": "",
        "pids": [
          {
            "pid": "2283341",
            "pid_init": "ATSP6;ATST96;ATSH7E0;ATSH7E1;ATSH7E4;",
            "parameters": [
              {
                "name": "SOC",
                "expression": "(B4*100)/255",
                "unit": "%",
                "class": "battery",
                "period": "5000",
                "min": "0",
                "max": "100",
                "type": "MQTT_Topic",
                "send_to": "wican/e4b323d12a35/soc"
              }
            ]
          },
          {
            "pid": "2245F91",
            "pid_init": "ATSP6;ATSH7E4;",
            "parameters": [
              {
                "name": "HV_CAPACITY",
                "expression": "[B4:B5]*0.0032",
                "unit": "kWh",
                "class": "battery",
                "period": "5000",
                "min": "",
                "max": "",
                "type": "Default",
                "send_to": "wican/e4b323d12a35/cap"
              }
            ]
          }
        ]
      }

```