meatpiHQ/wican-fw #259: Sleep with AutoPID enabled not working.

smurf12345, 2024-11-13:
Hi All:
I just moved my WiCAN from using MQTT to AutoPID last week.  I have noticed now, that the sleep function may not be working.  Or maybe with autopid it polling more than I had MQTT set at.  I noticed when I walked out in the garage, my dashboard was lit up.  After watching it for a while, it would try to go to sleep, but then withing 5-30 seconds it would wake back up repeating this cycle. As soon as I removed the WiCAN it stopped doing this.  Previously using MQTT for more than a year, I have not seen this behavior.   I'm running v3.48.  Only changes is I'm on newer code and running Auto PID Mode.  I tried increasing the cycle time setting thinking it would poll less and car would go to sleep, but that doesn't seem to help.  Let me know if there is any logs I can provide to help troubleshoot.

Thanks,
Patrick

meatpiHQ, 2024-11-14:
@smurf12345 I had some else report similar issue, but i was never able to reproduce. What is the sleep voltage in the settings?

smurf12345, 2024-11-14:
Sorry, I'm using wrong term for sleep.  The WiCAN does go to sleep once it drops down under the set threshold.  I did some more testing.  The dashboard wakes up when the ECU is polled even when not using AutoPID mode (switched back to elm327/mqtt).  I don't think I ever noticed it because I used a much slower poll rate when using MQTT and once the ECU went to sleep (it would go to sleep even during charging), I had to poll using a wake up { "bus": "0", "type": "tx", "frame": [{ "id": 2015, "dlc": 8, "rtr": false, "extd": false, "data": [2, 1, 0, 170, 170, 170, 170, 170] }] }; which only used sparingly.  With AutoPid mode it doesn't seem to need any type of wakeup (or sends that on every poll).   While testing, I changed the cycle time parameter from 60,000 to 300,000ms and the dashboard waking up matched that exactly.

So, it looks like that is normal behavior for my car.  I'll close this issue out as it is working as expected.

meatpiHQ, 2024-11-14:
@smurf12345 thanks for the update. So the solution was to increase the cycle time to 5 min. 

I suggest you keep this issue open, I'd like to send you a new firmware update, with periodic wakeup. This would help reduce the battery drain.

smurf12345, 2024-11-14:
5 minutes I picked randomly on my latest test.  At 1st, I had 10second intervals.  Changing to 1 minute was too low as I think that is same as the timeout for the dash going dark/sleep, so it looked like an endless cycle of dashboard going on and off.  5 minutes is probably even more granular than I need. 

I'm happy to test updated firmware if you think there is still an issue.  Currently I'm using the default 13.1v for the sleep.  After watching the voltage on my car, I could probably up to 13.6 to get the WiCAN to go to sleep sooner.  Once it is asleep, it doesn't wakeup the dashboard anymore.

capi, 2024-11-18:
> I had some else report similar issue, but i was never able to reproduce. 

@meatpiHQ I'm not sure if I should open a new issue, if so, please tell me. I received both my first electric car (Dacia Spring 2024 model) and the WiCAN this week. I have noticed that the WiCAN is not going to sleep mode. 

And then I noticed, that I changed the sleep voltage to `13`. I reset it to `13.1` and suddenly it worked. And then I tried it with `13` again and it did not work. BUT it works with `13.0`. So I guess there is an issue with parsing the float if it's not actually a float?

In general, it would be awesome if I could set the timeout for sleep mode explicitly. Because I think I could get away with polling the SoC if the WiCAN went to sleep after ~1-2 minutes.

Other than that, thanks for developing this! I hope I can stay clear of Renault's cloud thanks to this :-)

mikegoubeaux, 2024-12-07:
I have the same issue as @smurf12345. Firmware 3.48, using AutoPID and the Home Assistant integration. The WiCAN seems to sleep under 13.1 v but then wakes up intermittently. Vehicle is a Mercedes Sprinter.

capi, 2024-12-08:
In the meantime I realized that the polling of SoC itself triggered the car to wake up and have short spikes of voltage >13V, which was in fact what woke up the WiCAN again. I personally migrated now to a self-implemented Python script based on ELM327 protocol to handle this specifically, since I expect waking up the HV battery too often is not too good for the entire system.

You could try to increase the poll interval way above the 3min sleep time for the WiCAN.

mikegoubeaux, 2024-12-08:
Interesting. In my case, the dance of sleep --> wake up --> sleep seems to stop once I lock the doors of the vehicle.