meatpiHQ/wican-fw #265: Newbie need some help - Vehicle profile for Volvo XC40 BEV

jomud9, 2024-11-21:
For EV I though information like SoC should be available via API, Volvo did that at the beginning but limited API access to some region only. So I looked for another solution and found this project. Thank you - This little device did what I want.

At this point, what I have
- A Mobile Hotspot provide internet access + create a Wireguard tunnel to my Home Router
- All WiFi devices at Car could access Home Assistant VM hosted at home
- mqtt server as HA Addon for things like z2m / espresence 

I guess if I could fix the routing issue to enable HA to connect WiCAN @ Car, I should be able to use WiCAN Integration for HA directly. But before that I afraid I need to create a vehicle profile and setup WiCAN for mqtt.

Here attached logs from car scanner. And I hope someone could help 
[xc40.txt](https://github.com/user-attachments/files/17847556/xc40.txt)

Thank you in advance

meatpiHQ, 2024-11-25:
@jomud9 I can help you create a profile. Can you please export another log file? but this time in car scanner press on the SoC sensor and let it log for a few seconds then immediately unplug the adapter so that the last readings in the logs would be of the SoC.

jomud9, 2024-11-26:
Hi, I tried to do some more in different files. Thank you in advance.

HV SoC
[Xc40_HV_SoC_log.txt](https://github.com/user-attachments/files/17920740/Xc40_HV_SoC_log.txt)

OBD Voltage
[Xc40_obd_volt_log.txt](https://github.com/user-attachments/files/17920747/Xc40_obd_volt_log.txt)

12V Battery Voltage
[Xc40_12V_volt_log.txt](https://github.com/user-attachments/files/17920724/Xc40_12V_volt_log.txt)

Please let me know if you need more.

meatpiHQ, 2024-11-26:
What was the SoC at the time of the log?

jomud9, 2024-11-26:
Ohh sorry, I missed that.

HV SoC should be around 73-74% but I might be wrong.

OBD was solid at 13.7 or 13.8V

OBD Volt - started at 13.7V, Dropped to 13.6V then 13.4. Return to 13.7 then 13.6 and ended at 13.7.

Thanks.

jomud9, 2024-11-27:
There attached two more file as I found.

Hybrid/EV battery pack remaining charge 100%
[HV_battery_log.txt](https://github.com/user-attachments/files/17929642/HV_battery_log.txt)
I just take this when I disconnect charger from my car.  Dashboard shown full charged (100%)

[HV_SoC_log.txt](https://github.com/user-attachments/files/17929655/HV_SoC_log.txt)

This is the HV SoC sensor which previous I think it should be the correct one. To my surprise it just shown [BECM] HV Battery SoC
: 96.43 %

meatpiHQ, 2024-11-27:
@jomud9 Can you please provide another log with a different data point. This will help in guess the equation.

meatpiHQ, 2024-12-06:
@ScuderiaNL sorry for the delay. I've created a new profile for xc40. Please try it an let me know if it works.

jomud9, 2024-12-07:
Thanks. Did I need to add something in order to let those entity show up?

I did those setting at WiCAN
[image]

but I did not found any new entity (for example SoC)

[image]

jomud9, 2024-12-13:
Okay. Seems I might miss up a step - Download xc40.json and upload to my device

Tried and I saw in HA new entity appeared. 

Let me have some fun and see. :-)

meatpiHQ, 2024-12-13:
@jomud9 so the profile is working now?

jomud9, 2024-12-24:
> @jomud9 so the profile is working now?

Negative :-(

I tried different way but seems I don't know anything else to make it work. :-(
and i found if I did too much, my adapter to stop working and only firmware flash could make it work again.

[image]

Provide that when i toggle it to ELM327 mode and I could connect car scanner, i think i did plug that well.
And from HA's integration it would read Voltage from the adapter, i assume integration works as well..

Any idea?

jomud9, 2024-12-31:
Okay. Updated to fw 4.00 (via web, then hanged again which requires another USB re-flash :-( )

I like new feature and I found while SoC did not work still, I could get something from Standard PIDs which I guess they are good!

[image]

I will try to figure out the custom PID calculation and keep update on this

meatpiHQ, 2025-01-02:
@jomud9 I've updated the xc40 profile, please sync and try it again.

jomud9, 2025-01-03:
> @jomud9 I've updated the xc40 profile, please sync and try it again.

1 bad news and 2 good news.
- Your update did not work.
- I finally know how to use car scanner and get correct log for you
- I figured out Custom PID for Odo

In addition, Standard PID: 5B-HybrBatPackRemLife work very nicely as SoC. Just FYI.

Let me put together my finding
1. Odo (as [IEM] Total distance
at 30733 km 
 >22DD011 
1EC6EE80 0662DD01 00780D

at 30776km
>22DD011
1EC6EE80 0662DD01 007838

Seems it is just put last 4 digits as Hex! 
as attached. It would be great if u could inspire me how to make that "expression"
[20250102_Odo_30733km.txt](https://github.com/user-attachments/files/18296172/20250102_Odo_30733km.txt)
[20250103_Odo_30776km.txt](https://github.com/user-attachments/files/18296173/20250103_Odo_30776km.txt)

2. SoC (as [BECM] HV Battery SoC 
82.65852
>2240281
1EC6AE80 0762402800 7E7461

64.94598
>2240281
1EC6AE80 0762402800 630454

Seems another Hex to Dec conversation plus some small calculation.
[20250102_SoC_6494598.txt](https://github.com/user-attachments/files/18296296/20250102_SoC_6494598.txt)
[20250103_SoC_8265852.txt](https://github.com/user-attachments/files/18296297/20250103_SoC_8265852.txt)

Please let me know if u need more logs. :-)

jomud9, 2025-01-03:
Okay @meatpiHQ. I made the Odo expression from try and error.

Init: ATCP1D;ATCRA1EC6EE80; ATFCSH1DD01637;
PID: 22DD011
Expression: ([B4:B7])/256

I think I could fork then commit for few more entity. However would you please help to rename this xc40.json to xc40 bev.json? 
It is because this is for pure electric version of xc40 instead.

jomud9, 2025-01-03:
> Okay @meatpiHQ. I made the Odo expression from try and error.
>
> Init: ATCP1D;ATCRA1EC6EE80; ATFCSH1DD01637; PID: 22DD011 Expression: ([B4:B7])/256
>
> I think I could fork then commit for few more [...]

Tried to create this pull request for the mentioned
https://github.com/meatpiHQ/wican-fw/pull/303

meatpiHQ, 2025-01-05:
@jomud9 This is some really good work!

closed 2025-01-06 by jomud9