meatpiHQ/wican-fw #266: I can only see one frame. SavvyCan and on Wican Monitor web ui

jameslouiz, 2024-11-26:
I can only seem to see a single frame in monitor 
[image]

In savvy can its the same (I have enabled overwrite mode)

[image]

Can anyone give any pointers on how I might diagnose this? 

Thanks

meatpiHQ, 2024-11-27:
@jameslouiz What's the make and model of the car?

This is normal, some cars only have a "I'm alive" frame, and do not broadcast the internal CAN data on the OBD port. 

You are still able to get data using OBD PID request.

jameslouiz, 2024-11-28:
@meatpiHQ Ah I see, so I need to know before hand what frame I want to request/ listen to. I was hopping I could just use savvy can to pin point exactly what I'm after. 

It's a 2018 VW crafter.

I just want to listen to when doors have been locked/unlocked using the fob and wether they are open/closed

meatpiHQ, 2024-12-06:
@jameslouiz Try searching, maybe someone has found the PID request for the door sensor.

BTW is it always showing a single frame when the car is OFF or ON?