meatpiHQ/wican-fw #267: [request] please add vehicle config for smart eq

user86000, 2024-11-27:
official smart eq servers will close on 31.12.2024
we need an alternative

meatpiHQ, 2025-01-02:
@user86000 are you able to provide car scanner logs of the smart eq?

user86000, 2025-01-11:
hello

here are some logs and datas for the smart eq

https://www.smart-emotion.de/forum/thread/1433-meatpi-wican-obd-2-wifi-ble-to-can-adapter-ovms-alternative-soc-mqtt/

user86000, 2025-01-11:
[image]
[image]

user86000, 2025-02-23:
Code: Car_Scanner.logs for Smart #1

[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK
>ATD0
ATD0
OK
>ATE0
ATE0
OK
>ATH1
OK
>ATSP0
OK
>ATE0
OK
>ATH1
OK
>ATM0
OK
>ATS0
OK
>ATAT1
OK
>ATAL
OK
>ATST64
OK
>ATAR
OK
>ATCP18
OK
>ATSP6
OK
>ATFCSM0
OK
>ATSH7DF
OK
>0100
7E806410000000000
>ATST64
OK
>0100
7E806410000000000
>
[SetRunning(True,From Start #2518)]
ATSHD01635
OK
>ATCP1D
OK
>ATSP7
OK
>ATCRA1EC6AE80
OK
>ATFCSH1DD01635
OK
>ATFCSD300000
OK
>ATFCSM1
OK
>2248011
1EC6AE80056248015492

................................................................................

{
  "init_commands": {
    "command": [
      "ATD", // Set all settings to defaults
      "ATD0", // Turn printing of CAN DLC off
      "ATE0", // Turn off terminal echo
      "ATH1", // Turn display of headers on
      "ATSP0", // Set OBD protocol preset
      "ATM0", // Memory Off
      "ATS0", // Print space off
      "ATAT1", // Adaptative Timing auto1
      "ATAL", // Allow long (>7 byte) messages
      "ATST64", // Set timeout to64 x 4 ms
      "ATAR", // Automatically receive (deprecated)
      "ATSH7DF", // Set the header of transmitted OBD 
      "0100", // returns 7E806410000000000
      "ATSHD01635", // Set the header of transmitted OBD 
      "ATCP1D", // Set CAN priority to 1D (29 bit only)
      "ATSP7", // Set to protocol 7
      "ATCRA1EC6AE80" // Set the CAN hardware filter
    ]
  },
  "data_commands": {
    "command": [
      "224801", // SoC
      "224802", // Current (Amps)
      "224803", // Voltage
    ]
  },
  "obd_protocol": "7",
  "soc": {
    "equation": "",
    "minValue": "0",
    "maxValue": "100",
    "type": "Number",
    "command": "224801",
    "ecu": "1EC6AE80"
  },
  "current": {
      "equation": "0.1*((B*256+C)-16384)", // A is always 02
      "minValue": "-400",
      "maxValue": "400",
      "type": "Number",
      "command": "224802",
      "ecu": "1EC6AE80"
  },
  "voltage": {
      "equation": "0.01*(A*256+B)", // A is always 03
      "minValue": "300",
      "maxValue": "480",
      "type": "Number",
      "command": "224803",
      "ecu": "1EC6AE80"
  }
}

...................................................................

# command 224802 for current is answered with 1EC6AE8005624802[B][C]
# Current = "0.1*((B*256+C)-16384)"
# command 224803 for voltage is answered with 1EC6AE8005624803[B][C]
# Voltage = "0.01*(B*256+C)"

# SoC command 224801 is answered with 1EC6AE8005624801[B][C]
# For SoC, [A] is always 01
# SoC B and C to translate into 47.18% to 59.77%

|  B |  C |
| -- | -- | 60%
| 81 | 0B |
| 78 | 21 |
| 77 | 24 |
| 77 | 13 |
| 74 | BE |
| 5C | 13 |
| 57 | E1 |
            47%
            
example
> 2248011
1EC6AE80056248017310      # 7310 for 58.9x %

-----------------------------

# SoC command 224801 is answered with 1EC6AE8005624801[B][C]
# For SoC, [A] is always 01
# SoC B and C to translate into 47.18% to 59.77%

|  B |  C |
| -- | -- |
| 81 | 0B |
| 78 | 21 |
| 77 | 24 |
| 77 | 13 |
| 74 | BE |
| 5C | 13 |
| 57 | E1 |

Ocramius, 2025-03-16:
@user86000 I'm also interested in tracking Smart EQ (Fortwo): Mercedes just closed off their API ( used https://github.com/ReneNulschDE/mbapi2020 before - they've also been doing shady stuff, like locking out users: https://github.com/ReneNulschDE/mbapi2020/issues/287 ).

Let me know if I can help with logs from my end: I'm an absolute newb, when it comes to CAN.

lucabacchipg, 2025-03-22:
Hello, please can add Smart EQ ? Also me i'm interested...Thanks

user86000, 2025-04-20:
{
  "car_model": "smart eq",
  "init": "ATST64;ATSH7DF;ATSP7;",
  "pids": [
    {
      "pid": "224801",
      "parameters": {
        "SOC": "(B3)"
      }
    }
  ]
}

Ocramius, 2025-04-26:
I couldn't make sense of the reply by @user86000 above ( https://github.com/meatpiHQ/wican-fw/issues/267#issuecomment-2677070097 ).

The reason is just that the Markdown is all garbled up: here's the markdown, unquoted / adjusted. I believe some segments are personal notes, but pasted into the github UI, they came out very cryptic to read afterwards.

```md
> Code: Car_Scanner.logs for Smart #1

 [VTSupported=False(ByResponse)] [ELM RESET] ATD OK

 > ATD0
 > ATD0
 > OK
 > ATE0
 > ATE0
 > OK
 > ATH1
 > OK
 > ATSP0
 > OK
 > ATE0
 > OK
 > ATH1
 > OK
 > ATM0
 > OK
 > ATS0
 > OK
 > ATAT1
 > OK
 > ATAL
 > OK
 > ATST64
 > OK
 > ATAR
 > OK
 > ATCP18
 > OK
 > ATSP6
 > OK
 > ATFCSM0
 > OK
 > ATSH7DF
 > OK
 > 0100
 > 7E806410000000000
 > ATST64
 > OK
 > 0100
 > 7E806410000000000

 [SetRunning(True,From Start #2518)] ATSHD01635 OK

 > ATCP1D
 > OK
 > ATSP7
 > OK
 > ATCRA1EC6AE80
 > OK
 > ATFCSH1DD01635
 > OK
 > ATFCSD300000
 > OK
 > ATFCSM1
 > OK
 > 2248011
 > 1EC6AE80056248015492

 ................................................................................

    {
    "init_commands": {
        "command": ["ATD", // Set all settings to defaults 
            "ATD0", // Turn printing of CAN DLC off 
            "ATE0", // Turn off terminal echo 
            "ATH1", // Turn display of headers on
            "ATSP0", // Set OBD protocol preset 
            "ATM0", // Memory Off
            "ATS0", // Print space off
            "ATAT1", // Adaptative Timing auto1 
            "ATAL", // Allow long (>7 byte) messages 
            "ATST64", // Set timeout to64 x 4 ms 
            "ATAR", // Automatically receive (deprecated) 
            "ATSH7DF", // Set the header of transmitted OBD 
            "0100", // returns 7E806410000000000 
            "ATSHD01635", // Set the header of transmitted OBD
            "ATCP1D", // Set CAN priority to 1D (29 bit only)
            "ATSP7", // Set to protocol 7 
            "ATCRA1EC6AE80" // Set the CAN hardware filter 
        ]
    },
    "data_commands": {
        "command": ["224801", // SoC 
            "224802", // Current (Amps)
            "224803", // Voltage 
        ]
    },
    "obd_protocol": "7",
    "soc": {
        "equation": "",
        "minValue": "0",
        "maxValue": "100",
        "type": "Number",
        "command": "224801",
        "ecu": "1EC6AE80"
    },
    "current": {
        "equation": "0.1*((B_256+C)-16384)", // A is always 02 
        "minValue": "-400", "maxValue": "400", "type": "Number", "command": "224802", "ecu": "1EC6AE80"
    },
    "voltage": {
        "equation": "0.01_(A*256+B)", // A is always 03 
        "minValue": "300", "maxValue": "480", "type": "Number", "command": "224803", "ecu": "1EC6AE80"
    }
    }
 ...................................................................
 # command 224802 for current is answered with 1EC6AE8005624802[B][C]
 # Current = "0.1*((B*256+C)-16384)"
 # command 224803 for voltage is answered with 1EC6AE8005624803[B][C]
 # Voltage = "0.01*(B*256+C)"
 # SoC command 224801 is answered with 1EC6AE8005624801[B][C]
 # For SoC, [A] is always 01
 # SoC B and C to translate into 47.18% to 59.77%

 | B | C | 
 | -- | -- | 60%
 | 81 | 0B |
 | 78 | 21 |
 | 77 | 24 | 
 | 77 | 13 |
 | 74 | BE |
 | 5C | 13 | 
 | 57 | E1 | 47%

 example

 > 2248011
 > 1EC6AE80056248017310      # 7310 for 58.9x %

 # SoC command 224801 is answered with 1EC6AE8005624801[B][C]
 # For SoC, [A] is always 01
 # SoC B and C to translate into 47.18% to 59.77%
 B 	C
 81 	0B
 78 	21
 77 	24
 77 	13
 74 	BE
 5C 	13
 57 	E1
```

If I understand correctly, @user86000 posted logs for a `Smart #1`, **not** a `Smart EQ`?

How would one go about adding these expressions created by @user86000 into this UI from MeatPI?

[image]

meatpiHQ, 2025-05-04:
I've created a profile for Smart # 1.

Is anyone able to export car scanner logs for the EQ in case it's different from # 1 ? 

https://meatpihq.github.io/wican-fw/car_scanner/logs

lucabacchipg, 2025-05-04:
I can export from an #3 if you want create a profile. #3 #1 are similar.

Ps please can consolidate Smart fortwo profile into master ?

If i send you car scanner pro logs from BMW CE02 EV motorcicle, can add to
standard profile ?

*Luca Bacchi*
* Digital & Security Sales Manager*

Cell: (+39) 334.7412934

Cell: (+39) 348.3549377

***@***.*** ***@***.***> *

*Linkedin* <https://www.linkedin.com/in/lucabacchi/>

Questo messaggio contiene informazioni confidenziali .

Il presente messaggio è rivolto unicamente ai destinatari. La divulgazione
o copia, anche parziale e non autorizzata, è proibita. Se non siete i
designati riceventi di questo messaggio, cancellatelo immediatamente dal
vostro sistema e comunicate al mittente l’errore dell’indirizzo di consegna
e la cancellazione del messaggio.

This message contains confidential information. This message is intended
for the recipients only. Disclosure or copy, even partial and unauthorized,
is prohibited. If you are not the intended recipient of this message,
please delete it immediately from your system, notify the sender of the
wrong delivery and confirm the mail deletion.

Ocramius, 2025-05-04:
I just tried to connect car scanner, but apparently, it can't detect the car itself? 🤔

[image]

Ocramius, 2025-05-04:
Attached is an export of the car-scanner logs I managed to export by:

1. downloading car scanner ( https://play.google.com/store/apps/details?id=com.ovz.carscanner ) 
2. pointing car scanner at `<local-ip-address>:3333`
3. adjusting the car profile to `Smart Fortwo EV / Smart Forfour EV`

[car-scanner-exported-logs-redacted.txt](https://github.com/user-attachments/files/20028441/car-scanner-exported-logs-redacted.txt)

meatpiHQ, 2025-05-04:
@Ocramius what was the SoC value at the time of reading?

Ocramius, 2025-05-04:
Where would I see that, please? :D

meatpiHQ, 2025-05-04:
Yes, state of charge

meatpiHQ, 2025-05-04:
@Ocramius can you please make another log at a different SoC value?

lucabacchipg, 2025-05-04:
Here read from Smart #3 log and screenshoot of data (SOC, SOC DISPLAY, ODOMETER)
Readed using ODBlink since Wican seem don't see data from Controller

[image]
[log (1).txt](https://github.com/user-attachments/files/20028641/log.1.txt)

lucabacchipg, 2025-05-04:
Ps. I have used no profile #1 to my #3 Smart and not work. SOC unavailable.

Another question: add please SmartForTwo configuration to SMART manufacurer (you have all data)
To be clear:

#1 and #3 are a car produced by Geely
SmartForTwo are made by Smart/Reanault 

so #1 #3 are same platform
FORTWO are old platform

Thanks

[image]

Ocramius, 2025-05-05:
> [@Ocramius](https://github.com/Ocramius) can you please make another log at a different SoC value?

Attached is a log with the battery charge at 80% - readings start `21:55:34`

[car-scanner-exported-logs-at-80-percent-charge-redacted.txt](https://github.com/user-attachments/files/20033663/car-scanner-exported-logs-at-80-percent-charge-redacted.txt)

user86000, 2025-05-05:
here is a correct log for **smart eq** (Baureihe W453]

last two sensors in the log are the "[BMS ] SOC" and the "[BMS] car contactor status"

BMS: 87,7%
Contactor: closed / Status 2

[log.txt](https://github.com/user-attachments/files/20037203/log.txt)

meatpiHQ, 2025-05-05:
@Ocramius @user86000 the logs don't make a lot of sense... The PID seems to be 223045 but the return value is always the same in all logs, when it's 100%, 80% or 87.7%

>223045
7EC056230458000

Ocramius, 2025-05-05:
Could it be related with the initialization sequence that @user86000 [suggested above](https://github.com/meatpiHQ/wican-fw/issues/267#issuecomment-2817079576)?

I can clear it and retry, perhaps?

meatpiHQ, 2025-05-05:
@Ocramius yes please. Do you see the correct value in car scanner ? Can you please take a screenshot?

user86000, 2025-05-05:
here is another log with 99,79% and conductor open / value 1

[log 1.txt](https://github.com/user-attachments/files/20039044/log.1.txt)

car scanner shows the correct values...

[image]

[image]

user86000, 2025-05-05:
here it is....

[log 2.txt](https://github.com/user-attachments/files/20039211/log.2.txt)

Ocramius, 2025-05-05:
Aha, now the logs do indeed look very different! :-) 

I've truncated the logs so they start at today's scan.

The State of Charge (SoC) is at 50.4%

[car-scanner-logs-after-checking-values-redacted.txt](https://github.com/user-attachments/files/20039850/car-scanner-logs-after-checking-values-redacted.txt)

user86000, 2025-05-05:
Ecu 7E4 pid 223045 is the connection check and the keep alive 

[image]

lucabacchipg, 2025-05-05:
Excuse me , but there are a big confusion in this thread.

Let's talk about Smart ForTwo or  Smart #1/ Smart #3 ?

For SmartForTwo, all data needed to display SOC and Remain KM  has been reported by me. 
It's time to simple add to SMART Manufacturer and create file.

For Smart #1 and #3 i send you a log yesterday taken from #3    

Thanks

meatpiHQ, 2025-05-06:
@Ocramius @user86000 I've created a profile for the EQ please try it and let me know.

@lucabacchipg I'm sorry, not sure I understand. Are you saying that the `#1`, `#3` and `EQ` are the same profile?

Ocramius, 2025-05-06:
> @Ocramius @user86000 I've created a profile for the EQ please try it and let me know.

Reconfigured Home Assistant and tried it out: `SOC` value being reported, but at `0%`. Is there a way to see the value being sent by WiCAN in the web UI somewhere?

meatpiHQ, 2025-05-06:
@Ocramius try this in you browser 

http://your_ip/autopid_data

user86000, 2025-05-06:
Soc=0
In real soc=72

there ist an other way to read the soc from smart eq, don't know of this works in the vsr profile....
Without pid (-1)
Can id 1620 decimal,  Start bit 0, length 1, expression B3

Ocramius, 2025-05-06:
> [@Ocramius](https://github.com/Ocramius) try this in you browser
> 
> http://your_ip/autopid_data

The endpoint does indeed report `0`

meatpiHQ, 2025-05-07:
@Ocramius I've made a change to the profile please try it.

@lucabacchipg Thank you, this is great information. Currently the profiles do not support filters, so I cannot add this information to the profile. The PID request should work.

lucabacchipg, 2025-05-07:
> [@Ocramius](https://github.com/Ocramius) I've made a change to the profile please try it.
>
> [@lucabacchipg](https://github.com/lucabacchipg) Thank you, this is great information. Currently the [...]

Retested after update also so 4.12fw 
Smart EQ Don't Work : NOW SOC Works ! (tested into 2 different SMART EQ)

Note:;
I had connected the ODB, flashed it and set the SMART EQ and it didn't work. Then when I disconnected it from the ODB it gave an error on the display, at this point I moved it to another Smart EQ and it worked. Disconnected and reconnected to the previous one it immediately worked. A strange thing.

user86000, 2025-05-07:
nice job, thanks

has anyboy the pid for contractor open / closed?

would be good to auto detect the correct car on the wallbox

Ocramius, 2025-05-07:
Got PID data reporting SoC, thank you, @meatpiHQ! This is exactly why I bought your product, BTW 😁

[image]

lucabacchipg, 2025-05-07:
Tested SMART #1 profile plugged into SMART #3 : NOT WORK 
"Soc unavailable"

meatpiHQ, 2025-05-08:
@user86000 

> has anyboy the pid for contractor open / closed?

Can you please provide car scanner logs just like you did for the EQ?

user86000, 2025-05-08:
i have found out the ecu and pid for smart eq for odometer from the ed4san project 
https://github.com/MyLab-odyssey/ED4scan/blob/master/ED4scan/_BMS_ED4_dfs.h

these parameters are not in the logs from odb scanner
can you help here?

Odometer:  

request id 743, response id 763
pid 220207
must be init: ATSH743;ATFCSH743;ATCRA763;

bits 3 to 5? i cant't interpretate these values..
myBMS->ODO = (unsigned long) data[3] * 65536 + (uint16_t) data[4] * 256 + data[5];

i get an value, but not the correct 

[image]

[image]

meatpiHQ, 2025-05-08:
@user86000 what is the correct odometer value?

user86000, 2025-05-11:
https://github.com/meatpiHQ/wican-fw/pull/409

lucabacchipg, 2025-05-14:
Hi, I see that the smart EQ you have is the old model. I have the EQ 453 (latest) and the current SOC profile works on the Smart EQ profile with .a 4.12fw. Are you sure that these parameters are ok for the 453 too?

user86000, 2025-05-14:
i have an eq 453 ed4 (2017-2024 i think) should be the same, you can test the parametes manually, they should work

lucabacchipg, 2025-05-14:
> i have an eq 453 ed4 (2017-2024 i think) should be the same, you can test the parametes manually, they should work

Ok, tested. 
Soc, KM Remain , Range Works
Connector Status = Not work , no change

user86000, 2025-05-14:
What is the answer from rhe canbus? No answer or a number?
Don't use [] for connector status.

user86000, 2025-05-14:
> > i have an eq 453 ed4 (2017-2024 i think) should be the same, you can test the parametes manually, they should work
> 
> Ok, tested. Soc, KM Remain , Range Works Connector Status = Not work , no change

The are to different loader, 7kW and 22kW
Check this, should annswer with A, B, C, D

ATSH719;ATFCSH719;ATCRA739;
226155
B4

lucabacchipg, 2025-05-15:
> > > i have an eq 453 ed4 (2017-2024 i think) should be the same, you can test the parametes manually, they should work
> > 
> > 
> > Ok, tested. Soc, KM Remain , Range Works Connector Status = Not work , no change
> 
> The are to different charger, 7kW and 22kW Check this, should annswer with A, B, C, D
> 
> ATSH719;ATFCSH719;ATCRA739; 226155 B4

Hi, my is 22kw. 
Thested value but give me unavailable.

closed 2025-05-19 by user86000