meatpiHQ/wican-fw #271: [New Vehicle] Volkswagen Passat GTE 2015

ScuderiaNL, 2024-12-03:
Hi there,

I would like to request support for our car the Volkswagen Passat GTE 2015.
Main reason of purchase: WeConnect/Carconnect doesn't work with my car, because i don't have a simcard-slot.
I am very much interested in the hybrid battery parameters.
- SOC
- Renaming electrical Range
- Battery Voltage
- ChargerSocket Voltage 

Also nice to know would be
- ODO
- Locked / unlocked

Unfortunately i haven't  had much succes using the json files for ID4, e-golf or e-up
The SOC Wican reports to HomeAssistant is the 12V battery i think. Values dont match the readout in the car.

I can't find the right parameters in Carscanner; i think they are in the hidden part of the CANBUS.
Our car is still very much a petrol car CANBUS-wise.
I am worried only VAGCOM or VCDS can expose these parameters. 

I did find the ODO with carscanner
[intake air temperature log.txt](https://github.com/user-attachments/files/17998088/intake.air.temperature.log.txt)
[odometer log.txt](https://github.com/user-attachments/files/17998089/odometer.log.txt)
selection of interesting values i could read with carscanner:
[interesting values.txt](https://github.com/user-attachments/files/17998288/interesting.values.txt)

I did try to build a very short JSON file, but none of the parameters are recognized in HA.
[heelkort.json](https://github.com/user-attachments/files/17998126/heelkort.json)

I also had some advise from the user typxxi, but this was not enough.

NOTE:
I bought the original meatpi, not the PRO.
Is the Wican USB capable of exposing the Volkswagen parameters?

ScuderiaNL, 2024-12-03:
https://forums.ross-tech.com/index.php?threads%2F38868%2F
Here some more info about "proprietary" VAG Blocks.

ScuderiaNL, 2024-12-03:
Maybe this can help to:
https://www.reddit.com/r/CarHacking/comments/16v0484/volkswagen_obd_pids/?rdt=51284
https://pastebin.com/9vuNU68y

meatpiHQ, 2024-12-06:
@ScuderiaNL 

> I did try to build a very short JSON file, but none of the parameters are recognized in HA.

Do you mean the values are not showing in the HA integration ?

Did you change the protocol to "AutoPid" ?

ScuderiaNL, 2024-12-08:
[image]
They are showing in HA, but they are all unavailable. I think i just use the wrong addresses and need "proprietary" VAG Blocks.
Besides a few tests on "ELM327" i've always used "AutoPID", as HA reports.

meatpiHQ, 2024-12-13:
@ScuderiaNL I suggest that we start adding parameters one by one. Let first focus on the SoC.

Can you please make a car scanner log for the SoC, make sure to click on the sensor and let it run for sometime like 20 sec. Also note the SoC at the time of taking the reading.

If you can take multiple logs at different SoC levels, it will make things easier guessing the equation.

ScuderiaNL, 2024-12-13:
I can't, its a "proprietary" VAG parameter. 
I cant find it in Carscanner and Torque.
I think is only available in the expensive tools, like VCDS, VAGCOM and maybe ODB11.

JabeBRD, 2024-12-25:
How you have setup your homeassistant and wican?
Do you use homeassistant to ask wican dongle parameters/values and then response is made in to sensor?
Or do you use wican automation function?

ScuderiaNL, 2024-12-28:
Hello JakeBRD, first of all tnx 4 ur time.
i hope i understand all your questions right.

setup HA
i only use the HA integration and only needed to fill in the IP address

Wican
It's setup to AutoPID, as advised by the HA integration
I have used several json vehicle files, im now running "heelkort.json", which i have uploaded in the OP (opening post).
I have before used the files from E-Up and E-Golf, which also gave me the SOC from the 12V battery.

I have downloaded the program MQTT explorer, but don't know yet how to configure the connection.
Your solution is a lot more complex then the standard HA integration.
I've attached the settings screen of MQTT explorer, which i am unsure how to fill in.
[image]

Also i was wondering if the builtin MQTT of HA is usable?
[image]

JabeBRD, 2024-12-29:
For MQTT Explorer you need your MQTT server IP and username and password.
Usually IP is same as your HA IP, you could try `homeassistant.local` for IP address, but it is better to use direct address.

meatpiHQ, 2024-12-29:
@JabeBRD 

> To read SoC i send message to wican and get raw value as response, message is this `{"bus":"0","type":"tx","frame":[{"id":2021,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,140,170,170,170,170]}]}`
> I have used same commands as e-Up and e-Golf use, most are same.

Can you please post a sample of the response? I'd like to create a profile for the Passat.

JabeBRD, 2024-12-29:
> @JabeBRD
>
>
> Can you please post a sample of the response? I'd like to create a profile for the Passat.

`{'bus': '0', 'type': 'rx', 'ts': 7217, 'frame': [{'id': 2029, 'dlc': 8, 'rtr': False, 'extd': False, 'data': [4, 98, 2, 140, 208, 170, 170, 170]}]}`

ScuderiaNL, 2025-01-03:
Hello There Jake, 

First of all best wishes for 2025.
I didnt succeed in connecting directly to meatpi, so i restored the connection to my mqtt broker on HA.
If im listening to the correct topic i can see the messages there too, although a bit messy.
[image]

In mqttexplorer i see a lot of data coming in to HA.
I think the response is {"error": "Timeout, pid: 010F1"}, this was with ignition off.
[image]

Then i tried the same with ignition on, the response is also looks like an error:
{"bus":"0","type":"tx","frame":[{"id":2021,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,140,170,170,170,170]}]}
[image]

Then i used the new command youve written (20**29**)
It also responses with an error as you can see here (the car is charging at ~95%)
[image]

Note:
In meatpi i have now selected: 
Vehicle Specific: Disable

JabeBRD, 2025-01-03:
It bit tricky to see those messages, there is lot of "noise" in RX topic with some VW models.
Try again, send my message in TX topic and then press PAUSE in MQTT Explorer, search RX topic for that answer.
Its same for me with Passat, there is lots of unwanted messages, those are allways there when car is awake.

ScuderiaNL, 2025-01-03:
That PAUSE made a difference
My message now looks very similar to yours
Is the "32,16" the data? 
Mine is currently 100% charged, connected to a grannycharger. 
The grannycharger is connected to the car, but not to the mains (HA disconnects when charging stops). 
[image]

ScuderiaNL, 2025-01-03:
> That PAUSE made a difference My message now looks very similar to yours Is the "32,16" the data? Mine is currently 100% charged, connected to a grannycharger. The grannycharger is connected to the car, but not to the mains (HA disconnects when charging stops). 

Btw this printscreen was with ignition offl

JabeBRD, 2025-01-03:
It does not matter if ignition is ON or OFF if car is charging, if not charging then it needs to be ignition ON. Control units that you are trying to talk to, needs to be awake, when car is charging some control units are awake and those respond.
`"data":[32,16,0,0,0,0,0,0]}]}` is not that data what we are after, that is message that needs to be ignored.
Search for message that looks like this, open that `History` tab and you can browse all messages.
`{'bus': '0', 'type': 'rx', 'ts': 7217, 'frame': [{'id': 2029, 'dlc': 8, 'rtr': False, 'extd': False, 'data': [4, 98, 2, 140, 208, 170, 170, 170]}]}`
In this message `208` is data for SOC level, it can be identified by control unit id `2029` and data `[4, 98, 2, 140, 208, 170, 170, 170]`
You have sent message that has data like this `"data":[3,34,2,140,170,170,170,170]`, then search for respond that has values `2, 140`, after those is datavalue that we are looking for (208 in example).
See explanation here https://www.goingelectric.de/wiki/Liste-der-OBD2-Codes/  <- there everything is in HEX, what we use here are in DEC, example DEC 2021 -> HEX 7E5, DEC 2029 -> HEX 7ED

JabeBRD, 2025-01-03:
I have made `sensor` in HA to log all messages, it listens to my wican RX topic, i have made filter that filters that unwanted message, it is much easier to read this log file than MQTT Explorer's unfiltered messages.
```
mqtt:
  sensor:  
    - name: "Passat_GTE_raw_data"
      unique_id: "passat_raw_data"
      state_topic: "wican/passat/can/rx"
      value_template: >-
        {% if not value_json.frame[0].data[0:5] == [32,16,0,0,0] %}
          {{ value_json }}
        {% else %}
        {{ this.state }}
        {% endif %}
```
Just replace `state_topic` with yours.
But again, im not using that Wican automation feature currently, i have all setup in HA, lots of sensors and automation that starts to ask wican/car for information when wican is online.
This sensor is only for debugging and should be but in `Logbook` ignore list, or it will flood your Logbook in HA.

What i think needs to be done, is determine how car responds (what ID it uses and where that data is located in respond message), by sending these messages manually to wican, after those are sorted, then it can be turned into automate.

ScuderiaNL, 2025-01-03:
I'm sorry Jake. i've read your message 20 times. But i just don't understand.
i thought the 208 was the return data. 
Would you be willing to help me onscreen?
I could share my screen in Microsoft Teams or maybe Google Meet?

JabeBRD, 2025-01-03:
Ill try explaining one more time here, as its not possible for me in Teams just right now.
1. Open MQTT Explorer
2. Turn Ignition ON
3. send message to TX topic via Publish tab,  select `json`, use this message:
  `{"bus":"0","type":"tx","frame":[{"id":2021,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,140,170,170,170,170]}]}`

4. Press PAUSE in MQTT Explorer and browse History tab in RX topic for message that contains ID `2029`, in that same message DATA should look something like `[4, 98, 2, 140, 208, 170, 170, 170]`, byte 4 is that value. (bytes start counting from 0), 208 in this example.
Whole message should look something like this
`{"bus":"0","type":"rx","ts":11195,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,208,170,170,170]}]}`
5. In this example my value 208 calculates to ~86% with formula (208-24)/2,15

Try to be quick pressing PAUSE after sending TX message, its easier to find that response when topic is not flooded with unwanted messages.
Messages that look like here below, needs to be ignored, your car has no fault.
`{"bus":"0","type":"rx","ts":14686,"frame":[{"id":401604624,"dlc":8,"rtr":false,"extd":true,"data":[32,16,0,0,0,0,0,0]}]}` 

Here is another response from my car
`{"bus":"0","type":"rx","ts":13626,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,92,170,170,170]}]}`
Here value is 92, that calculates to ~32% SOC.
See how 2,140 is followed by value that changes with battery level, that same 2,140 can be found from TX message, thats why we are looking for those numbers.

Just go through 1 - 4 until you find that response, it should be there, be quick to pause, it gets burried really quick.
This is just to verify that your car uses/responses to same messages.
If there really is no response, then we need to try something different, but this is common address for VW, thats why we try this first.

ScuderiaNL, 2025-01-03:
the response now is {"bus":"0","type":"rx","ts":24285,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,237,170,170,170]}]}

(237-24)/2,15=99%
which does fit the number in the dash (unfortunately no exact number in the dash of the GTE2015)
[image]

tomorrow i will do some more tests to make sure this number isn't from the 12V battery.
tomorrow the wife will drive it empty so good moment for some more tests (freezing here at the moment).

i've added the Sensor to HA
i included this in my configuration.yaml to stop logging:

recorder:
  exclude:
    entities:
      - sensor.Passat_GTE_raw_data

[image]

JabeBRD, 2025-01-04:
Great, now we can ask for other data.

Range is requested with message
`{"bus":"0","type":"tx","frame":[{"id":1808,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,182,170,170,170,170]}]}`

ODO is requested with message
`{   "bus": "0",   "type": "tx",   "frame": [     {       "id": 402391163,       "dlc": 8,       "rtr": false,       "extd": true,       "data": [ 3, 34, 2, 189, 170, 170, 170, 170 ]     }   ] }`

Both range and odo, need automation to be able to read all of the data, these messages are long messages, they consist of total 2 or 3 messages.
Car responds with message 1, after that is received, it is needed to ask rest of data with message
`{"bus":"0","type":"tx","frame":[{"id":1808,"dlc": 8,       "rtr": false,       "extd": false,       "data": [ 48, 0, 0, 170, 170, 170, 170, 170 ]     }   ] }`
(this is for range)

ScuderiaNL, 2025-01-04:
answer when she came back:
{"bus":"0","type":"rx","ts":22255,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,52,170,170,170]}]}
52 
(237-24)/2,15=13% seems reasonable value with the needle somewhere in te red erea.

is the response from e-range also id 2029?
2km response: {"bus":"0","type":"rx","ts":37453,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,62,170,170,170]}]}
3km response {"bus":"0","type":"rx","ts":27572,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,68,170,170,170]}]}

odo {"bus":"0","type":"rx","ts":17641,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,72,170,170,170]}]}
odo {"bus":"0","type":"rx","ts":57711,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,75,170,170,170]}]}

JabeBRD, 2025-01-04:
Now we know that SOC request is working, those values look reasonable, and all of those messages are for SOC.

If you want to read e-range, send only message
`{"bus":"0","type":"tx","frame":[{"id":1808,"dlc":8,"rtr":false,"extd":false,"data":[3,34,42,182,170,170,170,170]}]}`

Car responds here with ID 1914, and again look for numbers `42, 182`, so that you know you got right response (you can see those in TX message we have sent, just before numbers 170)
Message you will receive should look something like this
`{'bus': '0', 'type': 'rx', 'ts': 7729, 'frame': [{'id': 1914, 'dlc': 8, 'rtr': False, 'extd': False, 'data': [16, 10, 98, 42, 182, 0, 229, 0]}]}`
This message starts with number 16, so it is long message, just part1 of many messages.
Part2 needs to be requested with another message, but here it gets tricky, as passat is bit picky, this second message needs to be send fast enought be get part2. Here is where automation kicks in, i have automation that send those two message with 40ms delay, you cant do it manually fast enought.
Part2 starts with number 33 and part3 would start with number 34. (this message has only 2 parts)

In this range request, part1 contains compined range, part2 would contain separate range for petrol and battery.
This is how part2 would look like
`{'bus': '0', 'type': 'rx', 'ts': 7769, 'frame': [{'id': 1914, 'dlc': 8, 'rtr': False, 'extd': False, 'data': [33, 200, 0, 29, 0, 170, 170, 170]}]}`

In this example, my compined range is 0 x 256 + 229 = 229km, petrol 0 x 256 (last byte from part1) + 200(from part2), battery range is 0 x 256 + 29 = 29km
When your range is more than 256km, it will consist two numbers, 280km would be 1, 24 (1 x 256 + 24).

ScuderiaNL, 2025-01-04:
I dont see any id:1914 response (e-range)

After the odo request i get a lot of empty responses like this one:
{"bus":"0","type":"rx","ts":18711,"frame":[{"id":401604624,"dlc":8,"rtr":false,"extd":true,"data":[32,16,0,0,0,0,0,128]}]}
The funny thing is, Carscanner was able to show the ODO but not the SOC

ScuderiaNL, 2025-01-04:
the odo 2021 seems  to work; response is (also) from a 2029 id
{"bus":"0","type":"rx","ts":33518,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,236,170,170,170]}]}
current km readout on dash : 205367 km

for range i cant find anything near 1812, besides the tx itself

JabeBRD, 2025-01-04:
No that message is still soc, but that odo should respond to 2029.
Are you sure that `retained` is not selected, you can also clear tx topic with mqtt explorer by deleting whole topic, it will be back when new message is sent.

If you send soc, odo and range messages once, what would ha sensor log show?

ScuderiaNL, 2025-01-04:
I have uploaded the log from carscanner ODO in the OP.
The value that is repeated constantly is:

>0100
7E8064100BE3EA813

At dinner i suddenly realised i may have made a mistake.
I switched to vehicle specific mode vw e-up, to get the HA integration up again.
Could this have caused interference to what we are doing with mqtt?

ScuderiaNL, 2025-01-04:
I decided to check out for myself and switched back to vehicle specific mode and disable the HA integration
request odo
{"bus":"0","type":"tx","frame":[{"id":2021,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,189,170,170,170,170]}]}
no response from 2029

request for range
{"bus":"0","type":"tx","frame":[{"id":1812,"dlc":8,"rtr":false,"extd":false,"data":[3,34,34,224,170,170,170,170]}]} 
no reponse from something around 1812

Can you translate the value from carscanner in the post from 30min ago?

ScuderiaNL, 2025-01-04:
> If you send soc, odo and range messages once, what would ha sensor log show?

the funny this is i dont get RX value in HA anymore
yust the TX i post in MQTT

JabeBRD, 2025-01-05:
You need to disable autopid feature and all filters that are set to wican. It should be in default settings and mqtt enabled. 

To make log, use wican elm debug mode and open carscanner, look for values that you want to add to wican.
Use HA sensor to read log.
You need to enable this sensor in recorder, but disable in logbook.

ScuderiaNL, 2025-01-05:
Good afternoon Jake,

I assume i need to select ELM327 instead of AutoPID?

About your second sentence; i think this is my main problem as mentioned in the OP. 
I can't find any of the values that have anything to do with the high voltage battery in Carscanner.
I think that's my main problem, i have 12 screenshots like this one. 
I can't log what i can't see. A lot of information from the ICE engine, but nothing from the E-engine.

I think i 'll have to pause this issue for now. 
My wife is getting p*ssed with the amount of time i've spent behind the laptop this Christmas.

I 'll first try to contact the official Volkswagen dealer maybe they can confirm to me the PID's we're looking for actually exist in my old car.
They should be ale to prove this by reading them on the spot in their workshop with their official equipment. Hopefully software will include full PID and header.
This will take me some days to get an appointment at least.

1
[image]

2
[image]

3
[image]

4
[image]

5
[image]

6
[image]

7
[image]

8
[image]

9
[image]

10
[image]

11
[image]

12
[image]

JabeBRD, 2025-01-05:
Here is how your wican setup should look like, you can use your own topic name and url address, but basic settings need to be same.
[image]
Most important are CAN parameters to be same, also `MQTT elm327 log` needs to be `disable` 

[image]
[image]
Also `automate` should be now disabled and no filters set in `Settings` tab.

These setting are now for getting right messages going manually, lets forget carscanner for now, most of these values can be read with carscanner but profile/vehicle needs to be set something else, lets not go into that right now, lets test with messages that work with similar cars as its far more easier.

Then check that you have MQTT Explorer set up correctly
[image]
Note that we are in TX topic, `json` is selected and `retain` is unchecked

[image]
If you are seeing yellow `retained` flag in your TX topic (like i have on my eUp topic), before you have sent any messages, press that X on flag to delete retained topic.
There is reason why i have it like this, but you dont want retained messages right now.

After you have all set up correctly, just start sending those requests i have posted, and check your HA sensor, what car responses (you can see it in RX topic for like 1/4s)
[image]
Open `History` tab, select your sensor, select timeframe and under three dots, `Download data`.
Note that your **sensor must not be excluded from recorder**, only from logbook. (you can have it included in logbook, but then it will flood your logbook display)

Here is referenses that i have used to build these messages/requests, there is no secret behind, just try and error.
https://www.goingelectric.de/wiki/Liste-der-OBD2-Codes/  http://obd-amigos.linuxtech.net/files/amigos_PIDs.pm

ScuderiaNL, 2025-01-18:
Hello JakeBRD,

I finally found someone who could help confirming the undocumented PID's
So i drove to Dordrecht last wednesday to get things checked. 
In the end VAGCOM VCDS is still the ultimate software for  these things.
[https://www.facebook.com/Vagcomdordrecht/posts/pfbid0wKt2v7aT53fXwdrhAenfAmnFnuhb5QeRdZu5D7zs2W8cwHvigQF1cyxds3QcvSzTl](url)
We found the remaining charge time in C6, unfortunately i don't have a PID#
I also found out i don't have the BD Module (765/7CF)

I did what you mentioned here above and my values are coming into HA just fine now.
I attached what i have so far but i don't know how to combine values from 2 frames. 
I guess it's time now to build the automations in HA?
[new test.txt](https://github.com/user-attachments/files/18462058/new.test.txt)

As you can see have found in the vagcom log C6 / 744 HV Charger
-Actual HV charge current * A
-Remaining time for charge procedure * min
These are the undocumented values. 
[Log-VW PASSAT GTE 2015 ST - opgeschoond.txt](https://github.com/user-attachments/files/18462052/Log-VW.PASSAT.GTE.2015.ST.-.opgeschoond.txt)

JabeBRD, 2025-01-18:
When you get response `[3, 127, 34, 49, 170, 170, 170, 170]`, it basicly means that control unit responses error, i have not done any research what it means, it could be that you need to login before accessing data or its just saying that there is no data in that address.

Your kwh and odo reguests need second message, just build small automation that sends two messages with 50ms delay betwene and check result.

Now as we know what control unit holds data you want to extract, we can try different messages, vagcom log does not provide any other hint, those "addresses" do not correspond to can messages, they have done their research and that data is hidden and encrypted.

typxxi, 2025-02-02:
> When you get response `[3, 127, 34, 49, 170, 170, 170, 170]`, it basicly means that control unit responses error, i have not done any research what it means, it could be that you need to login [...]

REALLY GREAT SUPPORT FROM YOU AGAIN - THANKS FOR HELPING OUT OUR DUTCH FRIEND !

ScuderiaNL, 2025-02-02:
[image]

Here's a litte update of what i have right now.
I have around 75% of the parameters i want.

This week i will start looking for lock status in VAGCOM, cluster 9. 
The guy from Dordrecht confirmed already the data is there.
I bought a Y-splitter cable to sniff/log with Meatpi what VAGCOM is doing.

Also i still have some bugs in formula's.

JabeBRD, 2025-02-07:
Your welcome.
There is some serious data you got.
For values that use two or three messages, there is possibility to get false data, as it is not possible to identify data from messages that start with `33` or `34`, therefor i have come up with solution to compare these messages timestamp to first message.
example here 
```
    - name: "Passat_GTE_range"
      unique_id: "passat_range"
      state_topic: "wican/passat/can/rx"
      unit_of_measurement: "km"
      icon: mdi:road-variant
      state_class: "measurement"
      value_template: >-
        {% if value_json.frame[0].id == 1914 %}
          {% set PID = value_json.frame[0].data[0] %}
          {% if PID == 33 %}
            {% if 0 <= (value_json.ts - state_attr('sensor.passat_gte_full_range','ts')) <= 500 %}
              {% set AA = value_json.frame[0].data[3] %}
              {{ AA | round(0) | int }}
            {% endif %}
          {% endif %}
        {% endif %}
      json_attributes_topic: "wican/passat/can/rx"
      json_attributes_template: >-
        {% if value_json.frame[0].id == 1914 %}
          {% set PID = value_json.frame[0].data[0] %}
          {% if PID == 33 %}
            {"ts": {{value_json.ts | round(0) | int(1000)}} }
          {% else %}
          {"ts": {{(state_attr('sensor.passat_gte_range','ts')) | int(1000)}} }
          {% endif %} 
        {% else %}
        {"ts": {{(state_attr('sensor.passat_gte_range','ts')) | int(1000)}} }
        {% endif %}
        
    - name: "Passat_GTE_full_range"
      unique_id: "passat_full_range"
      state_topic: "wican/passat/can/rx"
      state_class: "measurement"
      unit_of_measurement: "km"
      icon: mdi:road-variant
      value_template: >-
        {% if value_json.frame[0].id == 1914 %}
          {% set PID = value_json.frame[0].data[0:5] %}
          {% if PID == [16,10,98,42,182] %}
            {% set AA = value_json.frame[0].data[5] %}
            {% set BB = value_json.frame[0].data[6] %}
            {% if AA <= 3 %}
              {{ ((AA*256) + BB) | round(0) | int }}
            {% else %}
              {{ this.state }}
            {% endif %}
          {% endif %}
        {% endif %}
      json_attributes_topic: "wican/passat/can/rx"
      json_attributes_template: >-
        {% if value_json.frame[0].id == 1914 %}
          {% set PID = value_json.frame[0].data[0:5] %}
          {% if PID == [16,10,98,42,182] %}
            {"ts": {{value_json.ts | round(0) | int(0)}} }
          {% else %}
          {"ts": {{(state_attr('sensor.passat_gte_full_range','ts')) | int(0)}} }
          {% endif %} 
        {% else %}
        {"ts": {{(state_attr('sensor.passat_gte_full_range','ts')) | int(0)}} }
        {% endif %}
```
First messages timestamp is saved into full range sensor as attribute.
E-range messages timestamp is then compared against full range timestamp, if it is fresh enought, then it is accepted and displayed.
As these messages should appear "at same time" in RX topic.

You need to add line `unit_of_measurement: "km"` to each sensor, ofcourse `"km"` is replaced by suitable, i think there is no list of "accepted" units of measurement, you can use anythink you like.
All sensors that share exactly same unit of measurement are drawn into same graf.

arcidodo, 2025-10-25:
i have an wecan pro, and want to use it on my Passat GTE to get the soc for my wall charger.  this is working perfect i using ScuderiaNL his automation code. but the car alarm was triggerd and im not sure why. my guess is that the automation is asking to often data.

now an other question is it possible to enable my auxiliary heater using this? that would be awesome 

Example: HA automation, if outsite temp is lower then 8 and calendar shows that im working at the office then turn on the heater;-)

I have an vcds but no Y cable atm.

arcidodo, 2025-10-26:
Yes i have an paid subscription but the 2G and 3G are gone soon.. so the car don't have an active internet connection anymore.. 

I use EVCC to charge my car smart but i need the SOC of my car to make it work, so i tought this is the solution.. but sadly it  
isn't because i cannot left open my car at night. Or let the alarm go off in the middle of the night is also not a good idea :-p 

Heater:
I think i go for an "SwitchBot" solution.. not an great solution but if it works... then i have an switch in HA to turn on my webasto heater.

Or i put the remote on my nightstand and click it when i wakeup :-p

ScuderiaNL, 2025-10-26:
@JabeBRD 
Maybe I can find the PID with VCDS sniffing?
But this wil always stay readonly.
You'll have to get behind the canbus gateway if you really want to control Clima.

@arcidodo 
Do you let the Meatpi sleep?

Power saving
Sleep mode: enabled
Sleep voltage: 12.8V
Sleep time: 1 min

arcidodo, 2025-10-26:
Yes en no.. I setup it to 12v to get to sleep.

I changed it to 13v so it only turn on when charging. Lets see what's happening the next couple of days the car is now full so can't test it now. 😬.

arcidodo, 2025-10-26:
https://github.com/meatpiHQ/wican-fw/issues/234#issuecomment-2480679095

I try to disable the OBD-Monitoring. but cant find it yet in my passat using obdeleven when i have time i do some more research. or do you guys already tried this?

arcidodo, 2025-10-29:
update on the Alarm thing:

After i change that setting (disable OBD alarm), The alarm din't go off anymore:-) so @typxxi maybe it is worth to do the same ;-)

happyjaxx, 2025-11-30:
@arcidodo Hey ! I'll give that a shot on my ID.Buzz ... the Wican was triggering the alarm and even a vLinker dongle was too when I was near the car with ABRP forgotten in the background on my phone...
This might avoid starting a war with my neighbors and family :p

happyjaxx, 2025-12-05:
Happy that it's an SFD1 actually... I can do other adaptations... i'm just not finding the correct setting for the MEB platform I suppose... I asked on the OBD11 forum (but in the ID.Buzz section where there's possible nobody around haha... I'll crosspost on ID.4 eventually)

alien0matic, 2026-08-28:
I have a 2022 GTE and thanks to you guys I got it working with Home Assistant and evcc. I would have never been able to do that on my own. thx