meatpiHQ/wican-fw #274: WiCAN stops mqtt sendings and needs re-plugging to send again

micha53, 2024-12-06:
I got my WiCAN this week and it works in principle. I use it in a Renault Zoe Ph. 2.
But I have one Problem: If the car ist standing and not connected to a wallbox, it stops sending mqtt telegrams. For  a while it sends an Error Information, later this also stops. This should not be a problem, when it would start sending per mqtt after starting to load the car. But the stick remains "Sleeping" when I start loading the car. (It is the same behavior with and without enabling sleep mode in the settings.) I think in this moment the stick should restart to send mqtt.
A Reboot of the stick makes no effect, only disconnecting and replugging brings success. 
I do not understand why replugging (hard reset) works but a software reboot brings no effect. But in general the stick should restart to send mqtt, when the loading process starts. I can't imagine that replugging can start the CAN communication of the car, so why does the stick not listen to the CAN?
The ...can/status topic says "offline" and the ...can/rx is not present.
At least with enabled sleep mode the power led is blinking and I cannot access the stick until I replug it. Also turning on the ignition of the car brings no change.
I would like to have that the wican starts to send mqtt if I start to load the car, without replugging the stick.

meatpiHQ, 2024-12-06:
can you please share a screenshot for your device configuration page? And the status tab as well.

micha53, 2024-12-06:
Config and Status:
[image]
[image]
[image]

meatpiHQ, 2024-12-06:
It's normal to stop sending MQTT if the car is not connected to the charger, because the ecu goes in low power mode. But it sounds like there is a bug in the firmware, when it's connected but ecu is not responding it fails. 

I will try to reproduce this issue here.

What happens if, when in this state you plug in the charger and then reboot?

micha53, 2024-12-06:
actually this happened:
i was coming home by car half an hour before. Just looked at the dongle: Blue and green LED' s were on, yellow one was flashing. I started the car and moved some meters. After that flashing the yellow LED stopped.
And now, additional 15 minutes later I opened the mqtt explorer and for some cycles I saw the right values and after some seconds they vanished  and I see only this:
[image]
**cfos_mqtt/set/M12** is the automate destination topic.

**Your Question:
If I plug in the charger and reboot nothing changes in the mqtt explorer.**

btw. using the monitor in this situation I do not see any can-message.

micha53, 2024-12-17:
I made another test today:
Wican is in sleep mode and sleeps, I open the door: Wican wakes up and publishes car data
I close the door: after one minute wican stops publishing and goes to sleep
I open the door again and open the charging connector and close the door again and plugin the charging cable: Wican wakes up and publishes car data

So the adapter works in a way like it should do, only I had done something that caused the error. But actually I cannot say what was the difference.

I will continue to observe the behavior and will give a feedback when I know more about it. So actually there's nothing to do for the developer.

micha53, 2024-12-27:
Actually it works fine, so I close this issue.

meatpiHQ, 2024-12-27:
@micha53 works fine on the latest firmware version or are you still on the old one?

micha53, 2024-12-27:
Ah, I see there is a new version from yesterday.. No I still have the old one.
But a question concerning the new one: 
I had problems with the value for the 12VBatt Voltage. In home assistant this variable causes an error. Adding a letter bevore the 12 solved the problem, but does not read the value published by the wican. Maybe it is not good to begin a value used here with a number. Does it still have the name 12vBatt?

meatpiHQ, 2024-12-27:
That's in the zoe vehicle profile right? I can change it in the profile, this is not hard coded in the firmware.

micha53, 2024-12-27:
yes in the zoe profile

meatpiHQ, 2024-12-27:
Ok, it's fixed now. Resync the profile and give it a go.

micha53, 2024-12-27:
thank you
and.. can you tell me, or put into the doc, what is the payload of an Battery alert?

meatpiHQ, 2024-12-27:
here you go

{"alert": "low_battery", "battery_voltage": 12.0}

micha53, 2024-12-27:
so, Christmas is over, I thank you for the presents and I wish you all the best and a happy new year!

meatpiHQ, 2024-12-27:
Happy new year! Cheers!

micha53, 2025-06-10:
I had today the situation again, made a reboot, but the wican does not send mqtt messages after the reboot. Also a 2nd reboot did not change the behavior.

micha53, 2025-06-11:
Yes, I enabled this immediately with th update to 4.12. It is a very smart feature.

meatpiHQ, 2025-06-12:
@micha53 please try this firmware version. I suspect the ECU polling is failing, and not the MQTT.

[wican-fw_obd_v412-63-gc8d257c.zip](https://github.com/user-attachments/files/20709232/wican-fw_obd_v412-63-gc8d257c.zip)

micha53, 2025-06-14:
I made the update and after the update at first it does not send mqtt. But after a reboot mqtt is working.
I will watch the behaviour the next days.

micha53, 2025-06-18:
little update, actually verified

- returning home it connects to the wifi and  **starts mqtt**

micha53, 2025-06-19:
Yesterday I took a look into the wican code and found that there exist log entries. If I can support you anyhow with log data then give me instructions how to get them.