meatpiHQ/wican-fw #275: WiCAN - BYD Atto 3 more Sensors

Ipirk88, 2024-12-06:
Hello, 
fist of all - to read out the SoC of the Battery is verry easy and works just fine - thanks for that!

But i whant to know more about my car :-)

Is it maby possible to read out more values for example:
 * min/max Cell Voltage to calculate the voltage spread between the cells
 * Battery total Voltage
 * min/max Cell Temp 
 * driven km
 * charge Power that goes into the Battery (to calculate the Charge efficency or what is used for the rest of the Car - Heatpump)
 * Voltage of the 12V battery
 
 And one Question more - does the WiCAN also support the BYD Seal U? 
 
 Thank you so much!

meatpiHQ, 2024-12-13:
@Ipirk88 I've add the battery voltage, you should sync the profile and reboot. For the other parameters we need to look at the logs from car scanner. 

Do you see the other parameters in car scanner?

meatpiHQ, 2024-12-13:
Yes the BYD seal U should be the same profile.

Ipirk88, 2024-12-13:
Great - thank you very much - I have this Car Scanner logs:
[2024-12-07 18-05-23.csv](https://github.com/user-attachments/files/18126496/2024-12-07.18-05-23.csv)
Did you need another one or is this enough?

Ipirk88, 2024-12-13:
sorry wrong file - here is the correct one 
[2024-12-07 13-12-19.csv](https://github.com/user-attachments/files/18126524/2024-12-07.13-12-19.csv)

Ipirk88, 2024-12-13:
OK - so how to do? I don't get it fully?!

So first connect, than go on "all Sensors" then click on the sensor?! Is that right
[image]
[image]
[image]

Is that right? And than?

l2smith2, 2024-12-18:
@meatpiHQ I updated my profile and now have the HV battery values except in HA it shows as 40000 instead of 400v

Ill have a look at doing some logs of the values I'd like to see

dalathegreat, 2025-05-14:
I have reverse engineered a few Atto 3 PID polls, available in this repo incase you want to copy it here :) 

https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/battery/BYD-ATTO-3-BATTERY.cpp#L15

loryanstrant, 2025-05-25:
> I have reverse engineered a few Atto 3 PID polls, available in this repo incase you want to copy it here :)
> 
> https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/battery/BYD-ATTO-3-BATTERY.cpp#L15

This is awesome, thanks for sharing. These look like mainly battery related though, unless I'm missing something?
Have you added these to your WiCAN at all? If so, are you able to share the configuration please?

loryanstrant, 2025-05-25:
> No I dont use WiCAN, I just found this thread while googling and looking for more info on BYD Atto 3 PID polling, so I thought I would reference back here incase you guys need it! Thanks for the original content here also 🙌

Thanks, it's appreciated!
I haven't been able to find a central location with PID listing, just scraps from people's code and conversations, so I thought I'd set up a repo to document them here: https://github.com/loryanstrant/BYD-PID-list/
It would be great if you could add any you've found!